cocotb/cocotb #377: trying to run cocotb 1.0 with 32-bit modelsim on 64-bit ubuntu wily

qubits1, 2016-01-07:
my gcc and g++ version is 5.2.1
simulator: modelsim 10.2 (32-bit)
I get an error when I try to run the adder example
>ARCH=i686 make SIM=modelsimmake results.xml
make[1]: Entering directory '/home/agohil/proj/cocotb/examples/adder/tests'
make -C /home/agohil/proj/cocotb/lib/gpi_log
make[2]: Entering directory '/home/agohil/proj/cocotb/lib/gpi_log'
gcc -m32 -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -Wstrict-prototypes -Waggregate-return -DFILTER -c -I/home/agohil/proj/cocotb/include -I/opt/pym32/include/python2.7  -o /home/agohil/proj/cocotb/build/obj/i686/gpi_logging.o gpi_logging.c
gcc -m32 -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -Wstrict-prototypes -Waggregate-return -DFILTER -shared -Xlinker -L/opt/pym32/lib -o /home/agohil/proj/cocotb/build/libs/i686/libgpilog.so /home/agohil/proj/cocotb/build/obj/i686/gpi_logging.o   -lpython2.7 -lpthread -ldl -lutil -lm 
make[2]: Leaving directory '/home/agohil/proj/cocotb/lib/gpi_log'
make -C /home/agohil/proj/cocotb/lib/embed
make[2]: Entering directory '/home/agohil/proj/cocotb/lib/embed'
gcc -m32 -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -Wstrict-prototypes -Waggregate-return -DPYTHON_SO_LIB=libpython2.7.so -c -I/home/agohil/proj/cocotb/include -I/opt/pym32/include/python2.7  -o /home/agohil/proj/cocotb/build/obj/i686/gpi_embed.o gpi_embed.c
gcc -m32 -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -Wstrict-prototypes -Waggregate-return -DPYTHON_SO_LIB=libpython2.7.so -shared -Xlinker -L/opt/pym32/lib -o /home/agohil/proj/cocotb/build/libs/i686/libcocotb.so /home/agohil/proj/cocotb/build/obj/i686/gpi_embed.o   -lpython2.7 -lpthread -ldl -lutil -lm -lgpilog -lcocotbutils -L/home/agohil/proj/cocotb/build/libs/i686
make[2]: Leaving directory '/home/agohil/proj/cocotb/lib/embed'
make -C /home/agohil/proj/cocotb/lib/gpi EXTRA_LIBS= EXTRA_LIBDIRS=
make[2]: Entering directory '/home/agohil/proj/cocotb/lib/gpi'
gcc -m32 -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -DVPI_CHECKING -DLIB_EXT=so -DSINGLETON_HANDLES -Wstrict-prototypes -Waggregate-return -shared -Xlinker -L/opt/pym32/lib -o /home/agohil/proj/cocotb/build/libs/i686/libgpi.so  /home/agohil/proj/cocotb/build/obj/i686/GpiCbHdl.o /home/agohil/proj/cocotb/build/obj/i686/GpiCommon.o  -lcocotbutils -lgpilog -lcocotb -lstdc++ -L/home/agohil/proj/cocotb/build/libs/i686
make[2]: Leaving directory '/home/agohil/proj/cocotb/lib/gpi'
make -C /home/agohil/proj/cocotb/lib/simulator
make[2]: Entering directory '/home/agohil/proj/cocotb/lib/simulator'
gcc -m32 -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -Wstrict-prototypes -Waggregate-return  -c -I/home/agohil/proj/cocotb/include -I/opt/pym32/include/python2.7  -o /home/agohil/proj/cocotb/build/obj/i686/simulatormodule.o simulatormodule.c
gcc -m32 -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -Wstrict-prototypes -Waggregate-return  -shared -Xlinker -L/opt/pym32/lib -o /home/agohil/proj/cocotb/build/libs/i686/libsim.so /home/agohil/proj/cocotb/build/obj/i686/simulatormodule.o   -lgpi -lgpilog -lpython2.7 -lpthread -ldl -lutil -lm -L/home/agohil/proj/cocotb/build/libs/i686
ln -sf libsim.so /home/agohil/proj/cocotb/build/libs/i686/simulator.so
make[2]: Leaving directory '/home/agohil/proj/cocotb/lib/simulator'
make -C /home/agohil/proj/cocotb/lib/vpi EXTRA_LIBS= EXTRA_LIBDIRS=
make[2]: Entering directory '/home/agohil/proj/cocotb/lib/vpi'
gcc -m32 -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -DVPI_CHECKING -Wstrict-prototypes -Waggregate-return -shared -Xlinker -L/opt/pym32/lib -o /home/agohil/proj/cocotb/build/libs/i686/libvpi.so  /home/agohil/proj/cocotb/build/obj/i686/VpiImpl.o /home/agohil/proj/cocotb/build/obj/i686/VpiCbHdl.o   -lgpilog -lgpi -lstdc++  -L/home/agohil/proj/cocotb/build/libs/i686
ln -sf libvpi.so /home/agohil/proj/cocotb/build/libs/i686/gpivpi.vpl
ln -sf libvpi.so /home/agohil/proj/cocotb/build/libs/i686/cocotb.vpi
make[2]: Leaving directory '/home/agohil/proj/cocotb/lib/vpi'
make -C /home/agohil/proj/cocotb/lib/fli EXTRA_LIBS= EXTRA_LIBDIRS=
make[2]: Entering directory '/home/agohil/proj/cocotb/lib/fli'
g++ -m32 -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -DFLI_CHECKING -DUSE_CACHE -c -I/home/agohil/proj/cocotb/include -I/opt/pym32/include/python2.7 -I/opt/modelsim/modelsim_dlx/bin/../include -o /home/agohil/proj/cocotb/build/obj/i686/FliImpl.o FliImpl.cpp
FliImpl.cpp: In function ‘void fli_mappings(GpiIteratorMapping<int, FliIterator::OneToMany>&)’:
FliImpl.cpp:517:24: error: ‘accElsifGenerate’ was not declared in this scope
     map.add_to_options(accElsifGenerate, &region_options[0]);
                        ^
FliImpl.cpp:518:24: error: ‘accElseGenerate’ was not declared in this scope
     map.add_to_options(accElseGenerate, &region_options[0]);
                        ^
FliImpl.cpp:520:24: error: ‘accCaseGenerate’ was not declared in this scope
     map.add_to_options(accCaseGenerate, &region_options[0]);
                        ^
FliImpl.cpp:521:24: error: ‘accCaseOTHERSGenerate’ was not declared in this scope
     map.add_to_options(accCaseOTHERSGenerate, &region_options[0]);
                        ^
FliImpl.cpp: In member function ‘virtual GpiIterator::Status FliIterator::next_handle(std::__cxx11::string&, GpiObjHdl**, void**)’:
FliImpl.cpp:693:30: error: cast from type ‘const char*’ to type ‘void*’ casts away qualifiers [-Werror=cast-qual]
         mti_VsimFree((void *)c_name);
                              ^
cc1plus: all warnings being treated as errors
/home/agohil/proj/cocotb/makefiles/Makefile.rules:49: recipe for target '/home/agohil/proj/cocotb/build/obj/i686/FliImpl.o' failed
make[2]: *** [/home/agohil/proj/cocotb/build/obj/i686/FliImpl.o] Error 1
make[2]: Leaving directory '/home/agohil/proj/cocotb/lib/fli'
/home/agohil/proj/cocotb/lib/Makefile:65: recipe for target '/home/agohil/proj/cocotb/build/libs/i686/libfli.so' failed
make[1]: *** [/home/agohil/proj/cocotb/build/libs/i686/libfli.so] Error 2
make[1]: Leaving directory '/home/agohil/proj/cocotb/examples/adder/tests'
/home/agohil/proj/cocotb/examples/adder/tests/../../../makefiles/Makefile.sim:71: recipe for target 'sim' failed
make: *** [sim] Error 2

stuarthodgson, 2016-01-09:
This seems a duplicate of #375

stuarthodgson, 2016-01-09:
@qubits1 does 2409ffb.. resolve this for you?

I suspect you where on master rather than 1.0, if you do not require the new features being worked on with FLI then maybe use the [1.0](https://github.com/potentialventures/cocotb/releases/latest) until these are resolved.

stuarthodgson, 2016-01-20:
closing as duplicate.

closed 2016-01-20 by stuarthodgson