cocotb/cocotb #381: IUS Support has some issues

leftink, 2016-01-11:
When running an ius simulation, the directory with the makefile is "dirtied" with several files that should be up in the sim_build folder.

Also, I noticed that the "results.xml" is not located in the same location as in the questa and aldec simulations.  In IUS the "results.xml", ends up in the same directory as the makefile.  In questa and aldec, the "reulsts.xml" ends up in the "sim_build" folder because of the change directory command that happens before running the sim.

@chiggs and @stuarthodgson What folder should results.xml be located in at the end of the simulation?  Currently it looks like ALDEC and Questa may behave differently.  And from the way the makefiles are, it seems like you want it to be in same directory as the makefile so the simulation won't run again if nothing has changed.

Also, I have noticed that there is an issue with vhpi simulations.  When starting from a clean build, if I run an VHPI simulation first, the test will always fail.  I see the following error message "VHPI ERROR: Registered callback isn't enabled!  Got 1000" in VhpiCbHdl.cpp:214.  I run the same test again without cleaning using VPI and it passes successfully.  I then run it one more time without cleaning, but this time targeting VHPI and works.  Something odd is happening that doesn't allow VHPI to work unless VPI sim has been run first.

leftink, 2016-01-11:
As for the last issue, it looks like there is just an issue with VHPI.  When I thought the VHPI test was passing, that was actually the VPI interface working again.

stuarthodgson, 2016-01-12:
@leftink , Taking the endian_swapper as an example, yes there is indeed something "odd" with IUS. I also seem not to have documented it anywhere. The IUS implementation of VHPI is far from complete. We can not query the state of a callback for instance. This leads to the the error 

```
VHPI ERROR: Registered callback isn't enabled! Got 1000
```

at the start of the simulation, VCS and Aldec work fine in this regard. The outcome of this is that on IUS the VHPI interface can only be used to modify values and read them back. Therefore a mixed language simulation will work fine as the VPI interface can be used to do the scheduling.

The current implementation will use the first registered interface layer as the scheduling one and IUS helpfully always loads VPI even under vhdl tests so that sort of works, but again only for mixed language. We also only use the first interface to obtain the root handle so in the endian_swapper test with TOPLEVEL_LANG=vhdl we do not work.

I do have some patches that no longer checked the state of the callback once we arm it, none of the the other interfaces implementations do this so seems safe enough. But then another problem with IUS hits, that the way it schedules events is different to VPI and seems wrong. A value change will result in a callback to before the function to modify the value has returned.

Since we are not IUS users and the only person we are aware of was using mixed language and so did not see these problems it has been on the back burner.

leftink, 2016-01-12:
@stuarthodgson That is fine, we aren't big users of IUS anymore either, but it was the only simulator I had that would let me do VHPI.  I think we are will be able to do some testing with Aldec in the next month or two so I can just wait until then.

What about the issue with the makefile?  Where are you guys expecting the results.xml to be located?  I have a feeling that the Aldec and Questa makefiles need to be fixed.

@chiggs and @stuarthodgson I also dug into the issue with IUS always loading the VPI interface regardless if it is told to or not.  It looks like IUS has a "feature" that it will search the shared libraries in its path and if any have the symbol "vlog_startup_routines[]" defined they will be automatically loaded.  According to their documentation if you want to control loading through the command line then this symbol shouldn't be present.  What would you guys think of adding a compile flag that will name mangle the "vlog_startup_routines" symbol?  Then we can load things normally for IUS.  I did a quick test with the name mangling and everything worked as expected.

stuarthodgson, 2016-01-20:
@leftink , we already have a -DIUS as there is another feature where they ignored the standard for VHPI and changed a structure members type, took a while to find that feature. So doing some name mangling based on that #define should be pretty easy. We where doing so well at not having any simulator specific features until IUS as well.

ktbarrett, 2025-03-27:
Some of these issues have been resolved. We are now testing Xcelium in CI. Open up new issues with specific failures going forward.

closed 2025-03-27 by ktbarrett