cocotb/cocotb #382: simulator time is not correctly handled above 4.3ms

chevillotn, 2016-01-19:
The following code:
--------------
@cocotb.test()
def fpga_test(dut):
    while True:
        yield Timer((100e-6)/1e-12)
        timeh, timel = simulator.get_sim_time()
        dut.log.info("DBG1: timeh=%d=0x%x timel=%d=0x%x" % (timeh, timeh, timel, timel))
--------------
(Questa simulator is configured for ps)
gives the following output:
--------------
#  100000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=100000000=0x5f5e100
#  200000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=200000000=0xbebc200
#  300000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=300000000=0x11e1a300
#  400000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=400000000=0x17d78400
#  500000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=500000000=0x1dcd6500
#  600000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=600000000=0x23c34600
#  700000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=700000000=0x29b92700
#  800000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=800000000=0x2faf0800
#  900000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=900000000=0x35a4e900
#  1000000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=1000000000=0x3b9aca00
#  1100000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=1100000000=0x4190ab00
#  1200000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=1200000000=0x47868c00
#  1300000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=1300000000=0x4d7c6d00
#  1400000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=1400000000=0x53724e00
#  1500000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=1500000000=0x59682f00
#  1600000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=1600000000=0x5f5e1000
#  1700000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=1700000000=0x6553f100
#  1800000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=1800000000=0x6b49d200
#  1900000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=1900000000=0x713fb300
#  2000000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=2000000000=0x77359400
#  2100000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=2100000000=0x7d2b7500
#  2200000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=2200000000=0x83215600
#  2300000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=2300000000=0x89173700
#  2400000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=2400000000=0x8f0d1800
#  2500000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=2500000000=0x9502f900
#  2600000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=2600000000=0x9af8da00
#  2700000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=2700000000=0xa0eebb00
#  2800000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=2800000000=0xa6e49c00
#  2900000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=2900000000=0xacda7d00
#  3000000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=3000000000=0xb2d05e00
#  3100000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=3100000000=0xb8c63f00
#  3200000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=3200000000=0xbebc2000
#  3300000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=3300000000=0xc4b20100
#  3400000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=3400000000=0xcaa7e200
#  3500000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=3500000000=0xd09dc300
#  3600000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=3600000000=0xd693a400
#  3700000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=3700000000=0xdc898500
#  3800000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=3800000000=0xe27f6600
#  3900000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=3900000000=0xe8754700
#  4000000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=4000000000=0xee6b2800
#  4100000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=4100000000=0xf4610900
#  4200000.00ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=0=0x0 timel=4200000000=0xfa56ea00
#   5032.70ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=1=0x1 timel=5032704=0x4ccb00
#  105032.70ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=1=0x1 timel=105032704=0x642ac00
#  205032.70ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=1=0x1 timel=205032704=0xc388d00
#  305032.70ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=1=0x1 timel=305032704=0x122e6e00
#  405032.70ns INFO     cocotb.fpga_tb_v                             fpga_tb.py:226  in fpga_test                       DBG1: timeh=1=0x1 timel=405032704=0x18244f00

luked80, 2016-01-22:
@chevillotn #115, #289 and #343 are all related to the same issue.
I have something working for simulations that include the VPI.
I'll polish it up a little and publish details.

stuarthodgson, 2016-04-01:
Closing as this was covered in the work from @luked80 and myself. If people disagree then please repoen.

closed 2016-04-01 by stuarthodgson