cocotb/cocotb #384: Transactions in driver

mciepluc, 2016-01-23:
Hello,

I am a bit confused with the code of the BusDriver

```python
    @coroutine
    def _driver_send(self, transaction, sync=True):
        if sync:
            yield RisingEdge(self.clock)
        self.bus <= transaction
```

If I understand the concept here, a __le__ operator is expected to be used for sending transaction objects. However, a Bus.drive() (__le__) method operates only on bus signal items. Could you please explain?

mciepluc

chiggs, 2016-01-24:
@mciepluc in this case, `self.bus` is an instance of a `Bus` class (see https://github.com/potentialventures/cocotb/blob/master/cocotb/bus.py).

This has a method called `drive` whereby we can apply any object to a bus, provided it has attributes that match the signal names of the bus.  This way we don't have to know up-front what types we might want to drive, provided there are members that match the signal names we can accept any type.

For example, if we had a simple bus comprising the signals:

```
bus_address
bus_read
bus_write
```

Then we could create a class that encapsulates each cycle on the bus:

```
class MyBusBeat(object):
    def __init__(self, address, read=False):
        self.address = address
        self.read = 1 if read else 0
        self.write = 0 if read else 1
```

Then we could pass an iterable to the `BusDriver` that yielded individual `MyBusBeat` instances for each cycle of the transaction.

mciepluc, 2016-01-24:
Thanks Chris,

According to this concept it is required that a ```transaction``` contains signal-level attributes.
I would rather prefer a UVM-based approach, where the driver is fully responsible for translating transaction objects into signals. Using the name "transaction" may be therefore a bit confusing and I suggest to discuss that. 

To give an example:

```python
class Transaction(object):
    def __init__(self, address, data, read=False, delay=0):
        ...

In the Driver code:
def execute_transaction(self, transaction):
    self.bus.address <= transaction.address
    self.bus.write <= 0 if transaction.read else 1
    self.bus.write <= 1 if not transaction.read else 0
    self.bus.data <= transaction.data
    for i in range(delay):
        yield clkedge

```

For me it is more verbose approach.
Instead of naming a ```_driver_send()``` argument a transaction (which is confusing) it could be named e.g. signal_sequence or something like that.