cocotb/cocotb #386: Modelsim libstdc++.so.6: version `GLIBCXX_3.4.21' not found

petspats, 2016-01-28:
Hi!
Trying to use this great work with 32 bit modelsim(Altera edition). Following error keeps popping up:
```
# ** Error (suppressible): (vsim-3197) Load of "/home/gkarm/cocotb/build/libs/i686/libvpi.so" failed: /home/gkarm/altera/modelsim_ase/gcc-4.7.4-linux/lib/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /home/gkarm/cocotb/build/libs/i686/libvpi.so).
# ** Error (suppressible): (vsim-PLI-3002) Failed to load PLI object file "/home/gkarm/cocotb/build/libs/i686/libvpi.so".
```
I believe i have tried all solutions found from Google, but no help..any suggestions?

Here is a longer segment from **ARCH=i686 make SIM=modelsim**

```
# 10.4b

# do runsim.do
# Model Technology ModelSim ALTERA vlog 10.4b Compiler 2015.05 May 27 2015
# Start time: 23:56:42 on Jan 28,2016
# vlog -work work -timescale 1ns/100ps -mfcu "+acc=rmb" -sv /home/gkarm/cocotb/tests/test_cases/test_cocotb/../../../tests/designs/sample_module/sample_module.v 
# -- Compiling package sample_module_v_unit
# -- Compiling module sample_module
# 
# Top level modules:
# 	sample_module
# End time: 23:56:42 on Jan 28,2016, Elapsed time: 0:00:00
# Errors: 0, Warnings: 0
# vsim -onfinish exit -pli "libvpi.so" work.sample_module 
# Start time: 23:56:42 on Jan 28,2016
# Loading sv_std.std
# Loading work.sample_module_v_unit
# Loading work.sample_module
# Loading /home/gkarm/cocotb/build/libs/i686/libvpi.so
# ** Error (suppressible): (vsim-3197) Load of "/home/gkarm/cocotb/build/libs/i686/libvpi.so" failed: /home/gkarm/altera/modelsim_ase/gcc-4.7.4-linux/lib/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /home/gkarm/cocotb/build/libs/i686/libvpi.so).
# ** Error (suppressible): (vsim-PLI-3002) Failed to load PLI object file "/home/gkarm/cocotb/build/libs/i686/libvpi.so".
#    Time: 0 ps  Iteration: 0  Root: /
# Error loading design
Error loading design
# End time: 23:56:42 on Jan 28,2016, Elapsed time: 0:00:00
# Errors: 2, Warnings: 0
make[3]: Leaving directory '/home/gkarm/cocotb/tests/test_cases/test_cocotb'
make[2]: Leaving directory '/home/gkarm/cocotb/tests/test_cases/test_cocotb'
cd test_cases/test_configuration && make
make[2]: Entering directory '/home/gkarm/cocotb/tests/test_cases/test_configuration'
make results.xml
make[3]: Entering directory '/home/gkarm/cocotb/tests/test_cases/test_configuration'
mkdir -p sim_build
echo "if [file exists work] {vdel -lib work -all}" > sim_build/runsim.do
echo "vlib work" >> sim_build/runsim.do
echo "vcom -work work  /home/gkarm/cocotb/tests/test_cases/test_configuration/../../../tests/designs/vhdl_configurations/dut.vhd /home/gkarm/cocotb/tests/test_cases/test_configuration/../../../tests/designs/vhdl_configurations/configurations.vhd" >> sim_build/runsim.do    
echo "vlog -work work -timescale 1ns/100ps -mfcu +acc=rmb -sv  /home/gkarm/cocotb/tests/test_cases/test_configuration/../../../tests/designs/vhdl_configurations/testbench.sv" >> sim_build/runsim.do
echo "vsim -onfinish exit -pli libvpi.so work.testbench" >> sim_build/runsim.do
echo "run -all" >> sim_build/runsim.do
echo "quit" >> sim_build/runsim.do
cd sim_build && LD_LIBRARY_PATH=/home/gkarm/cocotb/build/libs/i686:/usr/lib/x86_64-linux-gnu:/opt/pym32/lib:/opt/pym32/lib:/opt/pym32/lib:/opt/pym32/lib:/opt/pym32/lib:/opt/pym32/lib MODULE=test_configurations TESTCASE= TOPLEVEL=testbench \
GPI_EXTRA=fli  TOPLEVEL_LANG=verilog PYTHONPATH=/home/gkarm/cocotb/build/libs/i686:/home/gkarm/cocotb:/home/gkarm/cocotb/tests/test_cases/test_configuration:/mnt/6720e4ef-2b8a-43fe-b047-5dd6b3f65cce/target/lib/python2.7/dist-packages:/home/gkarm/GNURadio/target/python/:/home/gkarm/GNURadio/target/lib/python2.6/site-packages/:/home/gkarm/GNURadio/target/lib64/python2.6/site-packages/:/home/gkarm/GNURadio/target/lib/python2.6/dist-packages/:/home/gkarm/GNURadio/target/lib64/python2.6/dist-packages/:/home/gkarm/GNURadio/target/lib/python2.7/site-packages/:/home/gkarm/GNURadio/target/lib64/python2.7/site-packages/:/home/gkarm/GNURadio/target/lib/python2.7/dist-packages/:/home/gkarm/GNURadio/target/lib64/python2.7/dist-packages/:/home/gkarm/GNURadio/target/python/:/home/gkarm/GNURadio/target/lib/python2.6/site-packages/:/home/gkarm/GNURadio/target/lib64/python2.6/site-packages/:/home/gkarm/GNURadio/target/lib/python2.6/dist-packages/:/home/gkarm/GNURadio/target/lib64/python2.6/dist-packages/:/home/gkarm/GNURadio/target/lib/python2.7/site-packages/:/home/gkarm/GNURadio/target/lib64/python2.7/site-packages/:/home/gkarm/GNURadio/target/lib/python2.7/dist-packages/:/home/gkarm/GNURadio/target/lib64/python2.7/dist-packages/ \
vsim -c -do runsim.do 2>&1 | tee sim.log
Reading pref.tcl
```

stuarthodgson, 2016-01-29:
@petspats this looks to be because the version of libstd++ used at compile time can not be found at runtime. 

You seem to have a prior LD_LIBRARY_PATH set in your environment before running the test. The last line of your output shows

```
cd sim_build && LD_LIBRARY_PATH=/home/gkarm/cocotb/build/libs/i686:/usr/lib/x86_64-linux-gnu:/opt/pym32/lib:/opt/pym32/lib:/opt/pym32/lib:/opt/pym32/lib:/opt/pym32/lib:/opt/pym32/lib MODULE=test_configurations TESTCASE= TOPLEVEL=testbench \
GPI_EXTRA=fli  TOPLEVEL_LANG=verilog 
```

I would not expect to see ```/usr/lib/x86_64-linux-gnu``` being used on a 32 bit system and is not on the output of a test when I run with modelsim altera edition 32bits under a 64bit ubuntu based distro.

Before running the test can you do 

```
$> echo $LD_LIBRARY_PATH
```

If this set then unset with

```
$> unset LD_LIBRARY_PATH
```

and try the test again. If this does not resolve the issue then can you post the output of the compilation stage.

petspats, 2016-01-29:
@stuarthodgson thanks for the help, that lead me to fix.
Here is the fix:
` mv YOUR_ALTERA_PATH/modelsim_ase/gcc-4.7.4-linux/lib/libstdc++.so.6 YOUR_ALTERA_PATH/modelsim_ase/gcc-4.7.4-linux/lib/libstdc++.so.6.BACKUP`
This renames the problematic library and forces the use of system library, correct me if i am wrong.
Probably related to the use of Ubuntu 15.10 and the GCC version of 5.2.1.This did not happen with Ubuntu 14.

Problem solved :)

closed 2016-02-05 by stuarthodgson