cocotb/cocotb #387: Debugging python testbench

petspats, 2016-01-29:
After writing a testbench one would like to debug it, how could we archive this in cocotb?
I tried adding `import pdb; pdb.set_trace()` - this just stucks the make process (no debugger controls).
Currently there is a "makefile barrier", I am unable to understand what is going on under the hood :(

Ideally i could run my python testbench from Eclipse Pydev, but i would be happy with pdb as-well.

leftink, 2016-01-30:
I have successfully used the pydev debug server and their version of the set_trace() that communicates with the server to be able to debug scripts.

petspats, 2016-02-05:
@leftink thanks for the infos, i got the remote debugging working.
I am a bit new to this,  could someone share tips on  developing with cocotb. Currently i tried using Eclipse and PyDev but the design flow seems to be really painful, mainly because i have to run makefiles on different console.. there must be a way to improve this.

Thanks for the help guys :)

closed 2016-10-15 by petspats