cocotb/cocotb #389: update to Makefile.questa to support win7 in VHDL mode

diarmuid, 2016-02-03:
I'm using Win7 and questasim.

When running a VHDL test,, line 50 of the Makefile.questa should not have the explicit file extension on libfli.so but libfli.$(LIB_EXT)

stuarthodgson, 2016-02-05:
@diarmuid , simple change but if you could test I would appreciate it. 441d71

diarmuidcwc, 2016-02-05:
Ok. I've forked the repo and I'm making some updates to add support for Aldec Active-HDL. Works ok for me so I'll test and commit them all back in one go

elgorwi, 2016-02-12:
@stuarthodgson, just tried the endian swapper example and it runs fine using either VPI or FLI.
However, I had to make following change to line 133 of Makefile.questa to make it run:
```makefile
INT_LIBS := $(COCOTB_VPI_LIB) $(COCOTB_FLI_LIB)
```
With this change windows and linux versions of INT_LIBS are the same.

In addition, there is an issue with the runsim.do not being updated when only the variables passed from the command line are changed between different runs. eg. changing TOPLEVEL_LANG=vhdl to TOPLEVEL_LANG=verilog will not work correctly.
I've solved this a while ago by changing runsim.do to a PHONY target. I can bring my old PR (Fli under windows #214 ) up to date if that would be of interest.

stuarthodgson, 2016-02-12:
@elgorwi I'll make the change to line 133 and close this off, thanks for the testing. 

As for #214 does much of that apply any more? Been quite a few changes to FLI and the quests/modelsim makefiles of late. If you think it does though then please update the PR and I'll take a look.

Stu

stuarthodgson, 2016-04-01:
@elgorwi have added your changes (a bit late sorry). reopen if not fixed.

closed 2016-04-01 by stuarthodgson

elgorwi, 2016-04-04:
@stuarthodgson,

Thanks for the update.
However, the updating of runsim.do was already fixed in #408 similar to Makefile.aldec (c8bdb8df877bba5e7f7ff10f721801775f3cc8e6)
Not sure if the .PHONY target is still necessary now.

Thanks