cocotb/cocotb #393: Clarification/Consistency Needed for Array Support across VHPI, VPI, and FLI

leftink, 2016-02-25:
@chiggs and @stuarthodgson 

With the delivery of #392, all three interfaces (VPI, VHPI, and FLI) have roughly the same level of implmentation completed.  During my clean-up of the FLI interface, I came across some inconsistencies between the interfaces as well as some current implementations that I think should be changed in regards to the handling of Array-like structures, i.e. Arrays, Arrays of Arrays, Multi-dimensional arrays, and generate loops.

**Note: Any issues mentioned below were tested on QuestaSim and/or IUS.**

## Generate Loops
Assume we have the following VHDL (or Verilog equivalent) generate loop under the "dut" (i.e. sample_module):
```vhdl
a_gen : for i in 6 to 10 generate
begin
  ...
end generate a_gen;
```

The first thing I noticed with VHPI, is that it doesn't find anything with the name ":sample_module.a_gen".  Is this the expected behavior?  FLI actually has a valid "handle" for "/sample_module/a_gen", but I forced the software to reject this so the behavior is the same between VHPI and FLI and so the python hierarchy stays the same.

The major draw back to the this approach is that the first time someone tries to navigate through a generate loop, i.e. "dut.a_gen[0]", the software will fail to find a handle for "dut.a_gen" which means it will revert to "discover_all" and that could take some time depending on the File.

The other issue with the way the generate loops are handle in python has to do with the indices.  The code in handle.py parses the handle name and builds a python list and continues to "extend" the list based on the indices received.  My example, however, shows that the indicies don't have to be zero-based which means two things happen.

1. len(dut.a_gen) == 11   -> this is wrong
2. for reg in dut.a_gen:  -> will log 6 warnings since the list will contain None in dut.agen[0:6]

One solution to this last problem would be to crete a dictionary instead of a list, but that would require handling things differently when iterating.  A better solution would be to have a specialized Hierarchy class that behave like the other handles, but only stores the necessary indicies and can return the proper len().

## Single Dimension Array and Vector Indexing
Assume we have the following VHDL (or Verilog equivalent) signals defined under the "dut" (i.e. sample_module):

```vhdl
signal a1 : std_logic_vector(7 downto 0);            -- reg [7:0] a1;
signal a2 : std_logic_vector(0 to 7);                -- reg [0:7] a2;
signal a3 : std_logic_vector(10 downto 6);           -- reg [10:6] a3;
```

Currently there is a discrepency between the interfaces for which handle will be returned.  I modified FLI to behave like VHPI and personally I am not a fan of how the VHPI interface behaves.

For Python code "dut.a1[0]" the following handles are returned:
>    VPI: ":sample_module.a1(0)"
>   VHPI: ":sample_module.a1(7)"
>    FLI: "/sample_module/a1(7)"
	
For Python code "dut.a2[0]" the following handles are returned:
>    VPI: ":sample_module.a2(0)"
>   VHPI: ":sample_module.a2(0)"
>    FLI: "/sample_module/a2(0)"
	
For Python code "dut.a3[10]" the following handles are returned:
>    VPI: Not Sure
>   VHPI: Not Found...Error valid indices range from 0 to 4
>    FLI: Not Found...Error valid indices range from 0 to 4
	
When indexing from python, what are the valid indices and which handles should be returned for the given signals?  My opinion is the indicies should match whatever is defined in the HDL, i.e. if the signal is defined as "10 downto 6" then in python one should be able to access a bit by "dut.a3[10]".  Any sort of mapping from that index to an index that makes the interface happy should be handled by that interface.

Depending on the decision for what the valid python indices are, we will have to make sure when iterating over the array/vector that generated indicies follow the same pattern.  Currently, 0 to len()-1 is assumed for the valid indicies when iterating.

## Multi-Dimension Array Indexing
Assume we have the following VHDL (or Verilog equivalent) signal defined under the "dut" (i.e. sample_module):

```vhdl
type exType is array (0 to 3, 0 to 6) of signed(10 downto 0);

signal sig_ex : exType;     -- reg [10:0] sig_ex [0:3] [0:6];
```

### Direct Access
How would you expect the python code to look when trying to access a signle bit in the array?

In my opinion, the python code should be "dut.sig_ex[x][y][z]".  I tried this with the different interfaces ang got varying results.

For Python Code "dut.sig_ex[x][y][z]":
>   FLI (Questa): Works, with the caveat mentioned in the previous section, i.e. z=0 will return bit index 10
>     VHPI (IUS): Doesn't work, seems to be treating the first two dimensions as a single index, i.e. ":sample_module.sig_ex(x,y)" would be for "dut.sig_ex[w]"
>   VPI (Questa): Doesn't work, reports that it can't find index 'x'.  "dut.sig_ex" was identified as (vpiArrayVar) with 28 elements
>      VPI (IUS): Works as one would expect.  "dut.sig_ex" was identified as (vpiRegArray) with 28 elements
	  
### Iterating
In my opinion, when iterating, the python hierarchy should be the same whether using the "discover_all" iterating capability or going through direct access.  Currently this is not the case and then it ends up with multiple handles in the _sub_handle dictionary pointing to the same thing and you don't get the benefit of the caching.

Further, some of the names aren't being handled properly and the names are being registered int he _sub_handles dictionary, but there is no "pythonic" way to access them.  An example of this is the VHPI interface would have the following keys in the _sub_handles dictionary for handle "dut.sig_ex": "sig_ex(0,0), sig_ex(0,1), etc".  Due to the parenthesis in the names, these can't be accessed through the standard hierarchy navigation.

Ideally, the python hierarchy should be the same regardless of the interface used, but because all three interface report the names differently when iterating the hierarchies are not the same and none of them build the hierarchy that is created through direct access like mentioned above which is what I think it should look like.

I have seen issues with all three interfaces when iterating.  FLI and VHPI primarily have the issue of creating "handle" names that aren't accessible through the python hierarchy navigation.  The VPI seems to have an issue because "dut.sig_ex" reports there are 28 elements and so iterating starts out fine, but has issues because it doesn't stop iterating at "dut.sig_ex[3][6]", but instead tries to index "dut.sig_ex[4]".

Another issue is that becasue multi-dimensional arrays are being treated as "HierarchyObject"'s the python hierarchy is being created correctly.  After iteration, valid paths are "dut.sig_ex", "dut.sig_ex.sig_ex[0]", "dut.sig_ex.sig_ex[0].sig_ex__0[0]", etc.  We may need to consider a special python type and GPI_TYPE to have an "array" object that behaves like an array, but doesn't support driving/reading values.

## Conclusion
These issues are the last major things that need to get resoolved to allow me to role this out to my company for verification use.  If we can define how the above examples are expected to behave, I can work on making the changes to get everything to behave that way.

chiggs, 2016-02-25:
@leftink thanks for your work going through and summarising the inconsistencies.

You are correct that we should be aiming for an identical interface from Python regardless of the underlying language implementation and I think you're spot on with the expected behaviour.

#### Single Dimension Array and Vector Indexing

I agree with all of your points:

* `dut.a1[0]` should map onto RTL index 0 regardless of the direction of the range.
* `dut.a3[10]` should be possible if an array is defined as `10 downto 6` or `6 to 10`

In short, the Pythonic access should always behave as you would expect, mapping onto the same RTL object.

#### Direct Access

> In my opinion, the python code should be `dut.sig_ex[x][y][z]`

Agreed.  We should be able to delve through however many dimensions of array and access either the whole object when we reach a signal/variable, or index again to access individual bits (if the signal/variable is an array type).

#### Iterating

This is slightly special - there are various things (like generate loops) which it seems are impossible to traverse via names access.  Instead we have to iterate (`discover_all`) and build up an equivalent hierarchy in Python.

This is slightly nasty as we may end up creating far more handles than required, just to satisfy a Pythonic named access through the hierarchy.  Perhaps we could limit the discover_all fall-back to only create hierarchy objects.

In theory returned generate statements are caught by a regex and translated into traversable hierarchy.  There doesn't appear to be a consistent naming convention between languages and simulators, so this is less than ideal.  If you can find a better mechanism please feel free to suggest alternatives.

leftink, 2016-03-12:
@chiggs  @stuarthodgson 
I could really use your opinion on how to handle some different array types in Verilog

Here are examples of four different types I am trying to support:
```Verilog
wire             sig_t1 [0:3];
wire [7:0]       sig_t2 [7:4];
wire [7:0]       sig_t4 [0:3] [7:4];
wire [2:4] [7:0] sig_t6 [0:1];
```
**Signal** | **Type**
------------ | -------------
sig_t1 | vpiNetArray
sig_t1[x] | vpiNet
sig_t2 | vpiNetArray
sig_t2[x] | vpiNet
sig_t4 | vpiNetArray
sig_t4[x] | vpiArrayVar
sig_t4[x][y] | vpiNet
sig_t6 | vpiNetArray
sig_t6[x] | vpiNet

My question, is for these 4 "signals"  what would you consider read/writable and what would you consider hierarchical.  I am especially curious about sig_t6.

leftink, 2016-03-12:
After looking at this more, I think it makes sense to treat the packed arrays as R/W and the un-packed arrays as more hierarchical.  This would be a change from the current implementation because VPI currently maps vpiNetArrays to GPI_ARRAY which maps to a modifiable object.

Would you agree with my first statement?

chiggs, 2016-03-13:
@leftink I would agree - that matches the behaviour in Verilog, e.g. packed arrays can be assigned using any arbitrary vector, however unpacked arrays must be indexed, so that provides a sensible distinction in terms of hierarchy vs. modifiable signal.

leftink, 2016-04-01:
Tested and works in the latest version.  Closing.

closed 2016-04-01 by leftink