cocotb/cocotb #395: test_cases/test_avalon does not work in Questasim v10.2c_5

chevillotn, 2016-02-25:
The following errors occur:
[chevillot@lappc-f561 ~/Projects/ATLAS/git/cocotb/tests/test_cases/test_avalon]$
$ make SIM=modelsim
make results.xml
make[1]: Entering directory `/userselec/chevillot/Projects/ATLAS/git/cocotb/tests/test_cases/test_avalon'
mkdir -p sim_build
echo "if [file exists work] {vdel -lib work -all}" > sim_build/runsim.do
echo "vlib work" >> sim_build/runsim.do
echo "vlog -work work -timescale 1ns/100ps -mfcu +acc=rmb -sv  /userselec/chevillot/Projects/ATLAS/git/cocotb/tests/test_cases/test_avalon/../../../tests/designs/avalon_module/burst_read_master.v" >> sim_build/runsim.do
echo "vsim -onfinish exit -pli libvpi.so work.burst_read_master" >> sim_build/runsim.do
echo "run -all" >> sim_build/runsim.do
echo "quit" >> sim_build/runsim.do
cd sim_build && LD_LIBRARY_PATH=/userselec/chevillot/Projects/ATLAS/git/cocotb/build/libs/x86_64:/userselec/chevillot/tools/lib::/userselec/chevillot/tools/lib:/userselec/chevillot/tools/lib:/userselec/chevillot/tools/lib:/userselec/chevillot/tools/lib MODULE=test_avalon TESTCASE= TOPLEVEL=burst_read_master \
        GPI_EXTRA= TOPLEVEL_LANG=verilog PYTHONPATH=/userselec/chevillot/Projects/ATLAS/git/cocotb/build/libs/x86_64:/userselec/chevillot/Projects/ATLAS/git/cocotb:/userselec/chevillot/Projects/ATLAS/git/cocotb/tests/test_cases/test_avalon: \
        vsim -c -do runsim.do 2>&1 | tee sim.log
Reading pref.tcl

# 10.2c_5

# do runsim.do
# QuestaSim-64 vlog 10.2c_5 Compiler 2013.11 Nov 19 2013
# -- Compiling module scfifo
# -- Compiling module burst_read_master
#
# Top level modules:
#       burst_read_master
# vsim -onfinish exit -pli libvpi.so work.burst_read_master
# ** Note: (vsim-3812) Design is being optimized...
# ** Note: (vsim-3865) Due to PLI being present, full design access is being specified.
# ** Error: /userselec/chevillot/Projects/ATLAS/git/cocotb/tests/test_cases/test_avalon/../../../tests/designs/avalon_module/burst_read_master.v(363): A parameter override is not allowed for localparam 'lpm_width'.
# ** Error: /userselec/chevillot/Projects/ATLAS/git/cocotb/tests/test_cases/test_avalon/../../../tests/designs/avalon_module/burst_read_master.v(364): A parameter override is not allowed for localparam 'lpm_numwords'.
# ** Error: /userselec/chevillot/Projects/ATLAS/git/cocotb/tests/test_cases/test_avalon/../../../tests/designs/avalon_module/burst_read_master.v(365): A parameter override is not allowed for localparam 'lpm_showahead'.
# ** Error: /userselec/chevillot/Projects/ATLAS/git/cocotb/tests/test_cases/test_avalon/../../../tests/designs/avalon_module/burst_read_master.v(366): A parameter override is not allowed for localparam 'use_eab'.
# ** Error: /userselec/chevillot/Projects/ATLAS/git/cocotb/tests/test_cases/test_avalon/../../../tests/designs/avalon_module/burst_read_master.v(367): A parameter override is not allowed for localparam 'add_ram_output_register'.
# ** Error: /userselec/chevillot/Projects/ATLAS/git/cocotb/tests/test_cases/test_avalon/../../../tests/designs/avalon_module/burst_read_master.v(368): A parameter override is not allowed for localparam 'underflow_checking'.
# ** Error: /userselec/chevillot/Projects/ATLAS/git/cocotb/tests/test_cases/test_avalon/../../../tests/designs/avalon_module/burst_read_master.v(369): A parameter override is not allowed for localparam 'overflow_checking'.
# Optimization failed
# Error loading design
Error loading design
make[1]: Leaving directory `/userselec/chevillot/Projects/ATLAS/git/cocotb/tests/test_cases/test_avalon'

chiggs, 2016-02-25:
I believe this should be fixed by https://github.com/potentialventures/cocotb/pull/392

chevillotn, 2016-02-29:
I can't manage to find on which branch this is located? I believe it's not yet merged to master but how can i get it?

elgorwi, 2016-02-29:
The pull request hasn't been merged in yet, so it won't appear in the branches list.
So you can either clone the cocotb repo of @leftink and then checkout the appropriate branch.
Or alternatively, you can checkout the pull request locally as described here:
https://gist.github.com/piscisaureus/3342247

chiggs, 2016-02-29:
Apologies for the delay, I've merged the change now so please try again with current master.

chevillotn, 2016-03-01:
The test works well with latest master, thanks!

closed 2016-03-01 by chevillotn