cocotb/cocotb #396: Use cocotb with 32 bits modelsim on 64 bits machine

chevillotn, 2016-02-25:
I have several machines having several versions of modelsim/questa. Versions provided by Mentor allow to run 64 bits and i have no issues. However, the modelsim version provided by Altera in the Quartus bundle is 32 bits only.

**** If i run one test with top of repository i get in modelsim (32 bits):
# Loading /userselec/chevillot/Projects/ATLAS/git/cocotb/build/libs/x86_64/libvpi.so
# ** Error: (vsim-125) The shared library /userselec/chevillot/Projects/ATLAS/git/cocotb/build/libs/x86_64/libvpi.so being loaded was built on a 64-bit machine. A 64-bit shared library cannot be loaded in a 32-bit simulation. Please use compatible machines to build and load the library
# ** Error (suppressible): (vsim-3197) Load of "/userselec/chevillot/Projects/ATLAS/git/cocotb/build/libs/x86_64/libvpi.so" failed: /userselec/chevillot/Projects/ATLAS/git/cocotb/build/libs/x86_64/libvpi.so: wrong ELF class: ELFCLASS64.
# ** Error (suppressible): (vsim-PLI-3002) Failed to load PLI object file "/userselec/chevillot/Projects/ATLAS/git/cocotb/build/libs/x86_64/libvpi.so".

**** If i uncomment the ARCH:=i686 in Makefile.modelsim i get while compiling cocotb library:
In file included from /userselec/chevillot/tools/include/python2.7/Python.h:58,
                 from gpi_logging.c:30:
/userselec/chevillot/tools/include/python2.7/pyport.h:886:2: error: #error "LONG_BIT definition appears wrong for platform (bad gcc/glibc config?)."

**** If i use a local compiled version of python for 32 bits, i get:
# Unable to open lib libpython2.7.so (libpython2.7.so: wrong ELF class: ELFCLASS64)
#      -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:91   in gpi_print_registered_impl       VPI registered
Traceback (most recent call last):
  File "/userselec/chevillot/Projects/ATLAS/git/cocotb/cocotb/__init__.py", line 35, in <module>
    import logging
  File "/userselec/chevillot/tools/python2.7_m32/lib/python2.7/logging/__init__.py", line 26, in <module>
    import sys, os, time, cStringIO, traceback, warnings, weakref, collections
ImportError: /userselec/chevillot/tools/python2.7_m32/lib/python2.7/lib-dynload/time.so: undefined symbol: PyExc_ValueError

--> maybe i'm close to get it working.. but i would hope i don't need a local 32 bits python version of my own to be running the cocotb...
Have you made some tests on your side?

chiggs, 2016-02-25:
@chevillotn sadly a 32-bit version of Python is required to run with Modelsim 32-bit, however this is usually available via your distribution package manager.  For example, the following commands should let you use Cocotb in 32-bit mode on Ubuntu:

```
sudo dpkg --add-architecture i386
sudo apt-get update
sudo apt-get install -y build-essential
sudo apt-get install -y gcc-multilib g++-multilib lib32z1 lib32stdc++6 lib32gcc1 expat:i386 fontconfig:i386 libfreetype6:i386 libexpat1:i386 libc6:i386 libgtk-3-0:i386 libcanberra0:i386 libpng12-0:i386 libice6:i386 libsm6:i386 libncurses5:i386 zlib1g:i386 libx11-6:i386 libxau6:i386 libxdmcp6:i386 libxext6:i386 libxft2:i386 libxrender1:i386 libxt6:i386 libxtst6:i386
```

Then `make SIM=modelsim ARCH=i686` should work with the Altera version of Modelsim.

AlexanderSpirin, 2017-07-11:
@chiggs I tried proposed commands, but it didn't help to use cocotb with Modelsim 32-bit on Ubuntu 14.04 x64.
I was afraid of installation 32-bit version python2.7:i386 into system, because of long removed packages list and possible system breaking.
After a lot of attempts I solved problem using [pyenv](https://github.com/pyenv/pyenv)
- Install 32-bit python version in local dir
`env CFLAGS="-m32" LDFLAGS="-m32" PYTHON_CONFIGURE_OPTS="--enable-shared" pyenv install 2.7.13`
- Create symbol link to python sources
`sudo ln -s ~/.pyenv/versions/2.7.13/include/python2.7 /usr/include/i386-linux-gnu/python2.7`
- Run make with path to local libpython2.7.so
`LIBRARY_PATH=/home/adun/.pyenv/versions/2.7.13/lib/ make SIM=modelsim ARCH=i686`

jahagirdar, 2019-02-12:
I had to do the following additional step for vsim to find the pli files
` cp /cocotbtest/cocotb/build/libs/i686/* sim_build/
`

lcapossio, 2019-02-14:
> I had to do the following additional step for vsim to find the pli files
> `cp /cocotbtest/cocotb/build/libs/i686/* sim_build/`

Commit b2b1ddb0ecdc3a89b2be4718e779bf4ad2e2cac7 broke my modelsim runs. It seems if I make USER_DIR=$(SIM_ROOT) it fixes it. The problem is associated with it trying to make some symbolic links and getting protocol error. Is this a bug or are we doing something wrong ?

imphil, 2019-02-14:
@lcapossio please open a new issue for your problem as it is not related to this issue.

eric-wieser, 2020-04-09:
@themperek, I assume this can be closed?

themperek, 2020-04-16:
It should be solved now. Please follow instructions and reopen if the issue is still there.

closed 2020-04-16 by themperek