cocotb/cocotb #397: vpi_user.h contains only part of the file defined by the standard

leftink, 2016-02-25:
@chiggs 
Is there a reason the file "cocotb/include/vpi_user.h" only contains parts of the file defined in the standard?

I realize you probably only put in what was necessary to build, but it makes it difficult to add new features that may utilize some of the constants defined in the standard, but not defined in COCOTB's version of vpi_user.h.

Would you be opposed to just updating this file to match the standard, or should we just add the constants as we need them?

chiggs, 2016-02-25:
@leftink only that when the project started back in the mists of time, the specification for [System]Verilog wasn't public.  Since it wasn't clear what the licensing situation was regarding supplying `vpi_user.h` (or other copies seen in the wild), it seemed safer from a legal perspective to write from scratch only the parts we required rather than including code with unknown restrictions.

Now that the standard is publicly available, I don't think there's any problem with using the full header from the specification - it saves going around the loop discovering something is missing!

stuarthodgson, 2016-04-01:
@leftink  This can be closed?

leftink, 2016-04-01:
@stuarthodgson Yes.

closed 2016-04-01 by stuarthodgson