cocotb/cocotb #398: [FLI] Accessing constants in VHDL package

schribl, 2016-02-26:
Hi,

I tried to update my local cocotb installation and found some problems with Modelsim FLI.
To access constants in a package I use some code like

    some_pkg = SimHandle(simulator.get_root_handle("some_pkg"))
    depth = some_pkg.DEPTH.value.integer

This worked quite well in the old version. However, in the current version it does not work. I investigated the problem and found out that iterating over accPackage is not supported. I added an entry to fli_mapping and found out that acc_fetch_fulltype returns 256 (accVectorPort) for the constants and as a result they are not considered as VHDL handles. However when converting them to mtiVariableIdT and calling mti_GetVarKind returns the right integer 1140.

Is there another way to access constants in a package?
Can I somehow help to get this working? I do not have an overview of the FLI and GPI interface and just looked at the sourcecode. But if provided with some info by the developer I might be able to fix it.

chiggs, 2016-03-01:
Hi  @schribl,

@leftink has done a huge chunk of work on FLI to help bring it back in line with the VPI and VHPI layers, this has just been merged to master.  Could you test again on the current tip?  Unfortunately we don't have access to an FLI capable simulator so can't test this.

schribl, 2016-03-01:
I tested the current master branch and can confirm that the issue is gone now. The variable cannot be accessed via the usual way as the class differs. However, it is now possible to access constants in packages.

elgorwi, 2016-03-01:
Hi @schribl, 

Been trying to read a constant from a package but I'm getting following error:  
Send raised exception: 'ConstantObject' object has no attribute 'value'

Could you explain what you did differently from the steps you described in your original post to access constants in packages?

Thanks!

schribl, 2016-03-02:
Hi @elgorwi,

the ConstantObject is declared in handle.py and has a member _value. So `your_const._value` will return the value. However it is not a BinaryValue like other signals you use in cocotb. To achieve this you can use `BinaryValue(your_const._value)` to create a new object. But only use all of this only with extreme caution as it is not the intended use and you are responsible for checking the values.

elgorwi, 2016-03-02:
Hi @schribl,

Thanks for the info, that did the trick.
Had a look at the ConstantObject class in handle.py. The package constants I wanted to read are integer types, so doing following worked as expected:
```python
some_pkg = SimHandle(simulator.get_root_handle("some_pkg"))
depth = int(some_pkg.DEPTH)
```

chiggs, 2016-03-02:
@elgorwi @schribl I think this has uncovered an issue in the new modifications to Cocotb to distinguish between constants etc.

We should preserve the `.value` member of the constant, since there's no guarantee it will be an integer - in fact it could be `X` or other constants that don't convert into Python integers.

I'll take a quick look at fixing this.

chiggs, 2016-03-02:
@schribl the original method should now work again: (`depth = some_pkg.DEPTH.value.integer`) so I'll close this.

closed 2016-03-02 by chiggs