cocotb/cocotb #401: Unable to locate "python-config" for python3 installation

leftink, 2016-03-03:
Using Anaconda3-2.1.0 for my python installation.

The command to get the libs in this installation is "python3-config"

Seems like the makefile.pylib.linux needs to be updated to detect either.

@chiggs when you guys test with Python3 does your distro have a "python3-config" or just the standard "python-config"

leftink, 2016-03-04:
So I poked around in several different python installations we have in our environment, and the only consistent thing I notice for the config is:

python<major_ver>.<minor_ver>-config

All others that had python-config or python3-config just linked back to the above.  Would you be opposed to changing the makefile to use the above for the python config?

stuarthodgson, 2016-03-04:
@leftink I dont.

stuarthodgson, 2016-04-01:
@leftink This can be closed?

leftink, 2016-04-01:
@stuarthodgson Yes this can.  I have been meaning to ask, how do you guys want to handle closing Issues?  Do you want to be the one to close them, or if we opened them and have verified they are fixed do you want us to close them, or do you just want us to put a comment in them saying they can be closed?

stuarthodgson, 2016-04-01:
@leftink Either works for us but always good to have the person that raised it check that it is indeed fixed after the merge to master and either mention that it can be closed or close it if their problem is no longer present. We randomly go thought the bugs and chase them up too.

closed 2016-04-01 by stuarthodgson

cyining, 2016-05-02:
Hi there! I found that this change breaks Darwin builds. And I have a fix for this: 6577ab8264