cocotb/cocotb #402: Issue with TOPLEVEL_LANG=verilog and SIM=aldec, unable to find MODULE

leftink, 2016-03-03:
@chiggs and @stuarthodgson 

Have you guys ever had any issues with the settings in the title?

I am using RivieraPro 2015.06, and I have tried both Anaconda_1.9.1 w/Python2.7 and Anaconda3_2.1.0 w/Python3.4 and all configurations give me the same result.

Basically, COCOTB can't find the MODULE defined in the environment.  I started to load up print statements and what I found is the following:
    In gpi_embed.c, right before embedding the Python module with Py_Intialized(), I print the environment variable for MODULE and it is set correctly.

  In cocotb/__init__.py, I added a loop to print all environment variables in os.environ[] that is executed during the loading of the module, there were 0 environment variables.

Somehow the environment never made it into the embedded Python, which is really strange since it should just inherit from the same process that embedded it.

The other odd thing is that if I set TOPLEVEL_LANG=vhdl and SIM=aldec, everything behaves correctly.

Questa and IUS both work just fine with TOPLEVEL_LANG=verilog.

I am stumped and not sure what to even try at this point.

jeremyherbert, 2016-03-03:
You don't mention an operating system, but if linux, perhaps you could try comparing `strace` dumps for both the working and non-working configuration. Perhaps the environment variables are being written or something like that.

It's a long shot though ;)

chiggs, 2016-03-03:
This is actually the default configuration we use internally, never seen
this error before on any versions of Riviera...

We tend to use system Python rather than anaconda, but I'm still struggling
to think of a failure mode that would break it like this!

leftink, 2016-03-04:
@jeremyherbert OS is Linux, but `strace` really didn't show me anything useful.

@chiggs I thought this was your standard configuration.  I managed to narrow the issue down even more, but it doesn't really tell me why it isn't working.

I print the environment variable MODULE right before Py_Initialize() and the value is good.  Immediately after Py_Initialize(), I execute PyRun_SimpleString() and have it print both "sys.path" and "os.environ.keys()", and from what I see sys.path looks to be initialized correctly, but os.environ.keys() is empty.  It behaves like the os module failed to initialize correctly.

leftink, 2016-03-04:
One final update and then I am going to close this issue.

With RivieraPro 2015.06, I could never get this to work with Anaconda_1.9.1 or Anaconda3_2.1.0.  I even tried a standard Python Install for 2.7.11.  All failed in the exact same way as mentioned in my previous comment.

I rolled back my simulator to RivieraPro 2014.06 and everything behaves as expected.  

@chiggs What version of RivieraPro are you using?  Would it be possible for you to try 2015.06 to see if you see the same thing?  I am mostly curious if a later version will solve this problem for me or if it just our enviornment.

closed 2016-03-04 by leftink

stuarthodgson, 2016-03-04:
@leftink , I have 15.06 and 15.10 both with python 2.7.6 on linux mint 17.3 , and noth TOPLEVEL_LANG=verilog and vhdl work as expected.

leftink, 2016-03-04:
@stuarthodgson Thanks for checking, it must be some oddity.  We are on a really old version of RedHat so hopefully when we upgrade the issue will go away.