cocotb/cocotb #405: Clock generator is not accurate due to simulator time resolution

chevillotn, 2016-03-09:
In modelsim (and potentially others) time resolution is limited to ns or ps.
In my case i want to generate a 5.12GHz clock which has a period of 195.3125ps.
The result is a clock of half period 97ps which is not what i expected.

Suggestion is to adjust the half period each time before the yield:
class Clock(BaseClock):
...
    @cocotb.coroutine
    def start(self, cycles=0):
        value = 1
        time_count_real = 0
        time_count_int = 0
        half_period_int = int(self.half_period)
        half_period_int_adjusted = half_period_int
        while True:
            self.signal <= value
            value = 1 - value
            t = Timer(half_period_int_adjusted)
            yield t
            del(t)
            delta_time = time_count_real+self.half_period-(time_count_int+half_period_int)
            half_period_int_adjusted = half_period_int + int(delta_time)
            time_count_real += self.half_period
            time_count_int += half_period_int_adjusted

A delta is computed between the expected time value and the one reach with truncated value.
This delta is added to the next truncated timer.
The generator clock is very much close to expected frequency.

LinuxFanatic777, 2016-06-06:
Yes, this is well known approach to do this :)
Also about this: http://testbench.in/TB_08_CLOCK_GENERATOR.html

stuarthodgson, 2016-06-06:
@chevillotn I'll try and incorporate this if you don't have a handy pull request around.

ktbarrett, 2025-08-26:
`Clock` will now try to divide the clock period in two for the typical 50% duty cycle and throw a `ValueError` if it can't. The user can pass the duty period explicitly to avoid this rather than making an assumption about rounding.

closed 2025-08-26 by ktbarrett