cocotb/cocotb #407: Support for 'bit' type?

nevsan, 2016-03-16:
After receiving the following,
```
# 0.00ns DEBUG cocotb.gpi  VpiImpl.cpp:186  in create_gpi_obj_from_handle VPI: Not able to map type vpiBitVar(620) to object.
```
I did some searching and found out that the VPI code doesn't support the 'bit' SystemVerilog type. It can't, currently, because the included vpi_user.h does not define it, and is not up to date with [sv_vpi_user.h](http://www.eda.org/sv-cc/hm/att-2845/sv_vpi_user_P1800-draft5.h).

Are there any far reaching problems with updating that header and getting this fixed?

leftink, 2016-03-21:
Related to #397. 

My next set of changes will bring in the update for #397 as well as some update for vpi which "may" resolve this issue.

stuarthodgson, 2016-03-23:
@nevsan Updating the header (probably from work on #397) and then changing vpi layer to support this type is not a hard exercise. Testing it is always more the hard part as each simulator likes to though up different types entities in the same code.

nevsan, 2016-03-23:
Sounds like I can just take a look when #397 gets resolved. Is that one actively being worked on?

leftink, 2016-03-24:
@nevsan Yes it is completed as part of another issue I am working.  I will be doing a pull request tomorrow for the changes.

closed 2016-04-02 by nevsan