cocotb/cocotb #409: Test case issue_142 fails under modelsim

elgorwi, 2016-03-19:
Support was added to run the cocotb test cases using modelsim under windows in #408.
I noticed test case issue_142 fails when running modelsim 10.4c under windows7. 

Here's the transcript when TOPLEVEL_LANG=verilog:

```shell
#      0.00ns INFO     cocotb                                      __init__.py:131  in _initialise_testbench           Seeding Python random module with 1458358386
#      0.00ns INFO     cocotb.regression                         regression.py:161  in initialise                      Found test issue_142.issue_142_overflow_error
#      0.00ns INFO     cocotb.regression                         regression.py:262  in execute                         Running test 1/1: issue_142_overflow_error
#      0.00ns INFO     ..ssue_142_overflow_error.0xc66a890       decorators.py:189  in send                            Starting test: "issue_142_overflow_error"
#                                                                                                                      Description: Tranparently convert ints too long to pass
#                                                                                                                             through the GPI interface natively into BinaryValues
#      0.00ns WARNING  ..ssue_142_overflow_error.0xc66a890       decorators.py:197  in send                            Expecting 0x0 but got 0xff020d1000000000 on stream_in_data_wide @0xc3d8fa8
#      0.00ns ERROR    cocotb.regression                         regression.py:246  in handle_result                   Test Failed: issue_142_overflow_error (result was TestFailure)
#      0.00ns ERROR    cocotb.regression                         regression.py:167  in tear_down                       Failed 1 out of 1 tests (0 skipped)
#      0.00ns INFO     cocotb.regression                         regression.py:176  in tear_down                       Shutting down...
```

When TOPLEVEL_LANG=vhdl, the test segfaults:

```shell
#      0.00ns INFO     cocotb                                      __init__.py:131  in _initialise_testbench           Seeding Python random module with 1458358753
#      0.00ns INFO     cocotb.regression                         regression.py:161  in initialise                      Found test issue_142.issue_142_overflow_error
#      0.00ns INFO     cocotb.regression                         regression.py:262  in execute                         Running test 1/1: issue_142_overflow_error
#      0.00ns INFO     ..ssue_142_overflow_error.0xc7fb7f0       decorators.py:189  in send                            Starting test: "issue_142_overflow_error"
#                                                                                                                      Description: Tranparently convert ints too long to pass
#                                                                                                                             through the GPI interface natively into BinaryValues
# ** Fatal: (SIGSEGV) Bad handle or reference.
#    Time: 0 ns  Iteration: 2  Foreign Process: /sample_module File: Foreign
# Fatal error at an unknown location
#
#      0.00ns ERROR    cocotb.scheduler                            __init__.py:172  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely
#      0.00ns ERROR    cocotb.regression                         regression.py:239  in handle_result                   Test error has lead to simulator shuttting us down
#      0.00ns ERROR    cocotb.regression                         regression.py:167  in tear_down                       Failed 1 out of 1 tests (0 skipped)
#      0.00ns INFO     cocotb.regression                         regression.py:176  in tear_down                       Shutting down...
```

Has anyone tried running the cocotb tests using modelsim under linux?

leftink, 2016-03-21:
I have not run it under modelsim, but it works fine with QuestaSim on Linux.

leftink, 2016-03-21:
FYI, the proposed solution in #412, may make this issue go away.

elgorwi, 2016-03-30:
Hi @leftink,

Tried running the test again with latest master. When TOPLEVEL_LANG=vhdl, the test passes (no more segfault) However, there's still and error when TOPLEVEL_LANG=verilog:
```shell
#   0.00ns INFO     cocotb                                      __init__.py:131  in _initialise_testbench           Seeding Python random module with 1459301731
#   0.00ns INFO     cocotb.regression                         regression.py:161  in initialise                      Found test issue_142.issue_142_overflow_error
#   0.00ns INFO     cocotb.regression                         regression.py:262  in execute                         Running test 1/1: issue_142_overflow_error
#   0.00ns INFO     ..ssue_142_overflow_error.0x62bda70       decorators.py:189  in send                            Starting test: "issue_142_overflow_error"
#                                                                                                                   Description: Tranparently convert ints too long to pass
#                                                                                                                          through the GPI interface natively into BinaryValues
#   0.00ns WARNING  ..ssue_142_overflow_error.0x62bda70       decorators.py:197  in send                            Expecting 0x0 but got 0xff01581000000000 on 18375064780391055360
#   0.00ns ERROR    cocotb.regression                         regression.py:246  in handle_result                   Test Failed: issue_142_overflow_error (result was TestFailure)
#   0.00ns ERROR    cocotb.regression                         regression.py:167  in tear_down                       Failed 1 out of 1 tests (0 skipped)
#   0.00ns INFO     cocotb.regression                         regression.py:176  in tear_down                       Shutting down...
```

leftink, 2016-03-30:
@elgorwi 
Have you tried looking at this in the GUI to see what the waveforms show?

Also it would be worth running with COCOTB_LOG_LEVEL=DEBUG

elgorwi, 2016-03-30:
@leftink, thanks for the suggestions. 
Just investigated a bit further and it looks like the VPI interface drives some bits of 'stream_in_data_wide' to 'x' and 'z' rather than '0', when value  = 0.

When I change the test from:
```python
    for value in [0, 0x7FFFFFFF, 0x7FFFFFFFFFFF, BinaryValue(0x7FFFFFFFFFFFFF,len(dut.stream_in_data_wide),bigEndian=False)]:

        dut.stream_in_data_wide <= value
        yield RisingEdge(dut.clk)
        _compare(value)
        dut.stream_in_data_wide = value
        yield RisingEdge(dut.clk)
        _compare(value)
```
to
```python
    for value in [BinaryValue(0,len(dut.stream_in_data_wide),bigEndian=False), 0x7FFFFFFF, 0x7FFFFFFFFFFF, BinaryValue(0x7FFFFFFFFFFFFF,len(dut.stream_in_data_wide),bigEndian=False)]:

        dut.stream_in_data_wide <= value
        yield RisingEdge(dut.clk)
        _compare(value)
        dut.stream_in_data_wide = value
        yield RisingEdge(dut.clk)
        _compare(value)
```
All bits of 'stream_in_data_wide' are correctly driven and the test passes without errors.
The FLI interface behaves differently, ie. all bits are driven correctly even when value=0, rather than BinaryValue(0,len(dut.stream_in_data_wide),bigEndian=False).
Have the recent changes to FLI in  #412 been ported to VPI as well?

Probably not related to the error, but I'm not sure why '=' is used rather than '<=' for the second assignment of value. Is this a typo?

leftink, 2016-03-31:
@elgorwi The VPI interface is handled then the FLI.  The difference between the different methods is setting the value to 0  is done by using the set_value(long) method and the others are done using a BinaryValues and providing a string to the VPI interface.

The '=' versus the '<=' just exercise different interfaces in handle.py.  Both are valid ways of setting the value.

leftink, 2016-03-31:
@stuarthodgson and @chiggs 
Based on what @elgorwi  has demonstrated, setting the value using a binary string works just fine, but setting a value using an integer doesn't seem to be setting all the bits, i.e. in this case to 0.  If I had to speculate as to why the works for me and not for @elgorwi is that Linux may be initializing all memory to 0 while Windows is not.

It seems to me that using vpiIntVal will only set 32-bits of the value and  may require converting the int to a string of the length of the signal being set.  Do you guys have an opinion on this?

leftink, 2016-03-31:
FYI, I tried modifying the test to see if I could reproduce the problem by moving the setting by an integer to the value of 0 to be the last one in the list to test instead of the first.  This would cause the upper bits to be set by the previous items in the list and then I was hoping to see that when set to 0 the upper bits were still set.

For Questa on Linux this did not fail which means Questa set all the bits appropriately.  @elgorwi are you able to try Questa on your machine to see if it the problem exists?  This may just be a bug in the Mentor VPI interface that has been resolved in Questa.

elgorwi, 2016-04-01:
@leftink,

Unfortunately, I don't have access to questa. But I'll try installing the 64bit version of modelsim to see if that makes any difference.

cfswanson, 2016-04-03:
elgorwi,
I ran this test and it passed.  I am running Virtualbox on Windows 7.  In Virtualbox I am running Ubuntu 14.04 with Modelsim DE 10.4c and Xilinx Vivado 2015.2.

Craig

cfswanson, 2016-04-03:
elgorwi,
Here are my results:
# Loading sv_std.std
# Loading work.sample_module_v_unit
# Loading work.sample_module
# Loading /home/craig/cocotb/build/libs/i686/libvpi.so
#      -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:91   in gpi_print_registered_impl       VPI registered
#      0.00ns INFO     cocotb.gpi                                  gpi_embed.c:248  in embed_sim_init                  Running on ModelSim DE version 10.4c 2015.07
#      0.00ns INFO     cocotb.gpi                                  gpi_embed.c:249  in embed_sim_init                  Python interpreter initialised and cocotb loaded!
#      0.00ns INFO     cocotb                                      __init__.py:115  in _initialise_testbench           Running tests with Cocotb v1.0 from /home/craig/cocotb
#      0.00ns INFO     cocotb                                      __init__.py:131  in _initialise_testbench           Seeding Python random module with 1459644188
#      0.00ns INFO     cocotb.regression                         regression.py:161  in initialise                      Found test issue_142.issue_142_overflow_error
#      0.00ns INFO     cocotb.regression                         regression.py:262  in execute                         Running test 1/1: issue_142_overflow_error
#      0.00ns INFO     ..sue_142_overflow_error.0x5777a7ec       decorators.py:189  in send                            Starting test: "issue_142_overflow_error"
#                                                                                                                      Description: Tranparently convert ints too long to pass
#                                                                                                                             through the GPI interface natively into BinaryValues
#     17.50ns INFO     cocotb.regression                         regression.py:209  in handle_result                   Test Passed: issue_142_overflow_error
#     17.50ns INFO     cocotb.regression                         regression.py:170  in tear_down                       Passed 1 tests (0 skipped)
#     17.50ns INFO     cocotb.regression                         regression.py:176  in tear_down                       Shutting down...
# ** Note: $finish    : UNKNOWN(-1)
#    Time: 17501 ps  Iteration: 0  Instance: /sample_module
# End time: 20:43:08 on Apr 02,2016, Elapsed time: 0:00:02
# Errors: 0, Warnings: 0
make[1]: Leaving directory `/home/craig/cocotb/tests/test_cases/issue_142'

cfswanson, 2016-04-03:
elgorwi,
I have no idea why it made the above message so HUGE.  I just copied and pasted.
Craig

elgorwi, 2016-04-04:
Hi @cfswanson,

Thanks for testing, the result is interesting. Did you modify the test to use 'BinaryValue(0,len(dut.stream_in_data_wide),bigEndian=False)' rather than '0' as outlined in my post earlier? Or was this an unmodified test? Which revision of cocotb did you run (git SHA)?

I installed a 64-bit modelsim version. However, I was only able to get the VPI interface up and running, the FLI interface gave some compile errors. Interestingly, the test passed without issues.

Note, the comments are interpreted as markdown. Because cocotb shell output starts with a hash everything is converted to headings. You can get around this by putting everything between 3 backticks as explained here:
https://help.github.com/articles/creating-and-highlighting-code-blocks/

cfswanson, 2016-04-04:
elgorwi,
Sorry but I ran an unmodifed test. It was version 1.0.  Do you want me to
modify the test next?
Craig

elgorwi, 2016-04-04:
Hi @cfswanson,

I think there might be some confusion as a test case for issue_142 was not yet included in cocotb v1.0. I'd like to know the git SHA of the revision that you're using. Could you cd to your cocotb folder and let me know what following command returns:
```shell
git show
```   
No need to modify the tests.
Thanks

TC01, 2017-11-03:
I'm not sure if it's failing for the same reason, but the test case for #142 is also failing on ius/cadence with the following output:

```
  0.00ns INFO     cocotb.gpi                                  gpi_embed.c:248  in embed_sim_init                  Running on ncsim(64) version 15.20-s028
  0.00ns INFO     cocotb.gpi                                  gpi_embed.c:249  in embed_sim_init                  Python interpreter initialised and cocotb loaded!
  0.00ns INFO     cocotb                                      __init__.py:115  in _initialise_testbench           Running tests with Cocotb v1.0 from /tape/mitch_sim/bjr/git/cocotb
  0.00ns INFO     cocotb                                      __init__.py:131  in _initialise_testbench           Seeding Python random module with 1509723129
  0.00ns INFO     cocotb.regression                         regression.py:167  in initialise                      Found test issue_142.issue_142_overflow_error
  0.00ns INFO     cocotb.regression                         regression.py:287  in execute                         Running test 1/1: issue_142_overflow_error
  0.00ns INFO     ..ssue_142_overflow_error.0x50ad550       decorators.py:191  in send                            Starting test: "issue_142_overflow_error"
                                                                                                                  Description: Tranparently convert ints too long to pass
                                                                                                                         through the GPI interface natively into BinaryValues
  0.00ns ERROR    ..ssue_142_overflow_error.0x50ad550           result.py:51   in raise_error                     Send raised exception: Unable to resolve to binary >zzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzz<
                                                                                                                    File "/tape/mitch_sim/bjr/git/cocotb/cocotb/decorators.py", line 197, in send
                                                                                                                      return self._coro.send(value)
                                                                                                                    File "/tape/mitch_sim/bjr/git/cocotb/tests/test_cases/issue_142/issue_142.py", line 27, in issue_142_overflow_error
                                                                                                                      _compare(value)
                                                                                                                    File "/tape/mitch_sim/bjr/git/cocotb/tests/test_cases/issue_142/issue_142.py", line 17, in _compare
                                                                                                                      if int(dut.stream_in_data_wide.value) != int(value):
                                                                                                                    File "/tape/mitch_sim/bjr/git/cocotb/cocotb/binary.py", line 389, in __int__
                                                                                                                      return self.integer
                                                                                                                    File "/tape/mitch_sim/bjr/git/cocotb/cocotb/binary.py", line 247, in get_value
                                                                                                                      return self._convert_from[self.binaryRepresentation](self._str)
                                                                                                                    File "/tape/mitch_sim/bjr/git/cocotb/cocotb/binary.py", line 154, in _convert_from_unsigned
                                                                                                                      return int(resolve(x), 2)
                                                                                                                    File "/tape/mitch_sim/bjr/git/cocotb/cocotb/binary.py", line 41, in resolve
                                                                                                                      raise ValueError("Unable to resolve to binary >%s<" % string)

  0.00ns ERROR    cocotb.regression                         regression.py:263  in handle_result                   Test Failed: issue_142_overflow_error (result was TestError)
  0.00ns ERROR    cocotb.regression                         regression.py:173  in tear_down                       Failed 1 out of 1 tests (0 skipped)
  0.00ns INFO     cocotb.regression                         regression.py:341  in _log_test_summary               ********************************************************************************************
                                                                                                                  ** TEST                                PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
                                                                                                                  ********************************************************************************************
[... cut ...]
                                                                                                                  ********************************************************************************************
```

I ran this on my laptop with icarus and I got the same output, but the test passed. I see that this is because of the following code in the test case:

```
@cocotb.test(expect_error=cocotb.SIM_NAME.lower().startswith(("icarus")))
```

I'm guessing then that it's known it fails on icarus, but perhaps not known it's failing on ius? Should this code be updated to expect an error on ius as well? Should I file a separate bug about this failure?

closed 2018-06-19 by elgorwi