cocotb/cocotb #413: Update Regression Testing for both VHPI and VPI

leftink, 2016-03-21:
Currently, when code is pushed, the regression suite runs test for python2 and python3 using the "default" languages for each test.

In one of my previous updates, I cleaned up the make files so that the regression suite can be run with TOPLEVEL_LANG set to either VHDL or Verilog.  Tests will be skipped if not appropriate for the TOPLEVEL_LANG.

It would probably be worth updating the "code check" to test with TOPLEVEL_LANG set to both VHDL and Verilog.  It would have caught issue #412 if testing occurred that way.

stuarthodgson, 2016-03-23:
@leftink . I am in agreement that the code check should run TOPLEVEL_LANG=vhdl, however we are not able to run tests using vhdl in the public domain at the moment due to license issues. We do run internally not using Travis and we should look at exposing the result of this to the outside world if we can.

ktbarrett, 2020-04-23:
Currently tox supports passing `TOPLEVEL_LANG` to tests, and there is a test using VHDL as the top level lang with GHDL (it doesn't work, but that's another issue).

closed 2020-04-23 by ktbarrett