cocotb/cocotb #415: cocotb doesn't work on accumulation in VHDL

sufengniu, 2016-03-24:
Hi,

I am new to cocotb and find it is pretty neat, but I got some incorrect results in cocotb. I am not sure whether this is the correct place to ask question since I didn't find any forums. I found that for a simple accumulator (something like sum = sum+input) in VHDL using fixed_pkg, the cocotb failed to get the correct output. I used a vhdl testbench and test in the modelsim, which show it is fully functional, but in cocotb, the sum value keeps 0 all the time, which doesn't do accumulation at all. my simulator is modelsim, I was wondering is this a bug or I did something wrong?

Thank you

Sufeng

leftink, 2016-03-24:
@sufengniu could you provide some more information.  What is your Python test doing?  Code snippets of bothe the vhdl and Python would be useful.

sufengniu, 2016-03-24:
Sure, I did it by yield rising edge of the clock to load, following is my a test vhdl code:

`library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
use IEEE.fixed_float_types.all;
use IEEE.fixed_pkg.all;
use IEEE.float_pkg.all;

entity adder is
    generic
    (
    	DATA_INT		: integer := 2; -- integer part
    	DATA_FRA		: integer := 14; -- fractional part
	DATA_WIDTH		: integer := 16
    );
    port
    (
	clk				: in std_logic;
        rst				: in std_logic;
        ce				: in std_logic;
        data1_in		: in std_logic_vector(DATA_WIDTH-1 downto 0); 
        data_out		: out std_logic_vector(DATA_WIDTH-1 downto 0)
    );
end adder;

architecture rtl of adder is

signal data_a_buf: sfixed(DATA_INT-1 downto -DATA_FRA);
signal data_sum: sfixed(DATA_INT-1 downto -DATA_FRA);

begin 

	process(clk, rst)
	begin
		if rst = '1' then
			data_a_buf <= (others => '0');
		elsif rising_edge(clk) then
			if ce = '1' then
				data_a_buf <= to_sfixed(data1_in, data_a_buf'high, data_a_buf'low);	
				data_sum <= resize(data_sum+data_a_buf, DATA_INT-1, -DATA_FRA);
				data_out <= to_slv(data_sum);
			end if;
		end if;
	end process;
end rtl;
`

Here is my test_adder.py code
`
import random
import logging

import cocotb
from cocotb.clock import Clock
from cocotb.triggers import Timer, RisingEdge, ReadOnly
from cocotb.result import TestFailure, ReturnValue
clock_period = 100

@cocotb.coroutine
def clock_gen(signal, period=100):
	while True:
		signal <= 0
		yield Timer(period/2)
		signal <= 1
		yield Timer(period/2)

@cocotb.test()
def add_basic_test(dut):
	dut.log.info("Running basic test!")

	data_width = int(dut.DATA_WIDTH.value)
	int_width = int(dut.DATA_INT.value)
	fraction_width = int(dut.DATA_FRA.value)
	dut.log.info('Detected DATA_WIDTH = %d, DATA_INT = %d, DATA_FRA = %d' %(data_width, int_width, fraction_width))	
	cocotb.fork(clock_gen(dut.clk, period=clock_period))
	dut.rst <= 1
	yield RisingEdge(dut.clk)

	yield RisingEdge(dut.clk)
	dut.rst <= 0
	dut.ce <= 1
	yield RisingEdge(dut.clk)
	dut.data1_in = 5
	yield RisingEdge(dut.clk)
	dut.data1_in = 10
	print "===================="

	yield RisingEdge(dut.clk)
	print "--------------", int(dut.data_out.value)
	yield RisingEdge(dut.clk)	
	print "--------------", int(dut.data_out.value)
	yield RisingEdge(dut.clk)
	print "--------------", int(dut.data_out.value)

	dut.data1_in = 7
	yield RisingEdge(dut.clk)
	dut.data1_in = 5
	print "--------------", int(dut.data_out.value)
	yield RisingEdge(dut.clk)
	print "--------------", int(dut.data_out.value)
	yield RisingEdge(dut.clk)
	dut.log.info("sum = %d", dut.data_out.value)
	yield RisingEdge(dut.clk)
	dut.log.info('sum = %d', dut.data_out.value)
`

after Make (SIM is modelsim). The result is:

> #      0.00ns INFO     ..ine.add_basic_test.0x7fffe65555d0       decorators.py:189  in send                            Starting test: "add_basic_test"
#                                                                                                                      Description: None
#      0.00ns INFO     cocotb.adder                              test_adder.py:26   in add_basic_test                  Running basic test!
#      0.00ns INFO     cocotb.adder                              test_adder.py:31   in add_basic_test                  Detected DATA_WIDTH = 16, DATA_INT = 2, DATA_FRA = 14
# ====================
# -------------- 0
# -------------- 0
# -------------- 0
# -------------- 0
# -------------- 0
#      0.95ns INFO     cocotb.adder                              test_adder.py:61   in add_basic_test                  sum = 0
#      1.05ns INFO     cocotb.adder                              test_adder.py:63   in add_basic_test                  sum = 0

leftink, 2016-03-24:
The only thing that really jumps out at me is in the VHDL.  In reset, you setting data_a_buf to 0, but  you are not also clearing the data_sum signal.   This means you are accumulating with "U"'s initially and I am not sure how well that will work.

Printing the int() will convert "X" and U's to 0 I believe.  Try printing the following `dut.log.info("sum=%d (%s)", int(dut.data_out), dut.data_out.value.binstr)`

Also, if you are using "make" then you can set GUI=1 WAVES=1 to launch the gui to make sure things are being set as you expect.

Lastly, this isn't your problem, but you don't need to write your own clock coroutine. You can instead just do:
```python
from cocotb.clock import Clock

@cocotb.test()
def add_basic_test(dut):
    cocotb.fork(Clock(dut.clk, clock_period).start())
```

chiggs, 2016-03-24:
> you don't need to write your own clock coroutine

In fact some of our examples are misdirecting people here and defining a clock generator instead of re-using `cocotb.clock.Clock`.

sufengniu, 2016-03-24:
@leftink  @chiggs  Thank you very much for your answers! actually originally in my code, I have reset them. but as you mentioned, their binary is actually "X". I found that I forget to initialize the input as 0 so in the next clock, the buff become X. Add the GUI is really awesome! Thank you, it is my problem and sorry about that.

sufengniu, 2016-03-24:
I will close this issue, thank you

closed 2016-03-24 by sufengniu