cocotb/cocotb #417: __str__() and __repr__() usage

leftink, 2016-03-24:
@stuarthodgson and @chiggs 
In my recent work, I noticed something in handle.py that I wasn't a huge fan of and was looking to understand why it was implemented the way it was.

The current implementation uses __str__() to return the handle name and pointer value.  Where this gets confusing is for the StringObject, the user has to use repr(dut.hdl) to get the actual string value when it seems like str(dut.hdl) would make more sense.

Also, for ModifiableObject, it would be nice to use str(dut.hdl) to get the binary string.

chiggs, 2016-03-24:
@leftink I agree, it's slightly counter-intuitive.  I think historically it's probably a remnant of when we were first bringing up the GPI interface and shoved this in to help debug.  We should definitely harmonise and follow standard Python conventions for `__str__` and `__repr__`.  I'm happy for us to change this since the current usage is anti-idiomatic.

For `__str__` we could just coerce to the most appropriate value, so binstr for arrays of bits, the full path for hierarchy objects, or str(value) for ints/floats etc.

For `__repr__` we should follow the standard Python convention of providing the class name and the path perhaps?  e.g. `ModifiableObject(this.that.something[2])`

leftink, 2016-04-01:
Tested and works in the latest version.

closed 2016-04-01 by leftink