cocotb/cocotb #419: gpi_embed fails to dynamically load Python Library

leftink, 2016-03-29:
While doing some testing with Anaconda3 v2.1.0 which runs python 3.4.1, I was seeing an error: "Failed to find python lib"

And a simulator error: "Unable to open lib libpython3.4.so (libpython3.4.so: cannot open shared object file: No such file or directory)"

I looked and sure enough there wasn't a library by that name, but there was one called libpython3.4m.so.

I dug into that issue a little bit more and that is a result of PEP3149 compliance with ABI Version Tagging support.

Rather than manually create PYTHON_DYN_LIB in the makefile, we should probably just be getting it from the sysconfig just like we do the version, library directory and the dynamic library directory.

The following seems to work for both Python2 and Python3:
`PYTHON_DY_LIB:=$(shell $PYTHON_BIN) -c 'from __future__ import print_function; from distutils import sysconfig; print(sysconfig.get_config_var("LDLIBRARY"))'`

@chiggs and @stuarthodgson 
While the above should fix this issue, I noticed the simulation ran just fine even with the error.  Given that we are linking to the library when we build gpi_embed, do we really need to be dynamically loading the python library?

stuarthodgson, 2016-03-30:
@leftink the secondary loading is a result of #146 . And is done as simulators like to ship the kitchen sink with them rather than use system libs. We encountered an issue where the python loaded at run time was not the one used at link time. By compiling the version needed into the library and dlopening it we replace the lib with the correct one.

leftink, 2016-03-30:
@stuarthodgson Sounds good.  I figured it was something like that.  The fix I have provided doesn't remove that, but only updates Makefiles to query the correct library name to load.

closed 2016-03-30 by stuarthodgson