cocotb/cocotb #421: Endian swapper errors using FLI

elgorwi, 2016-03-30:
Just tried running the endian swapper example again using the latest from master and getting some errors when using FLI. The example used to work before #416 was merged into master. I'm running modelsim 10.4c under windows7. 
Note the test is also marked as passing, where it should be marked as failing. (See issue #368)

```shell
#   0.00ns INFO     cocotb.regression                         regression.py:262  in execute                         Running test 1/33: run_test_001
#   0.00ns INFO     cocotb.coroutine._my_test.0xe6360d0       decorators.py:189  in send                            Starting test: "run_test_001"
#                                                                                                                   Description: Automatically generated test
#
#                                                                                                                       backpressure_inserter: None
#                                                                                                                       config_coroutine: None
#                                                                                                                       data_in: random_packet_sizes (random string data of a random length)
#                                                                                                                       idle_inserter: None
#
# 5000.00ns INFO     ..tb.scoreboard.endian_swapper_vhdl       scoreboard.py:177  in add_interface                   Created with reorder_depth 0
# 5000.00ns ERROR    cocotb.gpi                                FliObjHdl.cpp:337  in set_signal_value                FLI: Unable to set logic vector due to the string having incorrect length.  Length of 64 needs to be 3
# 165000.00ns ERROR    cocotb.gpi                                FliObjHdl.cpp:337  in set_signal_value                FLI: Unable to set logic vector due to the string having incorrect length.  Length of 64 needs to be 3
# 165000.00ns INFO     ..tb.endian_swapper_vhdl.stream_out           avalon.py:137  in _monitor_recv                   Received a packet of 108 bytes
# 305000.00ns ERROR    cocotb.gpi                                FliObjHdl.cpp:337  in set_signal_value                FLI: Unable to set logic vector due to the string having incorrect length.  Length of 64 needs to be 3
# 305000.00ns INFO     ..tb.endian_swapper_vhdl.stream_out           avalon.py:137  in _monitor_recv                   Received a packet of 98 bytes
# 405000.00ns ERROR    cocotb.gpi                                FliObjHdl.cpp:337  in set_signal_value                FLI: Unable to set logic vector due to the string having incorrect length.  Length of 64 needs to be 3
# 405000.00ns INFO     ..tb.endian_swapper_vhdl.stream_out           avalon.py:137  in _monitor_recv                   Received a packet of 65 bytes
# 495000.00ns ERROR    cocotb.gpi                                FliObjHdl.cpp:337  in set_signal_value                FLI: Unable to set logic vector due to the string having incorrect length.  Length of 64 needs to be 3
# 495000.00ns INFO     ..tb.endian_swapper_vhdl.stream_out           avalon.py:137  in _monitor_recv                   Received a packet of 57 bytes
# 525000.00ns ERROR    cocotb.gpi                                FliObjHdl.cpp:337  in set_signal_value                FLI: Unable to set logic vector due to the string having incorrect length.  Length of 64 needs to be 3
# 525000.00ns INFO     ..tb.endian_swapper_vhdl.stream_out           avalon.py:137  in _monitor_recv                   Received a packet of 14 bytes
# 585000.00ns ERROR    cocotb.gpi                                FliObjHdl.cpp:337  in set_signal_value                FLI: Unable to set logic vector due to the string having incorrect length.  Length of 64 needs to be 3
# 585000.00ns INFO     ..tb.endian_swapper_vhdl.stream_out           avalon.py:137  in _monitor_recv                   Received a packet of 40 bytes
# 675000.00ns ERROR    cocotb.gpi                                FliObjHdl.cpp:337  in set_signal_value                FLI: Unable to set logic vector due to the string having incorrect length.  Length of 64 needs to be 3
# 675000.00ns INFO     ..tb.endian_swapper_vhdl.stream_out           avalon.py:137  in _monitor_recv                   Received a packet of 57 bytes
# 865000.00ns ERROR    cocotb.gpi                                FliObjHdl.cpp:337  in set_signal_value                FLI: Unable to set logic vector due to the string having incorrect length.  Length of 64 needs to be 3
# 865000.00ns INFO     ..tb.endian_swapper_vhdl.stream_out           avalon.py:137  in _monitor_recv                   Received a packet of 138 bytes
# 985000.00ns ERROR    cocotb.gpi                                FliObjHdl.cpp:337  in set_signal_value                FLI: Unable to set logic vector due to the string having incorrect length.  Length of 64 needs to be 3
# 985000.00ns INFO     ..tb.endian_swapper_vhdl.stream_out           avalon.py:137  in _monitor_recv                   Received a packet of 83 bytes
# 1085000.00ns ERROR    cocotb.gpi                                FliObjHdl.cpp:337  in set_signal_value                FLI: Unable to set logic vector due to the string having incorrect length.  Length of 64 needs to be 3
# 1085000.00ns INFO     ..tb.endian_swapper_vhdl.stream_out           avalon.py:137  in _monitor_recv                   Received a packet of 65 bytes
# 1125000.00ns INFO     cocotb.endian_swapper_vhdl         .._endian_swapper.py:158  in run_test                        DUT correctly counted 10 packets
# 1125001.00ns INFO     cocotb.regression                         regression.py:209  in handle_result                   Test Passed: run_test_001

```

leftink, 2016-03-30:
@elgorwi and @chiggs 

I looked at the code and the issue is with the Avalon Driver.  The driver is creating a BinaryValue with the length of stream_in_data, I.e 64 bits. 

That BinaryValue is then used to write start_of_packet and end_of_packet which are only one bit.  It is also used to write the empty signal which is 3 bits and causing the error to be printed.

leftink, 2016-03-30:
@elgorwi I have just submitted a fix that will get rid of errors.

elgorwi, 2016-07-12:
Thanks for the fix @leftink

closed 2016-07-12 by elgorwi