cocotb/cocotb #428: Cocotb on Windows

Paebbels, 2016-04-01:
Hello,

We are planning to integrate Cocotb into our [open-source IP-core library PoC](https://github.com/VLSI-EDA/PoC). Currently, we are supporting classic VHDL testbenches for various simulators and platforms. Cocotb states that it also supports Windows tool chains, but I encountered several problems....

First off all, MinGW32 can not be used because, Windows, QuestaSim and Python are 64-bit installations. Even compiling in 64-bit mode results in linking processes to i686 dlls from MinGW32...

So I installed MinGW64/MSys2 and added several packages. I modified some of Cocotb's makefiles and got it to compile all required dlls. I think I can provide a PR for it soon.

Finally I git stuck at this dll loading error in QuestaSim:

    Paebbels@ite190 MINGW64 /d/git/PoC/lib/cocotb/examples/dff/tests
    $ make SIM=questa TOPLEVEL_LANG=verilog
    make results.xml
    make[1]: Verzeichnis „/d/git/PoC/lib/cocotb/examples/dff/tests“ wird betreten
    cd sim_build && LD_LIBRARY_PATH=/d/git/PoC/lib/cocotb/build/libs/x86_64::/c/msys64/usr/lib/python2.7/config:/c/msys64/usr/lib/python2.7/config:/c/msys64/usr/lib/python2.7/config:/c/msys64/usr/lib/python2.7/config && PATH=/usr/bin:/mingw64/bin:/usr/local/bin:/usr/bin:/bin:/c/Windows/System32:/c/Windows:/c/Windows/System32/Wbem:/c/Windows/System32/WindowsPowerShell/v1.0/:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/c/Mentor/QuestaSim64/10.4c/:/c/Mentor/QuestaSim64/10.4c/win64 MODULE=dff_cocotb TESTCASE= TOPLEVEL=dff \
    GPI_EXTRA= TOPLEVEL_LANG=verilog PYTHONPATH=/d/git/PoC/lib/cocotb/build/libs/x86_64:/d/git/PoC/lib/cocotb:/d/git/PoC/lib/cocotb/examples/dff/tests: \
    vsim -c -do runsim.do 2>&1 | tee sim.log
    Reading C:/Mentor/QuestaSim64/10.4c/tcl/vsim/pref.tcl
    
    # 10.4c
    
    # do runsim.do
    # ** Warning: (vdel-134) Unable to remove locked optimized design "_opt".  Locker is Paebbels@ITE190.
    #
    # ** Warning: (vlib-34) Library already exists at "work".
    #
    # QuestaSim-64 vlog 10.4c Compiler 2015.07 Jul 20 2015
    # Start time: 19:00:27 on Apr 01,2016
    # vlog -work work "+define+COCOTB_SIM" -timescale 1ns/100ps -mfcu "+acc=rmb" -sv ../../hdl/dff.v
    # -- Compiling module dff
    #
    # Top level modules:
    #       dff
    # End time: 19:00:27 on Apr 01,2016, Elapsed time: 0:00:00
    # Errors: 0, Warnings: 0
    # vsim -t 1ps -onfinish exit -pli "libvpi.dll" work.dff
    # Start time: 19:00:27 on Apr 01,2016
    # ** Note: (vsim-3812) Design is being optimized...
    #
    # ** Note: (vsim-3865) Due to PLI being present, full design access is being specified.
    #
    # //  Questa Sim-64
    # //  Version 10.4c win64 Jul 20 2015
    # //
    # //  Copyright 1991-2015 Mentor Graphics Corporation
    # //  All Rights Reserved.
    # //
    # //  THIS WORK CONTAINS TRADE SECRET AND PROPRIETARY INFORMATION
    # //  WHICH IS THE PROPERTY OF MENTOR GRAPHICS CORPORATION OR ITS
    # //  LICENSORS AND IS SUBJECT TO LICENSE TERMS.
    # //  THIS DOCUMENT CONTAINS TRADE SECRETS AND COMMERCIAL OR FINANCIAL
    # //  INFORMATION THAT ARE PRIVILEGED, CONFIDENTIAL, AND EXEMPT FROM
    # //  DISCLOSURE UNDER THE FREEDOM OF INFORMATION ACT, 5 U.S.C. SECTION 552.
    # //  FURTHERMORE, THIS INFORMATION IS PROHIBITED FROM DISCLOSURE UNDER
    # //  THE TRADE SECRETS ACT, 18 U.S.C. SECTION 1905.
    # //
    # Loading sv_std.std
    # Loading work.dff(fast)
    # Loading D:\git\PoC\lib\cocotb\build\libs\x86_64/libvpi.dll
    # ** Error: (vsim-3193) Load of "D:\git\PoC\lib\cocotb\build\libs\x86_64/libvpi.dll" failed: DLL dependent library not found.
    #
    # ** Error (suppressible): (vsim-PLI-3002) Failed to load PLI object file "D:\git\PoC\lib\cocotb\build\libs\x86_64/libvpi.dll".
    #    Time: 0 ps  Iteration: 0  Root: /
    # Error loading design
    Error loading design
    # End time: 19:00:27 on Apr 01,2016, Elapsed time: 0:00:00
    # Errors: 2, Warnings: 0
    make[1]: Verzeichnis „/d/git/PoC/lib/cocotb/examples/dff/tests“ wird verlassen

The error message in short:

> # Loading D:\git\PoC\lib\cocotb\build\libs\x86_64/libvpi.dll
> # ** Error: (vsim-3193) Load of "D:\git\PoC\lib\cocotb\build\libs\x86_64/libvpi.dll" failed: DLL dependent library not found.

The files libvpi.dll was found, otherwise it states:

> # Loading libvpi.dll
> # ** Error: (vsim-3193) Load of "libvpi.dll" failed: File not found.
> #
> # ** Error (suppressible): (vsim-PLI-3002) Failed to load PLI object file "libvpi.dll".

If I run `ldd.exe` on that file, I get these resolved libraries:

    Paebbels@ite190 MINGW64 /d/git/PoC/lib/cocotb/examples/dff/tests
    $ ldd.exe "D:\git\PoC\lib\cocotb\build\libs\x86_64/libvpi.dll"
            ntdll.dll => /c/Windows/SYSTEM32/ntdll.dll (0x779f0000)
            kernel32.dll => /c/Windows/system32/kernel32.dll (0x777d0000)
            KERNELBASE.dll => /c/Windows/system32/KERNELBASE.dll (0x7fefd940000)
            mtipli.dll => /c/Mentor/QuestaSim64/10.4c/win64/mtipli.dll (0x180000000)
            MSVCR90.dll => /c/Windows/WinSxS/amd64_microsoft.vc90.crt_1fc8b3b9a1e18e3b_9.0.30729.6161_none_08e61857a83bc251/MSVCR90.dll (0x70630000)
            msys-2.0.dll => /usr/bin/msys-2.0.dll (0x180040000)
            msys-gcc_s-seh-1.dll => /usr/bin/msys-gcc_s-seh-1.dll (0x5e8160000)
            msys-stdc++-6.dll => /usr/bin/msys-stdc++-6.dll (0x526840000)
            libgpi.dll => /d/git/PoC/lib/cocotb/build/libs/x86_64/libgpi.dll (0x401130000)
            libcocotb.dll => /d/git/PoC/lib/cocotb/build/libs/x86_64/libcocotb.dll (0x43dcf0000)
            msys-python2.7.dll => /usr/bin/msys-python2.7.dll (0x46b270000)
            libcocotbutils.dll => /d/git/PoC/lib/cocotb/build/libs/x86_64/libcocotbutils.dll (0x5290a0000)
            libgpilog.dll => /d/git/PoC/lib/cocotb/build/libs/x86_64/libgpilog.dll (0x5a6e30000)

All dependencies are resolved. According to `env`, no `LD_LIBRARY_PATH` is set in MinGW64. 

**What could cause such an error?**

My 64-bit Windows system:

  - Windows 7 64-bit (German)
  - QuestaSim 64-bit 10.4c
  - Python 2.7.9 64-bit 
  - Python 3.5.1 64-bit
  - MinGW64 with gcc 5.3 toolchain in C:\msys64
  - GNAT 2015 with MinGW and gcc 4.9.3

elgorwi, 2016-04-04:
Hi @Paebbels,

Just installed modelsim 64-bit 10.4c on my windows 7 system, together with WinPython-64bit-2.7.9.5. (this version of winpython comes bundled with gcc version 4.9.2 x86_64-w64-mingw32).

After removing a redundant 32 bit python27.dll from my system, cocotb seems to run fine using VPI, although with some warnings during compilation:

```
simulatormodule.c: In function 'register_value_change_callback':
simulatormodule.c:417:15: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
     sig_hdl = (gpi_sim_hdl)PyLong_AsUnsignedLong(pSihHdl);
               ^
simulatormodule.c: In function 'deregister_callback':
simulatormodule.c:955:11: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
     hdl = (gpi_sim_hdl)PyLong_AsUnsignedLong(pSihHdl);
```

When trying to compile FLI, I'm getting some compile errors:
```
FliObjHdl.cpp: In member function 'virtual int FliLogicObjHdl::set_signal_value(long int)':
FliObjHdl.cpp:315:65: error: cast from 'char*' to 'long int' loses precision [-fpermissive]
             mti_SetVarValue(get_handle<mtiVariableIdT>(), (long)m_mti_buff);
                                                                 ^
FliObjHdl.cpp:317:66: error: cast from 'char*' to 'long int' loses precision [-fpermissive]
             mti_SetSignalValue(get_handle<mtiSignalIdT>(), (long)m_mti_buff);
                                                                  ^
FliObjHdl.cpp: In member function 'virtual int FliLogicObjHdl::set_signal_value(std::string&)':
FliObjHdl.cpp:353:65: error: cast from 'char*' to 'long int' loses precision [-fpermissive]
             mti_SetVarValue(get_handle<mtiVariableIdT>(), (long)m_mti_buff);
                                                                 ^
FliObjHdl.cpp:355:66: error: cast from 'char*' to 'long int' loses precision [-fpermissive]
             mti_SetSignalValue(get_handle<mtiSignalIdT>(), (long)m_mti_buff);
                                                                  ^
FliObjHdl.cpp: In member function 'virtual int FliRealObjHdl::set_signal_value(double)':
FliObjHdl.cpp:449:61: error: cast from 'double*' to 'long int' loses precision [-fpermissive]
         mti_SetVarValue(get_handle<mtiVariableIdT>(), (long)m_mti_buff);
                                                             ^
FliObjHdl.cpp:451:62: error: cast from 'double*' to 'long int' loses precision [-fpermissive]
         mti_SetSignalValue(get_handle<mtiSignalIdT>(), (long)m_mti_buff);
                                                              ^
FliObjHdl.cpp: In member function 'virtual int FliStringObjHdl::set_signal_value(std::string&)':
FliObjHdl.cpp:500:61: error: cast from 'char*' to 'long int' loses precision [-fpermissive]
         mti_SetVarValue(get_handle<mtiVariableIdT>(), (long)m_mti_buff);
                                                             ^
FliObjHdl.cpp:502:62: error: cast from 'char*' to 'long int' loses precision [-fpermissive]
         mti_SetSignalValue(get_handle<mtiSignalIdT>(), (long)m_mti_buff);
                                                              ^
```

Currently running cocotb from a MSYS2 shell, with gnu make 4.1 installed.

Following dirs had to be added to PATH:
```
WinPython-64bit-2.7.9.5\tools\mingw32\bin 
WinPython-64bit-2.7.9.5\python-2.7.9.amd64
WinPython-64bit-2.7.9.5\python-2.7.9.amd64\Scripts
WinPython-64bit-2.7.9.5\python-2.7.9.amd64\Lib
WinPython-64bit-2.7.9.5\python-2.7.9.amd64\DLLs
```
Also had to 'register' the python distribution on my system using the WinPython Control Panel.

It may be worth trying this setup, to see if that makes any difference.

Paebbels, 2016-04-04:
Sorry I think I forgot to mention, which Python I used....

I'm compiling against Python 2.7 from MinGW64, not the Python 2.7 from Windows. This Python was also compiled by GCC.

@elgorwi 
Your suggested WinPython was compiled with VisualStudio, I think it's not good to mix VisualStudio and GCC compiled projects....

Im collecting my changes for MinGW64/MSys2 on this branch: [paebbels/MinGW64](https://github.com/VLSI-EDA/cocotb/tree/paebbels/MinGW64) at our forked repo.

jmorarity, 2016-04-04:
@Paebbels, it's possible that you need not make any modifications to the base makefiles as I already made the modifications necessary to run 64bit Windows w/ Msys2 in #373.

That being said, my configuration was different than yours in that I'm running Verilog test benches w/ ModelSim DE x64.  I'm also using Anaconda Python 2.7, though I've had success with other Python distributions as well.

Let me go ahead and post some info I put on a wiki for my team about the build process. I hope this will be useful to others in the cocotb community.

#### Install MSYS2 w/ MinGW-w64
Install msys2-x86_64 (comes w/ mingw32 and mingw64) via the instructions at https://msys2.github.io/
`pacman --needed -Sy bash pacman pacman-mirrors msys2-runtime`
`pacman -Su`

#### Set Up MSYS2 Build Environment
Follow the "building packages" guide at http://sourceforge.net/p/msys2/wiki/Contributing%20to%20MSYS2/to acquire the requisite toolchains
`pacman -S base-devel`
`pacman -S mingw-w64-i686-toolchain`
`pacman -S mingw-w64-x86_64-toolchain`

#### Fix libpython27.a (optional depending on Python source)
If using Anaconda or PythonXY, or ??, you'll need to regenerate your libpython27.a file in the "proper" format (don't ask me why ... something to do w/ the symbols). Navigate to your python directory within msys2 and type the following:
 `gendef python27.dll`
Then the following for 64-bit:
 `dlltool --as-flags=--64 -m i386:x86-64 -k --output-lib libpython27.a --input-def python27.def` 
or instead for 32-bit:
 `dlltool -m i386 -k --output-lib libpython27.a --input-def python27.def`
Finally, move the newly generated lib into the appropriate subdirectory:
 `mv libpython27.a libs/`

#### Build Cocotb Source
Note that since Msys2 has it's own internal python distribution, you will need to set a Windows environmental variable PYTHONHOME to your alternate Python root dir (C:\Python27 for instance). You can also inline the definition, similar to how ARCH is done below.
To build for 64-bit:
 `make SIM=modelsim`
or for 32-bit:
 `ARCH=i686 make SIM=modelsim`

Paebbels, 2016-07-05:
Hello,

I'm continuing my Cocotb experiments on Windows and as far as I can tell, nothing really change ...
before I'm diving into the details, I can say that non of the following combination works:
- MinGW32 (GCC 5.4.0) + Cocotb intended for GHDL
- MinGW64 (GCC 5.4.0) + Cocotb intended for GHDL
- GCC 4.5.0 (mingw64) from Mentor Graphics QuestaSim + Cocotb for QuestaSim 10.4c

**Let's start with the simple part: Using MinGW32 (i686) to compile Cocotb with GCC 5.4.0...**

At first I tried my own compile script, because the shipped makefiles have problem to handle MSys2/MinGW32 in the right way. I extracted all compiler options for gcc, g++ and ld (called via gcc) from a running example on Debian Linux.

This works fine until `libvpi.dll` is going to be assembled. The library issues `undefined reference` errors. I checked the makefile rules for MinGW/Windows compilation and added the additional options (`-Wl,-no-undefined`, `-Wl,-enable-runtime-pseudo-reloc-v2`, `-Wl,--enable-auto-import`), but nothing changed.

OK, to write this message here, I was also going the long way and fixed Cocotb's makefiles for my system, so they run on MinGW32 and MinGW64... But in the end, even with Cocotb's original makefiles, nothing changed.

So I claim, that compiling Cocotb on MinGW32 is broken!

Let's review these above mentioned linker options:
- `-no-undefined`, according to the documentation, the correct name is `--no-undefined`.
- `-enable-runtime-pseudo-reloc-v2`, according to the documentation, the correct name is `--enable-runtime-pseudo-reloc`. I know there are some posts on the net, naming a `-v2` option, but `ld --help` doesn't know such an option.

I added a verbose flag (`-v`) to all gcc, g++ and ld calls and figured that these parameters are passed to `ld.exe`. I also added a `#warning` directive to `vpi_user.h`, to check whether `__declspec(dllimport)` is used. It's used.

**Let's test the next configuration: MinGW64 (x86_64):**

There are integer to pointer cast errors:
>D:/git/PoC/lib/cocotb/lib/simulator/simulatormodule.c: In function 'register_value_change_callback':
>D:/git/PoC/lib/cocotb/lib/simulator/simulatormodule.c:417:15: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
>     sig_hdl = (gpi_sim_hdl)PyLong_AsUnsignedLong(pSihHdl);
>
>D:/git/PoC/lib/cocotb/lib/simulator/simulatormodule.c: In function 'deregister_callback':
>D:/git/PoC/lib/cocotb/lib/simulator/simulatormodule.c:955:11: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
>     hdl = (gpi_sim_hdl)PyLong_AsUnsignedLong(pSihHdl);

When I disable "handle all warnings as errors", then I get the same errors as in MinGW32.

**OK, the last combination: GCC 4.5.0 from QuestaSim**

As expected there is the same error, because `libvpi.dll` is compiled before it reaches `libfli.dll` ...

**Other remarks**

According to some posts on the Internet, Windows DLLs do not support undefined references.

---------------------------
Appendix:

Here is the full linker output:

```Bash
gcc -v -shared -g -DDEBUG -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -Wl,--no-undefined -Wl,--enable-runtime-pseudo-reloc -Wl,--enable-auto-import -fno-common  -DFILTER -DPYTHON_SO_LIB=libpython2.7.dll -DPYTHON_SO_LIB=libpython2.7.dll -DVPI_CHECKING -o /d/git/PoC/temp/precompiled/ghdl/cocotb/libvpi.dll /d/git/PoC/temp/precompiled/ghdl/cocotb/VpiImpl.o /d/git/PoC/temp/precompiled/ghdl/cocotb/VpiCbHdl.o -shared -Wl,--no-undefined -Wl,--enable-runtime-pseudo-reloc -Wl,--enable-auto-import -L/c/msys64/mingw32/lib -L/usr/lib -L/d/git/PoC/temp/precompiled/ghdl/cocotb -lgpi -lgpilog -lstdc++
Using built-in specs.
COLLECT_GCC=C:\msys64\mingw32\bin\gcc.exe
COLLECT_LTO_WRAPPER=C:/msys64/mingw32/bin/../lib/gcc/i686-w64-mingw32/5.4.0/lto-wrapper.exe
Target: i686-w64-mingw32
Configured with: ../gcc-5.4.0/configure --prefix=/mingw32 --with-local-prefix=/mingw32/local --build=i686-w64-mingw32 --host=i686-w64-mingw32 --target=i686-w64-mingw32 --with-native-system-header-dir=/mingw32/i686-w64-mingw32/include --libexecdir=/mingw32/lib --with-gxx-include-dir=/mingw32/include/c++/5.4.0 --enable-bootstrap --with-arch=i686 --with-tune=generic --enable-languages=c,lto,c++,objc,obj-c++,fortran,ada --enable-shared --enable-static --enable-libatomic --enable-threads=posix --enable-graphite --enable-fully-dynamic-string --enable-libstdcxx-time=yes --disable-libstdcxx-pch --disable-libstdcxx-debug --enable-version-specific-runtime-libs --disable-isl-version-check --enable-lto --enable-libgomp --disable-multilib --enable-checking=release --disable-rpath --disable-win32-registry --disable-nls --disable-werror --disable-symvers --with-libiconv --with-system-zlib --with-gmp=/mingw32 --with-mpfr=/mingw32 --with-mpc=/mingw32 --with-isl=/mingw32 --with-pkgversion='Rev1, Built by MSYS2 project' --with-bugurl=https://sourceforge.net/projects/msys2 --with-gnu-as --with-gnu-ld --disable-sjlj-exceptions --with-dwarf2
Thread model: posix
gcc version 5.4.0 (Rev1, Built by MSYS2 project)
COMPILER_PATH=C:/msys64/mingw32/bin/../lib/gcc/i686-w64-mingw32/5.4.0/;C:/msys64/mingw32/bin/../lib/gcc/;C:/msys64/mingw32/bin/../lib/gcc/i686-w64-mingw32/5.4.0/../../../../i686-w64-mingw32/bin/
LIBRARY_PATH=C:/msys64/mingw32/bin/../lib/gcc/i686-w64-mingw32/5.4.0/;C:/msys64/mingw32/bin/../lib/gcc/;C:/msys64/mingw32/bin/../lib/gcc/i686-w64-mingw32/5.4.0/../../../../i686-w64-mingw32/lib/../lib/;C:/msys64/mingw32/bin/../lib/gcc/i686-w64-mingw32/5 [... cut ...]
D:/git/PoC/lib/cocotb/lib/vpi/VpiCbHdl.cpp:744: undefined reference to `_imp__vpi_get'
D:/git/PoC/lib/cocotb/lib/vpi/VpiCbHdl.cpp:745: undefined reference to `_imp__vpi_get_str'
D:/git/PoC/lib/cocotb/lib/vpi/VpiCbHdl.cpp:779: undefined reference to `_imp__vpi_iterate'
D:/git/PoC/lib/cocotb/lib/vpi/VpiCbHdl.cpp:794: undefined reference to `_imp__vpi_get_str'
D:/git/PoC/lib/cocotb/lib/vpi/VpiCbHdl.cpp:797: undefined reference to `_imp__vpi_get'
D:/git/PoC/lib/cocotb/lib/vpi/VpiCbHdl.cpp:816: undefined reference to `_imp__vpi_get'
collect2.exe: error: ld returned 1 exit status
```

elgorwi, 2016-07-05:
Hi @Paebbels,

The makefiles should pick up everything automatically from your environment. I have set following environment variable to ensure that the sytem path is picked up correctly in Msys2: MSYS2_PATH_TYPE=inherit
Without this variable, the makefiles won't be able to pick up the required info to work correctly.

I remember having tried to compile with the mentor supplied gcc-4.2.1-mingw32vc9 a long time ago, but that stopped working at some stage, so I moved to using winpython with bundled mingw 4.9.2.

For a 64 bit modelsim system, I've had success with WinPython-64bit-2.7.9.5.

Paebbels, 2016-07-06:
After a long debug session, I can say it works for MinGW (GCC 5.4.0) from MSys2 + Python 2.7.12 (CPython for Windows 64 bit). So it's all based on standard packages and doesn't need WinPython or Anaconda Python.

1. Common changes for all simulators:
    - The GCC tool chain is hardcoded in many places and needs to be exchanged with make variables
    - The make files are not Python 3 compatible
    - The make files do not properly support space characters in path names
    - Make files do not detect if the Python library is called `libpython27` or `libpython2.7` and if the file needs an `.dll` extension, because one has named it `*.dll.a`.
    - Many paths in the make files contain Windows paths, which is wrong. The need to converted to Linux paths, because assembling Windows paths with for example `C:\` causes problem in colon separated `PATH` variables.
2. Changes for QuestaSim:
    - libvpi.dll needs to be linked against libvsim.dll, because *undefined references* are not allowed in DLLs.
3. Changes for GHDL:
    - Issue: Cocotb doesn't implement the `vpi_register_sim` function

closed 2020-04-09 by eric-wieser