cocotb/cocotb #430: BinaryValue needs to support don't care, i.e. '-'

leftink, 2016-04-04:
The BinaryValue supports 'x', 'z', and 'u', all of which are resolved to 0.  It should also support the don't care, '-', and resolve it to 0 like the others.

mzabeltud, 2016-04-07:
I agree with the proposed resolving of 'L', 'H', and '-'. (The latter is now fixed #431).

But, I disagree to throw an exception for 'X', 'Z', 'W', and 'U'. These values may be permitted on a signal `s` if other signals indicate that the signal `s` is currently not relevant. The callback function of a `BusMonitor` should decide this. But from my current understanding of the framework, the thrown exception cannot be catched there.

Thus, I propose to pack all BinaryValues (before resolving) into a transaction for the callback. And then the callback can check for the 'X'-like values with a new function `is_x` for example. (In analogy to the VHDL IEEE.std_logic_1164 function.)

leftink, 2016-04-07:
@mzabeltud 
With the current approach, an assumption is made that 'X', 'Z', 'W', and 'U' should be zero.  Why is that a valid assumption?

The error only occurs if you try to convert the BinaryValue().  You can still call `str(dut.sig_vec)` and get the raw bit values, i.e. X, U, Z, etc., but no one has enough information to resolve those appropriately.  Further, it has been my experience that that if a vector has any of these values and it is actively being used in a design then there is a problem, either in the design or the testbench.

'X' -> Usually indicates a double driver
'Z' -> Usually indicates missing a "pull-up" or "pull-down" assignment
'U' -> Should have been initialized before use

I realize, that for instance a bus may have one of these values when not in use, but the monitor shouldn't be sampling at that time anyway.  If the bus indicates the data is valid, then all members of the bus should be valid at that time.

By automatically resolving these values  to zero, it can hide issues in your design.  This has already happened to me, where the test passes just fine, but when you open up a wave and look, I had one signal that actually had an error on it.  A prime example of this is the test "issue_142" on the Icarus Simulator.  Before this change, this test passed and everything looked ok, but after this change the test failed.  All the test was doing was setting a 64-bit vector to 0 using the set_signal_value(long) function and then reading it back and making sure it was zero.  With the change it turns out the setting was never happening and the vector was all 'Z's so the original way was actually suppressing an error.

Perhaps bus monitors should just return the BinaryValue and then the user can determine how to resolve things, but I still think it is correct that if the BinaryValue is asked to resolve something that it can't accurately resolve, it should raise an error so the user can address the issue.

mzabeltud, 2016-04-07:
You are right. Either the BusMonitor can ignore unsued signals, i.e., does not resolve them. Or the BusMonitor has to pass the binary value to the callback function to handle it. Thus, throwing an exception is appropriate.

I was mislead from my own BusMonitors which always capture all inputs/outputs and resolve them. Currently, they wouldn't detect 'X', 'Z' and 'U' which is bad.

closed 2016-06-19 by leftink

mzabeltud, 2016-06-20:
Was there a specific reason for closing this?

leftink, 2016-06-20:
@mzabeltud 
The changes have been merged into master.  I am not aware of anything else that is need for this.