cocotb/cocotb #432: Two or more signals trigger

kirknodeng, 2016-04-06:
I trying to use bus monitor.

I want to find some sequence with two signals.
example below :
two signals are A and B.
   11 - 00 - 01 - 00 

So, I used FallingEdge and RisingEdge method.
yield FallingEdge(A)
yield FallingEdge(B)

yield RisingEdge(B)
.....

But, cocotb couldn't detect both falling edge at same time. (detecting only fallingedge(A) )

Is it possible to detect two or more signals edge detection?

leftink, 2016-04-07:
With the approach you have, if both A and B change at the same time, yield FallingEdge(A) will pick up the change in A, but you have already missed the change in B, so the next line "yield FallingEdge(B)" will wait for the next edge.

I haven't used this, but have you tried the Combine() trigger.

```python
from cocotb.triggers import Combine

@cocotb.test()
def example(dut):
    yield Combine(FallingEdge(dut.A), FallingEdge(dut.B))
```

I believe that is how you use it.

kirknodeng, 2016-04-08:
I tested Combile() in my monitor class and in my @cocotb.test(). But, It's not working..
I have three different signals that indicate some state of transmission.
Sometimes their are change at same time. 
So, I want to get their state.

eric-wieser, 2019-03-19:
With #870 this can be written
```python
class Simultaneous(_AggregateWaitable):
    def _wait(self):
    	t = Timer(1)
    	all_t = Combine(triggers)
    	any_t = First(triggers)

    	while True:
    		triggers = list(self.triggers)

    		# wait for one to fire
    		@cocotb.coroutine
    		def wait_all():
    			yield all_t

    		# wait for the first one and all of them at the same time
    		all_task = cocotb.fork(wait_all)
    		yield First(triggers)
    		f = yield First(all_task, t)
    		if f is not t:
    			break

    		# stop waiting on the combination
    		all_task.kill()

    	raise ReturnValue(self)

```

ktbarrett, 2020-05-05:
xref #1689.

Another way to write would be to wait on any trigger to fire (`First`) and then checking the value of the signals afterwards

```python
# if you want them all to be high
async def all_high(handles):
    rising_edges = [RisingEdge(handle) for handle in handles]
    while not all(str(handle) = '1' for handle in handles):
         await First(*rising_edges)
```

closed 2020-05-05 by ktbarrett