cocotb/cocotb #435: Failed to write to packed structure input ports

zpan007, 2016-04-11:
modelsim failed with the following error:
```
Send raised exception: test_status[0] contains no object named value
#                                                                                                                                File "/home/pan/cocotb/cocotb/decorators.py", line 114, in send
#                                                                                                                                  return self._coro.send(value)
#                                                                                                                                File "/home/pan/code/cocotb/bug/test_testcase_top.py", line 52, in initialise
#                                                                                                                                  self.dut.test_status[i] = random.getrandbits(3)
#                                                                                                                                File "/home/pan/cocotb/cocotb/handle.py", line 441, in __setitem__
#                                                                                                                                  self.__getitem__(index).value = value
#                                                                                                                                File "/home/pan/cocotb/cocotb/handle.py", line 236, in __setattr__
#                                                                                                                                  return getattr(self, name)._setcachedvalue(value)
#                                                                                                                                File "/home/pan/cocotb/cocotb/handle.py", line 258, in __getattr__
#                                                                                                                                  raise AttributeError("%s contains no object named %s" % (self._name, name))
```

With Riviera Pro 2015.2, only the 9 LSB bits are set.

zpan007, 2016-04-11:
The testcase has been checked in to fork zpan007/cocotb/bug folder.

For modelsim, if line 52 is commented out and lines 53-55 are uncommented, the sim runs fine.

leftink, 2016-04-12:
@zpan007 
Even though the structure is packed, I don't believe you can drive a structure like you are.  You have to drive each field independently.  I see in you test you have code commented out to drive the three fields, does that work?

zpan007, 2016-04-12:
The commented out code works fine with the latest COCOTB on master branch
but not the released version v1.1(?). Modelsim fails with the current code
but Riviera doesn't. Better to fail if the structure can't be driven like
that.

Kind Regards,
Zhengjun

leftink, 2016-04-12:
@zpan007 
There was a lot of work after the initial v1.0 release so if you want to use structures I would definitely recommend using the latest version of the master branch.

As for the different behavior between the simulators, my guess is that this is due to what limitations/differences in their VPI interface.  When I worked on these interfaces, I saw a fairly wide range of support for structures in the VPI implementations.

I definitely saw some issues with ports versus nets and structures.  You might try creating a net and accessing the net directly to see if the behavior is the same as the port.

Fyi, what I saw was that in some cases, the "ports" were also seen as vpiReg/vpiNet so they had the right type information, but for more complex types like structures, some simulators didn't see structures at all and would just report the type as vpiPort.  COCOTB treats the vpiPort type as a Hierarchical construct.