cocotb/cocotb #436: not working on Windows

LinuxFanatic777, 2016-04-12:
No problems when running on Linux, but now I have to use Windows Server 2012 R2 Datacenter 64-bit.
I'm using Anaconda 32-bit (Python), ModelSim Altera Started Edition, cocotb from git of version 1.0.
Tried both MinGW 32-bit from Qt 5.6.0 installation and separate MinGW from SourceForge: the result is same, please see log:

```
python : Python 2.7.11 :: Anaconda 4.0.0 (32-bit)
At C:\Users\user\work\src\tests\testbench__project\run.ps1:5 char:1
+ python -V
+ ~~~~~~~~~
    + CategoryInfo          : NotSpecified: (Python 2.7.11 :: Anaconda 4.0.0 (32-bit):String) [], RemoteException
    + FullyQualifiedErrorId : NativeCommandError
 
mingw32-make : process_begin: CreateProcess(NULL, uname, ...) failed.
At C:\Users\user\work\src\tests\testbench__project\run.ps1:12 char:1
+ mingw32-make SIM=modelsim
+ ~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (process_begin: ...e, ...) failed.:String) [], RemoteException
    + FullyQualifiedErrorId : NativeCommandError
 
process_begin: CreateProcess(NULL, uname, ...) failed.

...
**here is many many times of "CreateProcess(NULL, uname, ...) failed"**
...

process_begin: CreateProcess(NULL, uname, ...) failed.
C:/Users/user/work/w/cocotb/makefiles/Makefile.pylib.:35: C:/Users/user/work/w/cocotb/makefiles/Makefile.pyli
b.: Too many open files

process_begin: CreateProcess(NULL, uname -m, ...) failed.
process_begin: CreateProcess(NULL, uname -m, ...) failed.
process_begin: CreateProcess(NULL, uname, ...) failed.
process_begin: CreateProcess(NULL, uname, ...) failed.
process_begin: CreateProcess(NULL, uname, ...) failed.
process_begin: CreateProcess(NULL, uname, ...) failed.
C:\Users\user\work\src\tests\testbench__project/../../../w/cocotb/makefiles/Makefile.inc:86: *** "Unsup
ported os " .  Stop.

```
How to use cocotb on Windows???
**Makefile.inc:86: *** "Unsupported os " .  Stop.** - why?

leftink, 2016-04-12:
@LinuxFanatic777 
Are you using the v1.0 release of COCOTB or are you using the latest on the master branch?  There has been a lot of work to get COCOTB to work on windows, but it is not in the v1.0 release.  The latest master should work on windows.

LinuxFanatic777, 2016-04-12:
No, this do not help. I'm using latest git version, now I tried again to update and run.
Still getting error with "process_begin: CreateProcess(NULL, uname, ...) failed."

leftink, 2016-04-12:
@LinuxFanatic777 
I poked around the internet and everything I am seeing is that you may not have properly installed mingw or at the very least installed all the "tools" required.  In your case it seems like the command "uname" is not being found on your machine.  At the terminal you issue make from, have you tried running the "uname" command by itself.  If that doesn't work then you need to look into your installation.

LinuxFanatic777, 2016-04-13:
Yes! I forgot to add path to msys bin dir, now I do not see any problems with uname, but another serious problems appeared:

```
python : Python 2.7.11 :: Anaconda 4.0.0 (32-bit)
At C:\Users\user\work\src\tests\testbench__project\run.ps1:7 char:1
+ python -V
+ ~~~~~~~~~
    + CategoryInfo          : NotSpecified: (Python 2.7.11 :: Anaconda 4.0.0 (32-bit):String) [], RemoteException
    + FullyQualifiedErrorId : NativeCommandError
 
mingw32-make : /usr/bin/sh: -c: line 0: syntax error near unexpected token `('
At C:\Users\user\work\src\tests\testbench__project\run.ps1:14 char:1
+ mingw32-make SIM=modelsim
+ ~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (/usr/bin/sh: -c...ected token `(':String) [], RemoteException
    + FullyQualifiedErrorId : NativeCommandError
 
/usr/bin/sh: -c: line 0: `C:\Program Files (x86)\Anaconda2/python.exe -c 'from distutils import sysconfig; print syscon
fig.get_config_var(\DESTSHARED\)''

/usr/bin/sh: -c: line 0: syntax error near unexpected token `('

/usr/bin/sh: -c: line 0: `C:\Program Files (x86)\Anaconda2/python.exe -c 'import distutils.sysconfig; print distutils.s
ysconfig.get_python_inc()''

/usr/bin/sh: -c: line 0: syntax error near unexpected token `('

/usr/bin/sh: -c: line 0: `C:\Program Files (x86)\Anaconda2/python.exe -c 'from distutils import sysconfig; print syscon
fig.get_config_var(\DESTSHARED\)''

/usr/bin/sh: -c: line 0: syntax error near unexpected token `('

/usr/bin/sh: -c: line 0: `C:\Program Files (x86)\Anaconda2/python.exe -c 'import distutils.sysconfig; print distutils.s
ysconfig.get_python_inc()''

/usr/bin/sh: -c: line 0: unexpected EOF while looking for matching `"'

/usr/bin/sh: -c: line 1: syntax error: unexpected end of file

sed: -e expression #1, char 8: unterminated `s' command

/usr/bin/sh: -c: line 0: syntax error near unexpected token `('

/usr/bin/sh: -c: line 0: `C:\Program Files (x86)\Anaconda2/python.exe -c 'from distutils import sysconfig; print syscon
fig.get_config_var(\DESTSHARED\)''

/usr/bin/sh: -c: line 0: syntax error near unexpected token `('

/usr/bin/sh: -c: line 0: `C:\Program Files (x86)\Anaconda2/python.exe -c 'import distutils.sysconfig; print distutils.s
ysconfig.get_python_inc()''

/usr/bin/sh: -c: line 0: syntax error near unexpected token `('

/usr/bin/sh: -c: line 0: `C:\Program Files (x86)\Anaconda2/python.exe -c 'from distutils import sysconfig; print syscon
fig.get_config_var(\DESTSHARED\)''

/usr/bin/sh: -c: line 0: syntax error near unexpected token `('

/usr/bin/sh: -c: line 0: `C:\Program Files (x86)\Anaconda2/python.exe -c 'import distutils.sysconfig; print distutils.s
ysconfig.get_python_inc()''

/usr/bin/sh: -c: line 0: unexpected EOF while looking for matching `"'

/usr/bin/sh: -c: line 1: syntax error: unexpected end of file

sed: -e expression #1, char 8: unterminated `s' command

mingw32-make[1]: *** No rule to make target '/home/user/work/src/project/project.v', 
needed by 'sim_build/runsim.do'.  Stop.

mingw32-make: *** [sim] Error 2
```

elgorwi, 2016-04-13:
Try installing python into a path without spaces or special characters.

LinuxFanatic777, 2016-04-13:
Yes, "path without spaces" fixed problem and It's getting better.
I have idea for future: what about environment checking script before start cocotb on Windows?
Now I see problems with **sed**:

```
starting...
python : Python 2.7.11 :: Anaconda 4.0.0 (32-bit)
At C:\Users\user\work\src\tests\testbench__project\run.ps1:7 char:1
+ python -V
+ ~~~~~~~~~
    + CategoryInfo          : NotSpecified: (Python 2.7.11 :: Anaconda 4.0.0 (32-bit):String) [], RemoteException
    + FullyQualifiedErrorId : NativeCommandError
 
mingw32-make : sed: -e expression #1, char 8: unterminated `s' command
At C:\Users\user\work\src\tests\testbench__project\run.ps1:14 char:1
+ mingw32-make SIM=modelsim
+ ~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (sed: -e express...ted `s' command:String) [], RemoteException
    + FullyQualifiedErrorId : NativeCommandError
 
C:/Users/user/work/w/MinGW/bin/mingw32-make results.xml
sed: -e expression #1, char 8: unterminated `s' command

mingw32-make[1]: Entering directory 'C:/Users/user/work/src/tests/testbench__project'
mingw32-make[1]: Leaving directory 'C:/Users/user/work/src/tests/testbench__project'
mingw32-make[1]: *** No rule to make target '/home/user/work/src/project/project.v', 
needed by 'sim_build/runsim.do'.  Stop.

C:\Users\user\work\src\tests\testbench__project/../../../w/cocotb/makefiles/Makefile.sim:71: recipe for target 'sim' failed
mingw32-make: *** [sim] Error 2
```

elgorwi, 2016-04-13:
I think the path of project.v is causing the issue. 
Is the cocotb folder located in a sensible location, eg. C:/cocotb ?
Try running the 'adder' example first to get the system up and running

closed 2020-04-09 by eric-wieser