cocotb/cocotb #437: Replace makefiles through Python

Paebbels, 2016-04-21:
Hello, 

we from PoC have integrated Cocotb with QuestSim backend as another simulator into our list of supported tool chains. But there are some issues.

Using makefile is a bit contra productive when being platform independent. Moreover these files are very hard to maintain. We gethered several workaround.

While reading and understanding these files, we figured out that more then 50% of the makefile code is sequential code using Bash code just in makefile format and using target rules as sub programs.

Wouldn't it be better to put all this code into a clever Python run? Controlling simulators from Python is much easier. And gcc runs only once for a hand full of files.

Maybe we can share some Python code to do these build jobs.

stuarthodgson, 2016-04-21:
@Paebbels 

When you mention work arounds where needed can you share these? they could be of use to others.

From our point of view we do not need to maintain the Makefiles as they just work for us. Which I accept may not be the case for everyone.

The problem with any build system is that it always has detractors, nothing is perfect with any of them particularity custom solutions that may appear better at first glance.

Please do share the python code that you have developed but i'm not convinced yet there is a need to make what would be a pretty sweeping change, I am open to being convinced though.

Thanks

Stu

Paebbels, 2016-04-21:
Hello Stuart,

I'll test my workarounds and improvements. If it works out, I'll create a pull-request for your repository.

You can find our Python code in the [`py/`](https://github.com/VLSI-EDA/PoC/tree/release/py) sub-directory. It's not perfect nor fully decoupled, but it's a major step in the direction of OOP compared to the initial version of our Python infrastructure behind our IP cores.

Major parts:
  - base classes for simulators, compilers, configurations, tool-chains...
  - one simulator package per supported simulator
      - Active-HDL
      - [Cocotb (with QuestaSim backend)](https://github.com/VLSI-EDA/PoC/blob/release/py/Simulator/CocotbSimulator.py)
      - GHDL (with mcode, gcc or llvm backend)
      - [QuestaSim](https://github.com/VLSI-EDA/PoC/blob/release/py/Simulator/QuestaSimulator.py)
      - ISE Simulator
      - Vivado Simulator
  - one compiler package per supported (synthesis) compiler
      - Altera Quartus II
      - Lattice Diamond LSE
      - Xilinx CoreGen
      - Xilinx XST
  - one ToolChain package for each tool-chain
      - Aldec_ActiveHDL
      - Altera_Quartus
      - GHDL_GTKWave
      - Lattice_Diamond
      - [Mentor_QuestaSim](https://github.com/VLSI-EDA/PoC/blob/release/py/ToolChains/Mentor/QuestaSim.py)
      - Xilinx_ISE
      - Xilinx_Vivado

We use a set of [`*.files`](https://github.com/VLSI-EDA/PoC/blob/release/src/common/common.files) files to specify which source files (VHDL, Verilo, UCF, SDC, XDC, ...) are required to compile or simulate a PoC entity. This universal format with conditional and include statements can be translated into any vendor specific project file (*.qsf or *.prj) or into a sequence of command line calls, for example to *vlib, vom, vsim*. To patch generated IP cores from a vendor, we also have [`*.rules`](https://github.com/VLSI-EDA/PoC/blob/release/src/xil/mig/mig_Atlys_1x128.rules) files.

I'll need a few more days to update our new (vendor) tool configuration process, so a user does not need to write this [config.private.ini](https://github.com/VLSI-EDA/PoC/blob/release/tools/Travis-CI/config.private.ini) file by hand, but if you like, you can already test Cocotb in PoC by filling out the QuestaSim section with the same options as in the GHDl section.

Running a cache simulation in PoC: `./poc.sh -v cocotb PoC.cache.par`

You can chose other simulators (asim, isim, vsim, xsim, ghdl) or synthesizers (lattice, quartus, xst, coregen) if you have a local installation. `--gui` opens the simulation in GUI mode.

leftink, 2016-07-05:
@stuarthodgson I know make works for you guys, but have you ever looked at [Waf](https://waf.io/)?  It is a python based build system that might fit nicely with COCOTB.

We are evaluating at our company now as our potential new build framework.

stuarthodgson, 2016-07-07:
@leftink It has come up recently for other work and I had a quick look and others have recommended too. There seems to be a new build system every hour these days and all have problems that I have yet to be convinced of enough to move to one rather than the standard on Linux of gnu Make. Don't get me wrong it's a pain as well but one that I already know (well enough anyway). I also worry about introducing dependencies to existing users. I'll take more of a look though.

leftink, 2016-07-11:
@stuarthodgson I am not necessarily suggesting you should change.  Frankly, if you have something that works, I am not sure I would bother, but as I have been looking at it, I just thought I would mention it.

As far as introducing dependencies, you have already done that by choosing "make".  Windows platforms don't come with make and requires additional tools to be installed, i.e. cygwin, etc.  Given that Waf is just a python script that I believe can be bundled with COCOTB, there would be no additional tools necessary for users.

Paebbels, 2016-07-11:
Installing new Python modules is harmless in contrast to Cygwin or a 3.6 GiB MSys2 / MinGW32/64 installation ...

@stuarthodgson
Last time, you ask for our fixes and work-arounds .... our pull request #467 is still waiting to be merged.

stuarthodgson, 2016-07-11:
@Paebbels Sorry about that, not excuse use apart from I did not notice.

stuarthodgson, 2016-07-11:
@leftink @Paebbels I would agree that ```waf``` or something else is less overhead for the Windows users. Please don't get the wrong idea that I am not open to making things better. 

The problem being that neither @chiggs  or I am windows users and the project started on Linux so make was the natural choice, The fact that Windows users might find the project useful did not really occur to us. I'm glad it has helped more people than we first envisaged.

But this does mean that unless a member of the community fancies doing something better not much is going to change since It is pretty far down our list of tasks.

ktbarrett, 2022-03-06:
Closed by #2634.

closed 2022-03-06 by ktbarrett