cocotb/cocotb #440: conda or pip installable (enhancement request)

amal-khailtash, 2016-05-04:
Make cocotb conda or pip installable package.  That would make it easy to install from conda repo or PyPI.

jfosorio, 2016-09-18:
Conda would make cocotb so powerfull,  there is already a receipt for iverilog. With conda would be  posible to use numpy/scipy/ski-learn/jupyter. 
I would like to spent some time on it but find difficult to getting started.

jeremyherbert, 2016-09-18:
I think the best thing would be to make it pip installable with wheels. Could probably do the wheel building automatically on appveyor and Travis, and it has the added bonus that Windows users won't have to individually go through the pain of installing toolchains. But this would also necessitate changing the build system to not be dependent on Make, otherwise Windows users will still have to go through that all. I'm not actually a Windows user, but I imagine in corporate environments it is quite common for HDL work.

This is something that I plan to work on when I get time, but I won't have any until the end of the year.

elmsfu, 2018-05-01:
I see #517 has most of the work for pip. Also https://pypi.org/project/cocotb has a version of this from early 2017.

themperek, 2019-08-08:
PyPI package is available now.

closed 2019-08-08 by themperek