cocotb/cocotb #441: how can cocotb libraries be compiled as static?

chevillotn, 2016-05-12:
I have several users willing to use the cocotb on different machines hence with different versions of GLIBCXX installed.

I do not want them to compile the cocotb libraries each time they want to simulate their code and would like therefore to provide them with pre-compiled libraries.

If i compile the cocotb on my machine, it will be compiled against a specific GLIBCXX version. If the user does not have this version on his/her computer then it won't work.

Is there a way to statically link the GLIBCXX library with cocotb libraries?

luked80, 2016-06-07:
Adding ```-static-libgcc``` or ```-static``` to gcc seems to be the way to do this (disclaimer, I haven't actually tried this with cocotb).
More details at these links:
http://www.systutorials.com/5217/how-to-statically-link-c-and-c-programs-on-linux-with-gcc
http://stackoverflow.com/questions/13187499/link-glibc-statically-but-some-other-library-dynamically-with-gcc

stuarthodgson, 2016-06-18:
This is not so much making cocotb build static libs since the simulator will always need to load a shared object as an entry point, ```-static-libgcc``` does not pull in libstdc++. What you want to do is link in the C++ runtime into the cocotb libs such that the simulator loads it all. Unfortunately this is not easy/prone to problems. What you could do is install the GLIBCXX version that you need on a network mount and then set LD_LIBRARY_PATH to point to that location. You can then shared the cocotb libs more easily.

I'll be doing some work shortly to update cocotb to build a set of simulation libraries for a given set of simulators as part of a better installation in order to solve a similar problem with rebuilding all the time. Should be doing that in a couple of weeks.

eric-wieser, 2020-04-09:
@themperek, any idea if this still applies? Do we need to worry about libc++ version mismatches?

themperek, 2020-04-09:
@eric-wieser No idea.

themperek, 2020-04-09:
Since all this is different now I propose to close this. If problem persist needs a way to reproduce to try to solve this.

ktbarrett, 2020-04-09:
@themperek You should be able to pass `CFLAGS="-static-libgcc -static-libstdc++"` when building. I'm not sure how to do that with pip, I tried something and it didn't seem to work, `nm` shows no static linking.

themperek, 2020-04-09:
`CFLAGS="-static-libgcc -static-libstdc++" pip install -v cocorb` should work.

themperek, 2020-04-09:
And it works for me 👍

ktbarrett, 2020-04-09:
I think I was having a problem because it was using the build cache and not rebuilding the objects when I ran pip.

The results I see are that libstdc++ objects are linked weakly. My hunch is that we are using `-static-libstdc++` and `_build_libs.py` specifically calls out `libstdc++` for linking. This probably deserves some investigation because I have a feeling those weak links will be overloaded if a user has their own libstdc++ loaded (the concern of the OP).

This might also be an issue for me, some of our environments are using Anaconda 4.4.0 with it's ancient GCC 4.4.7-era libs that don't support C++11.

themperek, 2020-04-09:
>@themperek sourcing a newer libstdc++11 isn't always an option.

Why in case case this would not work?

ktbarrett, 2020-04-09:
Strict or bureaucratic software approval processes. Closed labs. Those libraries might exist, but it could be a huge hurdle to get them approved and distributed to users.

themperek, 2020-04-09:
@ktbarrett Just copy to libs folder ?

cmarqu, 2020-04-09:
Not when there is a risk of being visited by security.

themperek, 2020-04-10:
So conclusion.

Because we need to make dynamic libraries to be able to load by simulators some solutions may be (all in master for now): 

-  one can try `CFLAGS="-static-libgcc -static-libstdc++" pip ...` but this may not work see: https://stackoverflow.com/questionhs/26304531/compiling-with-static-libgcc-static-libstdc-still-results-in-dynamic-depende

- copy dependencies to `cocotb/libs` folder see: https://stackoverflow.com/questions/13636513/linking-libstdc-statically-any-gotchas (we set `$ORIGIN` on all libraries)

ktbarrett, 2021-01-30:
Does this deserve to remain open? Is there any additional work we should do to combat this? It seems the proper answer is to work around the simulator trying to load their libstdc++ over the system one. Really it should be whatever libstdc++ is the highest version, but `ld` doesn't support that behavior. Static compilation isn't the "correct" fix in any case. We *could* do the same thing as we do with libpython with *all* of our dynamic dependencies, but I really don't think that's reasonable.

I can understand simulators shipping with their own libpython and python installations to force their use (since their scripts probably depend on a particular version of Python), but it's just not reasonable in the case of libstdc++ that is designed to be totally backwards compatible. However, they don't separate the libraries so they can be inserted into the paths in particular orders.

closed 2021-08-05 by github-actions[bot]