cocotb/cocotb #444: Ubuntu 16.04 and anaconda core dump

jeremyherbert, 2016-05-29:
Hi,

I just tried to install cocotb on a fresh Ubuntu 16.04 (actually Xubuntu, but it should be the same).

The first issue I came across was that anaconda (and the inbuilt python for that matter) do not use `python-config` for python 3, but rather `python3-config`. I fixed this by softlinking `python-config` in the anaconda bin directory:

    ln -s python3-config python-config

The next problem I had was along the lines of this

```
...
ln -sf libvpi.so /home/jeremy/code/cocotb/build/libs/x86_64/gpivpi.vpl
ln -sf libvpi.so /home/jeremy/code/cocotb/build/libs/x86_64/cocotb.vpi
make[2]: Leaving directory '/home/jeremy/code/cocotb/lib/vpi'
PYTHONPATH=/home/jeremy/code/cocotb/build/libs/x86_64:/home/jeremy/code/cocotb:/home/jeremy/code/cocotb/examples/adder/tests:/home/jeremy/code/cocotb/examples/adder/tests/../model: LD_LIBRARY_PATH=/home/jeremy/code/cocotb/build/libs/x86_64::/home/jeremy/anaconda3/lib:/home/jeremy/anaconda3/lib:/home/jeremy/anaconda3/lib:/home/jeremy/anaconda3/lib MODULE=test_adder \
        TESTCASE= TOPLEVEL=adder TOPLEVEL_LANG=verilog \
        /usr/local/bin/vvp -M /home/jeremy/code/cocotb/build/libs/x86_64 -m gpivpi sim_build/sim.vvp   
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Fatal Python error: Py_Initialize: Unable to get the locale encoding
ImportError: No module named 'encodings'

Current thread 0x00007f5480c4c740 (most recent call first):
Aborted (core dumped)
/home/jeremy/code/cocotb/makefiles/simulators/Makefile.icarus:77: recipe for target 'results.xml' failed
make[1]: *** [results.xml] Error 134
make[1]: Leaving directory '/home/jeremy/code/cocotb/examples/adder/tests'
/home/jeremy/code/cocotb/examples/adder/tests/../../../makefiles/Makefile.sim:71: recipe for target 'sim' failed
make: *** [sim] Error 2
```

`strace` shows that the python interpreter is looking in the wrong place for the built in libraries:

```
...
stat("/opt/anaconda1anaconda2anaconda3/lib/lib-dynload", 0x7ffed09d51a0) = -1 ENOENT (No such file or directory)
stat("/opt/anaconda1anaconda2anaconda3/lib", 0x7ffed09d51a0) = -1 ENOENT (No such file or directory)
stat("/opt/anaconda1anaconda2anaconda3", 0x7ffed09d51a0) = -1 ENOENT (No such file or directory)
stat("/opt", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
stat("/opt/anaconda1anaconda2anaconda3/lib/lib-dynload", 0x7ffed09d4ac0) = -1 ENOENT (No such file or directory)
write(2, "Fatal Python error: Py_Initializ"..., 69Fatal Python error: Py_Initialize: Unable to get the locale encoding
) = 69
write(2, "ImportError", 11ImportError)             = 11
write(2, ": ", 2: )                       = 2
write(2, "No module named 'encodings'", 27No module named 'encodings') = 27
write(2, "\n", 1
)                       = 1
write(2, "\n", 1
)                       = 1
write(2, "Current thread 0x", 17Current thread 0x)       = 17
write(2, "00007fdb506ca740", 1600007fdb506ca740)        = 16
write(2, " (most recent call first):\n", 27 (most recent call first):
) = 27
rt_sigprocmask(SIG_UNBLOCK, [ABRT], NULL, 8) = 0
tgkill(7729, 7729, SIGABRT)             = 0
--- SIGABRT {si_signo=SIGABRT, si_code=SI_TKILL, si_pid=7729, si_uid=1000} ---
+++ killed by SIGABRT (core dumped) +++
Aborted (core dumped)
```

The solution is to set the `$PYTHONHOME` variable like so:

`export PYTHONHOME=/home/jeremy/anaconda3/`

After this, the adder sim works properly. I am not sure if this is a bug in cocotb, python or anaconda, but I thought I should report it anyway.

leftink, 2016-05-29:
@jeremyherbert 
I have already submitted a fix for the first issue. The second issue I believe is just set a python configuration deal. I had to do the same thing with setting PYTHONHOME.

themperek, 2019-08-08:
This should be fixed in other cases please reopen.

closed 2020-04-16 by themperek