cocotb/cocotb #445: Reusing a cocotb.external thread

luked80, 2016-06-01:
I'm attempting to embed a tcl interpreter into python/cocotb.

I'm passing some cocotb function pointers into the tcl interpreter, so that it can call back into cocotb/python.
I can start the interpreter and make calls back into cocotb in the tcl interpreter's ``` __init__ ```function.
I start the interpreter and save a reference to is as follows (I'm using [plumage](https://code.google.com/archive/p/plumage) to embed the tcl interpreter):
```python
import plumage
interp = yield cocotb.external(plumage.Interp)(use_tk=0)
```
So far so good.
Next thing I want to do is call some tcl code inside my interpreter.
Naively I try and call some tcl as:
```python
tcl_result = yield cocotb.external(interp.call)("array", "get", "tcl_platform")
print(tcl_result)
```
But this results in the following error message:
```bash
Fatal Python error: PyEval_SaveThread: NULL tstate
```

Removing the yield and cocotb.external and the above example works as expected.
But will obviously break the ability to call back into cocotb.

Digging a little deeper, the external decorator creates a threading.Thread object for every call to run in.
And I think this causes the problem as interp.call expects to find the interpreter in the current thread.

I'm thinking I'll need to create my own version of cocotb.external, that keeps track of the tcl interpter instance's thread.
And at the same time accepts commands from other threads and passes them to the interpreter (maybe with a Queue).

Am I on the right track here?
Has anyone had to do anything like this before?

stuarthodgson, 2016-06-06:
@luked80 we use cocotb.external quite extensively for co-simulation with software libraries that will be used in production systems. These then normally call back into python via some form of trap to read/write to registers in the DUT or sample other data.

cocotb.external was designed with the intention of running command and allowing simulation time to progress via a yield call from inside the command.

Have you tried the following approach. Wrap the TCL interpreter in a queue as you suggest looping over the head. This is started via a call to cocotb.external but inside a forked co-routine. You should then be able to call into the wrapper object to push onto the queue that is used by the interpreter. Sorry if not clear it's been a long day. The main idea is keep the interpreter alive.

luked80, 2016-06-07:
Thanks Stu.

I've implemented something similar to what you suggest.
And it seems to work ok.

I pass a queue into cocotb.external along with the command to start the tcl interpreter.
And then push commands to the tcl interpreter, running in the cocotb.external thread, through the queue.

I still need to figure out how to let each push into the queue block the main thread.
But I think I can achieve that in a similar fashion to the usage of test_locker in cocotb.external.

Once I've got something I'm happy with, I'll put together a working example and push to github.
I'd imagine others also have to carry around a bunch of legacy tcl configuration code like I do.
And might find it useful.

luked80, 2016-07-13:
@stuarthodgson I have a prototype for this on a branch here:
https://github.com/luked80/cocotb/tree/issue_445
with the sole commit here:
https://github.com/luked80/cocotb/commit/333b276bad5334016b5d3ca36513328cdc90608a

But I haven't bee able to get it to work so far.
The simulation just errors our with:
```
  0.00ns INFO     cocotb.sample_module                     test_cocotb.py:548  in test_tcl                        Creating the tcl_queue object
  0.00ns INFO     cocotb.sample_module                     test_cocotb.py:550  in test_tcl                        Starting up the thread that the tcl interpreter runs in.
  0.00ns INFO     cocotb.sample_module                     test_cocotb.py:551  in test_tcl                        And watches for commands on the queue.
  0.00ns ERROR    cocotb.scheduler                            __init__.py:172  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely
  0.00ns ERROR    cocotb.regression                         regression.py:239  in handle_result                   Test error has lead to simulator shuttting us down
  0.00ns ERROR    cocotb.regression                         regression.py:167  in tear_down                       Failed 1 out of 1 tests (0 skipped)
  0.00ns INFO     cocotb.regression                         regression.py:176  in tear_down                       Shutting down...
```

I can't really see what I'm doing wrong here, and there's no error messsage.
Wondering if you can take a quick look?

You can run with:
```
cd tests/test_cases/test_cocotb
make TESTCASE=test_tcl
```

luked80, 2016-07-14:
Some more progress on this.
I now have an example that works.
But it's not exactly nice.

I found that I need to keep the coroutine that starts the tcl interpreter thread alive and ticking over a Timer.
If not, we get the silent error as before.
This is kind of annoying as the simulation times can vary from run to run.
f7e6b0f0dcf2961cd1 has the change showing the working but less than ideal prototype.

ktbarrett, 2025-03-27:
The passage of time was fixed in #645. This is similar to what was described in #2811. There needs to be a way to re-enter the cocotb event loop from an external thread, `cocotb.function` (now `cocotb.resume`) is not doing that. This would also be useful for calling into cocotb from DPI functions.

closed 2025-03-27 by ktbarrett