cocotb/cocotb #447: parameters of top module

LinuxFanatic777, 2016-06-06:
Hello!
I have top module (Verilog language) with default parameters:
```
module my_nice_module
#(parameter COEF_FILE_A = "coef_a1.dat",
			COEF_FILE_B = "coef_a2.dat",
			MY_WIDTH = 16
)
```
and it's usage:
```
TOPLEVEL = my_nice_module
MODULE = testbench__my_nice_module
TOPLEVEL_LANG = verilog
```
Is there any way to override default parameters in Makefile or somewhere else?

pgfarley, 2016-06-06:
For modelsim you can do this with the -g command line arg which can be specified in the Makefile file by assigning VSIM_ARGS. e.g. VSIM_ARGS=-gMY_WIDTH=20

elgorwi, 2016-06-06:
There is a 'GENERICS' variable for modelsim/questa for that purpose:
```sh
GENERICS='COEF_FILE_A="coef_a1.dat" COEF_FILE_B="coef_a2.dat" MY_WIDTH=16'
```

kirknodeng, 2016-06-06:
You can use this below :
ncsim : EXTRA_ARGS=+defparam+MY_WIDTH=16
vcs : EXTRA_ARGS=-pvalue+MY_WIDTH=16

LinuxFanatic777, 2016-07-29:
I solved that problem... just by wrapping DUT into one more top module where I can set all parameters in Verilog language :)

closed 2016-07-29 by LinuxFanatic777