cocotb/cocotb #448: What does ReadOnly() actually do?

jeremyherbert, 2016-06-06:
Hi,

Sorry, I'm not sure this is a bug, but I can't seem to find any documentation on this. Using `yield ReadOnly()` seems to delay the simulation to the falling edge of my clock, but I have no idea how it is working this out. 

For example:

simple.v:
```
`timescale 1ns / 1ns

module simple(input wire clk, input wire a, input wire b);

initial begin
    $dumpfile("simple.vcd");
    $dumpvars(0, simple);
end

endmodule
```

simple_test.py:
```
import cocotb
from cocotb.triggers import RisingEdge, ReadOnly, Timer

@cocotb.coroutine
def clock(pin):
    while True:
        pin <= 1
        yield Timer(1000)
        pin <= 0
        yield Timer(1000)

@cocotb.test()
def do_test(dut):
    cocotb.fork(clock(dut.clk)) # make clock tick

    for i in range(5):
        yield RisingEdge(dut.clk)

    yield RisingEdge(dut.clk)
    dut.a = 1  # gtkwave shows this as aligned with the rising edge of dut.clk, makes sense

    yield ReadOnly()
    dut.b = 1  # gtkwave shows this as aligned with the falling edge of dut.clk ???

    for i in range(5):
        yield RisingEdge(dut.clk)
```

How/why does cocotb/icarus decide that my signals are read-only at the falling edge of the clock? How does it even know which signal is the clock when I didn't pass it anything? The D of a rising edge flipflop is supposed to be changing around the falling edge transition (kind of?), so that's clearly not read-only. 

I would have guessed that `yield ReadOnly()` would do something like: "process all of the assignment statements for this timestep, but yield just before the simulation actually advances time". But it seems like what it is really doing is "advance to the next scheduled event and yield".

Please enlighten me :)

jeremyherbert, 2016-06-06:
In fact, I just checked and that is exactly what it is doing. If I add the following coroutine:

```
@cocotb.coroutine
def spin():
    while True:
        yield Timer(10)

...
cocotb.fork(spin())
```

then `ReadOnly()` yields 10 simulator ticks later.

stuarthodgson, 2016-06-06:
@jeremyherbert ReadOnly() is there to provide a way to only allow the yielding co-routine continue when the current timestep no longer has any further delta steps. This should be point where all the signal values are stable as there are no more events scheduled for the timestep. The simulator should not allow scheduling of more events in this timestep. There are two testcases that check this works on a simulator. 

```
cd tests/test_cases/test_cocotb/
make TESTCASE=test_afterdelay_in_readonly
make TESTCASE=test_afterdelay_in_readonly_valid
```

First should fails as expected and the second should pass.

What these test cases do not seem to do is check what happens to simulation time. Is the behaviour the same on another simulator?

jeremyherbert, 2016-06-07:
After sleeping on it, I think I realised what is happening. Because there are no more events that can be scheduled, icarus queues the `dut.b = 1` assignment for the next simulated timestep. This is confirmed by printing out a debug message after `yield ReadOnly()` which shows that the simulation time hasn't yet advanced.

I'm not sure if other simulators behave this way (it would be nice if there was an exception) as I don't have access to any others. But my problem is sorted so I'm closing this ticket. Please reopen if you feel it needs further investigation.

jeremyherbert, 2016-06-07:
Just to add to this: it seems to be a "bug" in the vcd output? either that or a result of undefined behaviour. If you write to a signal during the read-only phase, cocotb will report the time as not changing but the values have changed. However, if you open the vcd file from icarus, it will report that the edge is located at  the next simulated event after the write.

top.v
```
module top(input wire clk, output reg a, input wire b);

    initial begin
        $dumpfile("top.vcd");
        $dumpvars(0, top);
    end

    always @(posedge clk) begin
        a <= clk;
    end

endmodule
```

test.py:
```
import cocotb
from cocotb.triggers import Timer
from cocotb.result import TestFailure
from cocotb.triggers import RisingEdge, ReadOnly, ReadWrite
from cocotb.clock import Clock

import simulator

@cocotb.test()
def test_write_during_read_only(dut):
    clock = Clock(dut.clk, 1000)
    cocotb.fork(clock.start())
    dut.b = 0

    for i in range(10):
        yield RisingEdge(dut.clk)

    yield RisingEdge(dut.clk)
    sim_time_before_read_only = simulator.get_sim_time()

    yield ReadOnly()
    sim_time_after_read_only = simulator.get_sim_time()

    dut.b = 1
    sim_time_after_read_only_and_write = simulator.get_sim_time()

    dut.log.info("time before: " + str(sim_time_before_read_only))
    dut.log.info("time after: " + str(sim_time_after_read_only))
    dut.log.info("time after write: " + str(sim_time_after_read_only_and_write))
    dut.log.info("has value been written yet: " + str(bool(dut.b)))

    for i in range(10):
        yield RisingEdge(dut.clk)
```

output:
```
10000000000000.00ns INFO     cocotb.top                                      test.py:27   in test_write_during_read_only     time before: (0, 10000)
10000000000000.00ns INFO     cocotb.top                                      test.py:28   in test_write_during_read_only     time after: (0, 10000)
10000000000000.00ns INFO     cocotb.top                                      test.py:29   in test_write_during_read_only     time after write: (0, 10000)
10000000000000.00ns INFO     cocotb.top                                      test.py:30   in test_write_during_read_only     has value been written yet: True
20001000000000.00ns INFO     cocotb.regression                         regression.py:209  in handle_result                   Test Passed: test_write_during_read_only
20001000000000.00ns INFO     cocotb.regression                         regression.py:170  in tear_down                       Passed 1 tests (0 skipped)
20001000000000.00ns INFO     cocotb.regression                         regression.py:176  in tear_down                       Shutting down...
```

jeremyherbert, 2016-06-07:
And it seems this issue is further documented in the MyHDL docs: http://docs.myhdl.org/en/stable/manual/cosimulation.html#delta-cycle-implementation

jeremyherbert, 2016-06-07:
Ok, I'm reopening this as something fishy is going on. When I run the cocotb unit tests, I get the following message on stderr from iverilog: `SCHEDULER ERROR: read-only sync events created RW events!`

But in my code above, I don't get any errors?

jeremyherbert, 2016-06-07:
Yes, it looks like the test case is not working correctly. Here is the code:

```
exited = False

@cocotb.coroutine
def do_test_readwrite_in_readonly(dut):
    global exited
    yield RisingEdge(dut.clk)
    yield ReadOnly()
    dut.clk <= 0
    yield ReadWrite()
    exited = True

@cocotb.test(expect_error=True,
             expect_fail=cocotb.SIM_NAME.lower().startswith(("icarus",
                                                             "riviera",
                                                             "modelsim",
                                                             "ncsim")))
def test_readwrite_in_readonly(dut):
    """Test doing invalid sim operation"""
    global exited
    exited = False
    clk_gen = cocotb.fork(Clock(dut.clk, 100).start())
    coro = cocotb.fork(do_test_readwrite_in_readonly(dut))
    yield [Join(coro), Timer(10000)]
    clk_gen.kill()
    if exited is not True:
        raise TestFailure
```

The line which actually causes the error is `yield ReadWrite()`, not `dut.clk <= 0`, and the error is presumably because you can't reschedule after the `ReadOnly()` event. But this test is misleading, because you actually can write to `dut.clk` without an error, so that line doesn't actually do anything. In fact, you can comment out the line and the test still passes.

jeremyherbert, 2016-06-08:
After reading about this for a bit today trying to understand what is happening, I need some help from those of you with commercial simulator licenses.

According to the MyHDL folks, the behaviour of `ReadWrite()` is not consistent across simulators. If anyone has access to the big four commercial simulators (I don't) I'd really appreciate if you could test this behaviour to see if it is consistent. If you could run the following test case for me and paste the output here, that would be great! :)

Here are the test files:

Makefile:
```
VERILOG_SOURCES = top.v
TOPLEVEL_LANG=verilog
TOPLEVEL=top
MODULE=test
include $(COCOTB)/makefiles/Makefile.inc
include $(COCOTB)/makefiles/Makefile.sim
```

top.v
```
`timescale 1ns/1ps

module top(input wire clk, output wire clk_out, output reg [3:0] counter);

    assign clk_out = clk;

    initial begin
        counter <= 0;
    end
    always @(posedge clk) counter <= counter + 1;

endmodule
```

test.py:
```
import cocotb
from cocotb.triggers import RisingEdge, ReadOnly
from cocotb.clock import Clock

@cocotb.test()
def test_order_of_assignments(dut):
    """
    Testing the order of assignments
    """
    clock = Clock(dut.clk, 1000)
    cocotb.fork(clock.start())

    dut.log.info("starting order of assignment test")

    for i in range(10):
        yield RisingEdge(dut.clk) # this is also implicitly `yield ReadWrite()`
        counter_before = dut.counter.value
        clk_out_before = dut.clk_out.value

        yield ReadOnly()
        counter_after = dut.counter.value
        clk_out_after = dut.clk_out.value

        if counter_before != counter_after:
            dut.log.info("non-blocking assignment value was different after ReadOnly(), was {}, now {}".format(counter_before, counter_after))
        else:
            dut.log.info("non-blocking assignment value was the same after ReadOnly()")

        if clk_out_before != clk_out_after:
            dut.log.info("continuous assignment value was different after ReadOnly(), was {}, now {}".format(clk_out_before, clk_out_after))
        else:
            dut.log.info("continuous assignment value was the same after ReadOnly()")
```

Just put them all in the same folder and run `SIM=modelsim make`, where `modelsim` is your simulator.

[... 25 lines cut ...]
  1.00ns INFO     cocotb.top                                      test.py:25   in test_order_of_assignments       non-blocking assignment value was different after ReadOnly(), was 1, now 2
  1.00ns INFO     cocotb.top                                      test.py:30   in test_order_of_assignments       continuous assignment value was different after ReadOnly(), was 0, now 1
  2.00ns INFO     cocotb.top                                      test.py:25   in test_order_of_assignments       non-blocking assignment value was different after ReadOnly(), was 2, now 3
  2.00ns INFO     cocotb.top                                      test.py:30   in test_order_of_assignments       continuous assignment value was different after ReadOnly(), was 0, now 1
  3.00ns INFO     cocotb.top                                      test.py:25   in test_order_of_assignments       non-blocking assignment value was different after ReadOnly(), was 3, now 4
  3.00ns INFO     cocotb.top                                      test.py:30   in test_order_of_assignments       continuous assignment value was different after ReadOnly(), was 0, now 1
  4.00ns INFO     cocotb.top                                      test.py:25   in test_order_of_assignments       non-blocking assignment value was different after ReadOnly(), was 4, now 5
  4.00ns INFO     cocotb.top                                      test.py:30   in test_order_of_assignments       continuous assignment value was different after ReadOnly(), was 0, now 1
  5.00ns INFO     cocotb.top                                      test.py:25   in test_order_of_assignments       non-blocking assignment value was different after ReadOnly(), was 5, now 6
  5.00ns INFO     cocotb.top                                      test.py:30   in test_order_of_assignments       continuous assignment value was different after ReadOnly(), was 0, now 1
  6.00ns INFO     cocotb.top                                      test.py:25   in test_order_of_assignments       non-blocking assignment value was different after ReadOnly(), was 6, now 7
  6.00ns INFO     cocotb.top                                      test.py:30   in test_order_of_assignments       continuous assignment value was different after ReadOnly(), was 0, now 1
  7.00ns INFO     cocotb.top                                      test.py:25   in test_order_of_assignments       non-blocking assignment value was different after ReadOnly(), was 7, now 8
  7.00ns INFO     cocotb.top                                      test.py:30   in test_order_of_assignments       continuous assignment value was different after ReadOnly(), was 0, now 1
  8.00ns INFO     cocotb.top                                      test.py:25   in test_order_of_assignments       non-blocking assignment value was different after ReadOnly(), was 8, now 9
  8.00ns INFO     cocotb.top                                      test.py:30   in test_order_of_assignments       continuous assignment value was different after ReadOnly(), was 0, now 1
  9.00ns INFO     cocotb.top                                      test.py:25   in test_order_of_assignments       non-blocking assignment value was different after ReadOnly(), was 9, now 10
[... cut ...]
  9.00ns INFO     cocotb.regression                         regression.py:209  in handle_result                   Test Passed: test_order_of_assignments
  9.00ns INFO     cocotb.regression                         regression.py:170  in tear_down                       Passed 1 tests (0 skipped)
  9.00ns INFO     cocotb.regression                         regression.py:176  in tear_down                       Shutting down...
make[1]: Leaving directory '/home/jeremy/code/cocotb_test_cases/order_of_assignments'
```

thanks in advance!

stuarthodgson, 2016-06-18:
@jeremyherbert You happy with this after merging?

jeremyherbert, 2016-06-19:
I am. I think there is still perhaps an underlying issue here to do with differences between VPI implementations, but I'm happy to close this as-is. 

I think for this to be investigated any further, there would need to be a CI setup going which has more simulators than just icarus (and given the way software licenses work in this industry, that seems extremely unlikely)

closed 2016-06-19 by jeremyherbert

stuarthodgson, 2016-06-20:
@jeremyherbert The VPI implementations are pretty much the same in this specific regard. VHPI is not the case and there are differences.