cocotb/cocotb #449: Support for timing annotated gate level simulations?

nevsan, 2016-06-07:
I saw in an older presentation for cocotb that testing with simulations that have annotated timing (via SDF, for example) is not supported. Is this still the case? Is there a fundamental reason it is not supported or is it that not enough testing has been done to guarantee it?

I realize that portability across simulators may be an important port of the limitation. Perhaps I should ask more specifically whether this might work with Questa or NCSim.

Thanks!

kirknodeng, 2016-06-07:
I think it is possible. I didn't try that but I establish analog-mixed signal simulation environment with VCS and FineSIM. It is work well. 

You can annotate sdf file at module. (or compile options)

sdf_annotate.v

module sdf_annotate();
initial begin
$sdf_annotate("../best.sdf",tb_top.u_design_top,,"../sdf_annotate_best.log","MINIMUM");
end
endmodule

add with your files in makefile.

brunopen, 2016-11-11:
Hi all,
From this thread, I am a little bit confused about the status of the support of annotated GLS. Is there a clear status about its support?
 
Thanks !

ktbarrett, 2020-05-05:
If anyone in this thread could write up the process they followed to get this to work on the [wiki](https://github.com/cocotb/cocotb/wiki), that would be fantastic! But it seems like the question is resolved, cocotb does work with back-annotated sims.

closed 2020-05-05 by ktbarrett

nevsan, 2020-05-06:
I was able to get this working without issues. There wasn't really any thing to solve related to CocoTB, it's just a matter of getting the right flags to the simulator (same as you would without CocoTB). I agree that this issue can be closed.