cocotb/cocotb #452: ERROR: Unable to find a valid python-config..  Stop.  (OSX 10.11)

tmeissner, 2016-06-15:
On OSX 10.11 I get following error, when trying the endian_swapper example:

```
TMs-Air:tests torsten$ make SIM=ghdl TOPLEVEL_LANG=vhdl
/Users/torsten/Projects/Git/cocotb/makefiles/Makefile.pylib.Darwin:60: *** ERROR: Unable to find a valid python-config..  Stop.
TMs-Air:tests torsten$ which python-config
/usr/bin/python-config
TMs-Air:tests torsten$ which python3-config
/usr/local/bin/python3-config
```

jeremyherbert, 2016-06-19:
does the simulation work with icarus? Or is it ghdl specifc?

jeremyherbert, 2016-06-19:
ok, just tried it on my mac. Same error with icarus. I'll try to have a quick look at this.

jeremyherbert, 2016-06-19:
The problem is that the `PYTHON_BIN` makefile variable has been clobbered at some point. Will submit a pull request shortly to get this fixed, but in the meantime just do

`export PYTHON_BIN=python`

and it should work. On OSX, I also need to set `PYTHONHOME` otherwise it gets confused about whether to use the system python or my anaconda install:

`export PYTHONHOME=/Users/jeremy/anaconda`

Looks like you have both system python and homebrew python, so you may need to softlink your `python3-config` to `python-config` and ensure that it gets picked up before the system python.

tmeissner, 2016-06-19:
I've also seen, that there was no definition of PYTHON_BIN, so I added it. Now, it gets further, but ends when loading the cocotb module after compiling:

```
loading VPI module '/Users/torsten/Projects/Git/cocotb/build/libs/x86_64/libvpi.so'
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:91   in gpi_print_registered_impl       VPI registered
VPI module loaded!
ImportError: No module named cocotb
Failed to load "cocotb"
```

jeremyherbert, 2016-06-19:
Can you please paste the entire output of the `make` command? the `PYTHONPATH` bit before the simulation command is the important part. Mine looks like this for the adder example:

```
make
/Applications/Xcode.app/Contents/Developer/usr/bin/make results.xml
PYTHONPATH=/Users/jeremy/code/cocotb/build/libs/x86_64:/Users/jeremy/code/cocotb:/Users/jeremy/code/cocotb/examples/adder/tests:/Users/jeremy/code/cocotb/examples/adder/tests/../model: LD_LIBRARY_PATH=/Users/jeremy/code/cocotb/build/libs/x86_64::/Users/jeremy/anaconda/lib:/Users/jeremy/anaconda/lib:/Users/jeremy/anaconda/lib:/Users/jeremy/anaconda/lib MODULE=test_adder \
        TESTCASE= TOPLEVEL=adder TOPLEVEL_LANG=verilog \
        /usr/local/bin/vvp -M /Users/jeremy/code/cocotb/build/libs/x86_64 -m gpivpi sim_build/sim.vvp
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:91   in gpi_print_registered_impl       VPI registered
  0.00ns INFO     cocotb.gpi                                  gpi_embed.c:248  in embed_sim_init                  Running on Icarus Verilog version 11.0 (devel)
  0.00ns INFO     cocotb.gpi                                  gpi_embed.c:249  in embed_sim_init                  Python interpreter initialised and cocotb loaded!
...
```

tmeissner, 2016-06-19:
```
TMs-Air:tests torsten$ make SIM=ghdl TOPLEVEL_LANG=vhdl 
/Applications/Xcode.app/Contents/Developer/usr/bin/make results.xml
cd sim_build && /usr/local/bin/ghdl -a /Users/torsten/Projects/Git/cocotb/examples/endian_swapper/tests/../hdl/endian_swapper.vhdl && /usr/local/bin/ghdl -e endian_swapper_vhdl
PYTHONPATH=/Users/torsten/Projects/Git/cocotb/build/libs/x86_64:/Users/torsten/Projects/Git/cocotb:/Users/torsten/Projects/Git/cocotb/examples/endian_swapper/tests:/Users/torsten/Projects/Git/cocotb/examples/endian_swapper/tests/../cosim: LD_LIBRARY_PATH=/Users/torsten/Projects/Git/cocotb/build/libs/x86_64:/Users/torsten/Projects/Git/cocotb/examples/endian_swapper/tests/../cosim:/Users/torsten/Projects/Git/cocotb/examples/endian_swapper/tests/../cosim::/System/Library/Frameworks/Python.framework/Versions/2.7/lib:/System/Library/Frameworks/Python.framework/Versions/2.7/lib:/System/Library/Frameworks/Python.framework/Versions/2.7/lib:/System/Library/Frameworks/Python.framework/Versions/2.7/lib MODULE=test_endian_swapper \
        TESTCASE= TOPLEVEL=endian_swapper_vhdl TOPLEVEL_LANG=vhdl \
	cd sim_build; /usr/local/bin/ghdl -r endian_swapper_vhdl --vpi=/Users/torsten/Projects/Git/cocotb/build/libs/x86_64/libvpi.so
loading VPI module '/Users/torsten/Projects/Git/cocotb/build/libs/x86_64/libvpi.so'
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:91   in gpi_print_registered_impl       VPI registered
VPI module loaded!
ImportError: No module named cocotb
Failed to load "cocotb"
```

jeremyherbert, 2016-06-20:
I'm no bash expert, but I'm pretty sure the problem here is the `cd sim_build;` part in the simulation command, because the ";" will consume the temporary environment variables like `PYTHONPATH`, and thus they won't be passed to the ghdl simulation and thus the embedded python interpreter won't be able to find the library. Cocotb depends heavily on environment variables to hook into the simulator.

I'm not very familiar with ghdl (well, I'm not familiar at all) so I'm not sure why the cd is even necessary, but I can't immediately see why the `cd` is being added after the environment variables; the makefile seems to indicate that the `cd` should be coming before them. There was a change only a few days ago: https://github.com/potentialventures/cocotb/commit/feaa765c3fe62134f28ed0d708ca3ba10080d2ee; are you running the latest git master?

tmeissner, 2016-06-20:
Argh, you are right. I've added the directory change, because in the git version, it did not work. You have to use the -r run command with a binary in the current directory. After moving the directory change to the beginning of the line before the settings of the environment variables, it works :)

jeremyherbert, 2016-06-20:
Great! Happy cocotb-ing!

I guess we'll leave this ticket open for one of the big dogs to close when they have the chance to merge.

chiggs, 2016-07-11:
@jeremyherbert merged - please close if you're happy this has been resolved.

jeremyherbert, 2016-07-12:
@chiggs I don't have the ability (authority?) to close this, but I think it should be closed nonetheless.

tmeissner, 2016-07-12:
As I initially created the issue, I have to close it. Which is what I do now.

closed 2016-07-12 by tmeissner