cocotb/cocotb #460: Error accessing top level port of VHDL DUT trying adder example using GHDL

tmeissner, 2016-06-21:
When trying the VHDL adder example with GHDL, each tests fails when the first access to a top level port occurs.

```
TMs-Air:tests torsten$ make SIM=ghdl TOPLEVEL_LANG=vhdl 
/Applications/Xcode.app/Contents/Developer/usr/bin/make results.xml
cd sim_build && /usr/local/bin/ghdl -a /Users/torsten/Projects/Git/cocotb/examples/adder/tests/../hdl/adder.vhdl && /usr/local/bin/ghdl -e adder
cd sim_build; \
	PYTHONPATH=/Users/torsten/Projects/Git/cocotb/build/libs/x86_64:/Users/torsten/Projects/Git/cocotb:/Users/torsten/Projects/Git/cocotb/examples/adder/tests:/Users/torsten/Projects/Git/cocotb/examples/adder/tests/../model: LD_LIBRARY_PATH=/Users/torsten/Projects/Git/cocotb/build/libs/x86_64::/System/Library/Frameworks/Python.framework/Versions/2.7/lib:/System/Library/Frameworks/Python.framework/Versions/2.7/lib:/System/Library/Frameworks/Python.framework/Versions/2.7/lib:/System/Library/Frameworks/Python.framework/Versions/2.7/lib MODULE=test_adder \
        TESTCASE= TOPLEVEL=adder TOPLEVEL_LANG=vhdl \
	/usr/local/bin/ghdl -r adder --vpi=/Users/torsten/Projects/Git/cocotb/build/libs/x86_64/libvpi.so
loading VPI module '/Users/torsten/Projects/Git/cocotb/build/libs/x86_64/libvpi.so'
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:91   in gpi_print_registered_impl       VPI registered
VPI module loaded!
  0.00ns INFO     cocotb.gpi                                  gpi_embed.c:248  in embed_sim_init                  Running on GHDL version 0.1
  0.00ns INFO     cocotb.gpi                                  gpi_embed.c:249  in embed_sim_init                  Python interpreter initialised and cocotb loaded!
  0.00ns INFO     cocotb                                      __init__.py:115  in _initialise_testbench           Running tests with Cocotb v1.0 from /Users/torsten/Projects/Git/cocotb
  0.00ns INFO     cocotb                                      __init__.py:131  in _initialise_testbench           Seeding Python random module with 1466546523
  0.00ns INFO     cocotb.regression                         regression.py:161  in initialise                      Found test test_adder.adder_basic_test
  0.00ns INFO     cocotb.regression                         regression.py:161  in initialise                      Found test test_adder.adder_randomised_test
  0.00ns INFO     cocotb.regression                         regression.py:262  in execute                         Running test 1/2: adder_basic_test
  0.00ns INFO     ..tine.adder_basic_test.0x101bc7a90       decorators.py:189  in send                            Starting test: "adder_basic_test"
                                                                                                                            Description: Test for 5 + 10
  2.00ns ERROR    ..tine.adder_basic_test.0x101bc7a90           result.py:51   in raise_error                     Send raised exception: adder contains no object named A
                                                                                                                              File "/Users/torsten/Projects/Git/cocotb/cocotb/decorators.py", line 194, in send
                                                                                                                                return self._coro.send(value)
                                                                                                                              File "/Users/torsten/Projects/Git/cocotb/examples/adder/tests/test_adder.py", line 16, in adder_basic_test
                                                                                                                                dut.A = A
                                                                                                                              File "/Users/torsten/Projects/Git/cocotb/cocotb/handle.py", line 239, in __setattr__
                                                                                                                                return getattr(self, name)._setcachedvalue(value)
                                                                                                                              File "/Users/torsten/Projects/Git/cocotb/cocotb/handle.py", line 261, in __getattr__
                                                                                                                                raise AttributeError("%s contains no object named %s" % (self._name, name))
                                                                                                                            
  3.00ns ERROR    cocotb.regression                         regression.py:246  in handle_result                   Test Failed: adder_basic_test (result was TestError)
  3.00ns INFO     cocotb.regression                         regression.py:262  in execute                         Running test 2/2: adder_randomised_test
  3.00ns INFO     ..adder_randomised_test.0x101bc7e90       decorators.py:189  in send                            Starting test: "adder_randomised_test"
                                                                                                                            Description: Test for adding 2 random numbers multiple times
  5.00ns ERROR    ..adder_randomised_test.0x101bc7e90           result.py:51   in raise_error                     Send raised exception: Attempt to access A which isn't present in adder
                                                                                                                              File "/Users/torsten/Projects/Git/cocotb/cocotb/decorators.py", line 194, in send
                                                                                                                                return self._coro.send(value)
                                                                                                                              File "/Users/torsten/Projects/Git/cocotb/examples/adder/tests/test_adder.py", line 37, in adder_randomised_test
                                                                                                                                dut.A = A
                                                                                                                              File "/Users/torsten/Projects/Git/cocotb/cocotb/handle.py", line 243, in __setattr__
                                                                                                                                name, self._name))
                                                                                                                            
  6.00ns ERROR    cocotb.regression                         regression.py:246  in handle_result                   Test Failed: adder_randomised_test (result was TestError)
  6.00ns ERROR    cocotb.regression                         regression.py:167  in tear_down                       Failed 2 out of 2 tests (0 skipped)
  6.00ns INFO     cocotb.regression                         regression.py:176  in tear_down                       Shutting down...

```

Running the example with icarus verilog works as expected. Maybe this is an GHDL specific error? The endian_swapper example works with both, icarus and GHDL.

tmeissner, 2016-06-22:
When I'm printing the attributes of the dut object, I an empty _sub_handles attribute:

`'_sub_handles': {}`

For endian_swapper, I get all DUT ports as ModifiableObjects in the _sub_handles attribute:

`'_sub_handles': {'stream_out_empty': ModifiableObject(endian_swapper_vhdl.stream_out_empty), 'stream_out_valid': ModifiableObject(endian_swapper_vhdl.stream_out_valid), 'csr_readdatavalid': ModifiableObject(endian_swapper_vhdl.csr_readdatavalid), 'stream_in_data': ModifiableObject(endian_swapper_vhdl.stream_in_data), 'csr_write': ModifiableObject(endian_swapper_vhdl.csr_write), 'csr_read': ModifiableObject(endian_swapper_vhdl.csr_read), 'stream_in_startofpacket': ModifiableObject(endian_swapper_vhdl.stream_in_startofpacket), 'clk': ModifiableObject(endian_swapper_vhdl.clk), 'stream_in_valid': ModifiableObject(endian_swapper_vhdl.stream_in_valid), 'stream_in_empty': ModifiableObject(endian_swapper_vhdl.stream_in_empty), 'csr_address': ModifiableObject(endian_swapper_vhdl.csr_address), 'csr_readdata': ModifiableObject(endian_swapper_vhdl.csr_readdata), 'stream_in_ready': ModifiableObject(endian_swapper_vhdl.stream_in_ready), 'stream_out_data': ModifiableObject(endian_swapper_vhdl.stream_out_data), 'stream_out_startofpacket': ModifiableObject(endian_swapper_vhdl.stream_out_startofpacket), 'csr_waitrequest': ModifiableObject(endian_swapper_vhdl.csr_waitrequest), 'stream_out_endofpacket': ModifiableObject(endian_swapper_vhdl.stream_out_endofpacket), 'stream_in_endofpacket': ModifiableObject(endian_swapper_vhdl.stream_in_endofpacket), 'byteswapping': ModifiableObject(endian_swapper_vhdl.byteswapping), 'stream_out_ready': ModifiableObject(endian_swapper_vhdl.stream_out_ready), 'reset_n': ModifiableObject(endian_swapper_vhdl.reset_n), 'csr_writedata': ModifiableObject(endian_swapper_vhdl.csr_writedata)}`

JarrettR, 2016-06-22:
Not sure if this is a bug in GHDL or in cocotb, but uppercase signal names in the Python testbench do not work.

If you change the signals A, B, and X to lowercase, this example will work, even if the signals are still uppercase in the HDL.

JarrettR, 2016-06-22:
Also tagging @tgingold

tmeissner, 2016-06-22:
Yeah, you are right :-) With lowercase port names in the test bench, it works with GHDL. But not with icarus. So, you can't use the same testbench for icarus & GHDL at the moment. For me, the problem seems to be like on the GHDL side, isn't it?

tgingold, 2016-06-22:
Yes, using a case insensitive string compare fixes the issue.

tmeissner, 2016-06-25:
I think we can close this issue.

closed 2016-06-25 by tmeissner