cocotb/cocotb #466: excessing memory consumption with Modelsim/Questa

LinuxFanatic777, 2016-06-24:
excessing memory consumption with Modelsim/Questa

Hello!
I have a large module that being tested with help of Cocotb+Modelsim/Questa on Ubuntu Linux 14.04.4 64-bit. After 20 minutes of simulation it runs out of system memory, so it can't run any longer. Also I found that it produce large vsim.wlf file of size of many gigs (how to turn this file off???).

Is there any way to determine the reason of such memory consumption? I'm using just few waves to output in GUI, what is eating so much? I suppose Python/Cocotb part allocating something and not releasing. Or may be Questa trying to capture state of whole module at every simulation tick - how to turn this off?

marcinwoj, 2016-06-27:
Hi, 

Not sure whether it helps but you can have a look at +acc=rmb flags. If you use $dumpvars you can also try to restrict levels and/or dump only specific modules. 

Marcin

LinuxFanatic777, 2016-06-27:
Thank you for reply, I'll try what you suggested.

LinuxFanatic777, 2016-07-29:
Thank you again :) That helped me! I just removed "+acc=rmb" and "echo "add wave -recursive /*" >> $@" in Makefile.questa and this reduced memory consumption in many times.
```
 ifeq ($(GUI),1)
-       echo "add wave -recursive /*" >> $@
+       echo "$$MY_WAVES" >> $@
```
And then in makefile of my project (for example):
```
define MY_WAVES
add wave -color White -radix unsigned /mymod/clk
add wave -color White -radix unsigned /mymod/output
endef
export MY_WAVES
```
Thus only required waves come to output.
It's funny, but console mode not really helps to reduce memory.
Problem solved and now I can simulate my project, it takes 12 Gb of ram. Before I did these optimisation steps I was unable to finish simulation, even on server with 128 Gb of memory :)

LinuxFanatic777, 2016-10-11:
Also interesting note: use **xrange** instead of **range** for Python 2.x - that helped me reduce memory consumption from 5 Gb RAM to ~200 Mb.

LinuxFanatic777, 2017-10-27:
Hello! I'm using latest Cocotb with Python 3.5.2 and Modelsim. With one of my large projects I found that it consume 10 megabytes per second and my system runs out of memory just for 10 minutes, while I need to simulate for many days on server.
To find out source of this problem I created pure-Verilog+Modelsim environment for same DUT. Without Cocotb it does not consume more than 1 megabyte for 10 minutes - great, but I want to use Cocotb.

How to debug this? I found variable MEMCHECK - how to use this? Is this debugging technique general for Python platform or Cocotb-related only?

AlexanderSpirin, 2017-10-27:
- One memory leak source fixed #536 (and there you can find comment about MEMCHECK). You should already have this patch as you use latest version.
- Another reason of memory leak is special define for Modelsim:https://github.com/potentialventures/cocotb/blob/f2ea9cb6a6f365f9b0950ec2aef35293ac806306/lib/vpi/VpiCbHdl.cpp#L107-L112
I have a feeling it can be removed, anyway I've removed it and not encountered with any errors and problems.
- Another problem: https://github.com/potentialventures/cocotb/blob/f2ea9cb6a6f365f9b0950ec2aef35293ac806306/cocotb/triggers.py#L373
I've tried to rewrite trigger code but there was some difficulties in it and I quickly lost enthusiasm.

Eventually after I applied the first two changes memory consumption growth became acceptable for my task, but it still exists.

LinuxFanatic777, 2017-10-30:
I'm using cocotb + Python3, finally got working dowser-py3, here is my log: http://paste.org.ru/?01fnqd
```
builtins.dict
Min: 3605 Cur: 16398 Max: 16567 TRACE

builtins.function
Min: 7041 Cur: 35083 Max: 35083 TRACE
```
Please, look at this link, could you find anything strange here? It eats 10 MBytes of RAM per second. Without Cocotb in pure System Verilog test memory leak is not present and memory almost not consuming.

AlexanderSpirin, 2017-11-01:
In my test memory growth is about 1 MB/s and I see count of objects in cocotb.log.SimLog, cocotb.triggers.Timer, cocotb.triggers._Event are constantly increasing. But there is no such thing in your log.
Can you post log of individual instances in builtins.dict and builtins.function (Dowser: Trace output), this should give some clue to the problem.
And I presume you have made changes in VpiCbHdl.cpp?

LinuxFanatic777, 2017-11-01:
No, it was without VpiCbHdl since I use latest repo. Here is new results with VpiCbHdl fixed - same results same memory hunger: http://paste.org.ru/?u9qe70 After a few seconds: http://paste.org.ru/?nc8yhp
Large traces for some variables: http://paste.org.ru/?enj451 http://paste.org.ru/?m822b3 http://paste.org.ru/?5rio1d http://paste.org.ru/?0wyddd

AlexanderSpirin, 2017-11-01:
All traces consists of quite diverse objects, so it's difficult to find reason and this differ from my case. I can only suggest to make diff of two traces at different time and find what new objects are created recently. But maybe better to use interactive debugging in IDE and some tool to analyze memory (I used [Heapy](https://stackoverflow.com/questions/110259/which-python-memory-profiler-is-recommended))
I think the main reason of leak is outside cocotb since cocotb.* objects increase is negligible. Another idea is reducing complexity of test, because now it uses a lot of libraries, and compare memory consumption.

LinuxFanatic777, 2017-11-17:
Tried a very simple example consisting of minimal binary counter without any libraries and Python modules. I'm shocked, but memory consumption remains the same: 10 MBytes/s, heh...
IcarusVerilog also suffers from memory leak, but at a lower rate of 1-2 MBytes/s.
As I noticed before, this problem not apparent when using pure-Verilog testbenches!

AlexanderSpirin, 2017-11-17:
@LinuxFanatic777 I've tried to increase number of packets in adder and endian_swapper examples with IcarusVerilog, and there is no memory consumption growth at all.
Please provide minimal working example for debug and will look at it.

LinuxFanatic777, 2017-11-17:
Here is my minimal example, where memory usage grows dramatically fast.
May be I do something completely wrong and insane? :)
My OS Ubuntu 14.04.5 64-bit, Modelsim 64-bit too, Anaconda3-4.2.0-Linux-x86_64

```
import cocotb
from cocotb.triggers import Timer as ps
from cocotb.triggers import RisingEdge as posedge
from cocotb.result import TestFailure as failed
import os

@cocotb.test()
def ctbml(dut):
	dut.log.info("ctbml start")
	cocotb.fork(clock_100mhz(dut.clock))

	dut.reset = 1
	yield posedge(dut.clock)
	dut.reset = 0
	yield posedge(dut.clock)

	while(int(dut.test) < 1000000):
		yield posedge(dut.clock)
	dut.log.info("ctbml finished")

@cocotb.coroutine
def clock_100mhz(signal):
	period = 10000
	while True:
		signal <= 1
		yield ps(period / 2, units = "ps")
		signal <= 0
		yield ps(period / 2, units = "ps")
```
```
`timescale 1ns / 1ns

module top_ctbml(
input clock,
input reset,
output reg [23:0] test
);

initial test = 0;

always @(posedge clock)
if(reset) test <= 0;
else test <= test + 1;

endmodule
```
```
VERILOG_SOURCES = \
$(SOURCE)/top_ctbml.v

MODULE = ctbml
TOPLEVEL = top_ctbml
TOPLEVEL_LANG = verilog
include $(COCOTB)/makefiles/Makefile.inc
include $(COCOTB)/makefiles/Makefile.sim
```

AlexanderSpirin, 2017-11-17:
```bash
/opt/cocotb/examples/mem on  master! ⌚ 14:43:30
$ ps aux | grep vvp
adun      4889  0.0  0.0   5360   784 pts/29   S+   14:43   0:00 /bin/sh -c PYTHONPATH=/opt/cocotb/build/libs/x86_64:/opt/cocotb:/opt/cocotb/examples/mem: LD_LIBRARY_PATH=/opt/cocotb/build/libs/x86_64::/usr/lib:/usr/lib:/usr/lib:/usr/lib MODULE=ctbml \         TESTCASE= TOPLEVEL=top_ctbml TOPLEVEL_LANG=verilog \         /usr/local/bin/vvp -M /opt/cocotb/build/libs/x86_64 -m gpivpi sim_build/sim.vvp   
adun      4890 90.6  0.0  97344 19532 pts/29   R+   14:43   0:04 /usr/local/bin/vvp -M /opt/cocotb/build/libs/x86_64 -m gpivpi sim_build/sim.vvp
```
```bash
/opt/cocotb/examples/mem on  master! ⌚ 14:44:28
$ ps aux | grep vvp
adun      4889  0.0  0.0   5360   784 pts/29   S+   14:43   0:00 /bin/sh -c PYTHONPATH=/opt/cocotb/build/libs/x86_64:/opt/cocotb:/opt/cocotb/examples/mem: LD_LIBRARY_PATH=/opt/cocotb/build/libs/x86_64::/usr/lib:/usr/lib:/usr/lib:/usr/lib MODULE=ctbml \         TESTCASE= TOPLEVEL=top_ctbml TOPLEVEL_LANG=verilog \         /usr/local/bin/vvp -M /opt/cocotb/build/libs/x86_64 -m gpivpi sim_build/sim.vvp   
adun      4890 98.8  0.0  97600 19760 pts/29   R+   14:43   1:01 /usr/local/bin/vvp -M /opt/cocotb/build/libs/x86_64 -m gpivpi sim_build/sim.vvp
```
```
Ubuntu 14.04.5 LTS
Python 3.4.3
Icarus Verilog version 11.0 (devel) (s20150603-474-g62e0f59b)
```
I have no idea what is the problem in your case. I can only suggest trying to run test on another system.

LinuxFanatic777, 2017-11-17:
I see you do not have changes in vvp used. Strange. Also I ran my tests on server with Debian 8.5 with same negative result. Will try test at home.

LinuxFanatic777, 2017-11-17:
Also I ran my tests on server with Debian 8.5 with same negative result.
Tried to measure with your method (same top_ctbml.v test). I found that **there is no memory leak** in case of IcarusVerilog, ps aux shows me constant memory usage, sorry - it's my error and disinformation. It seems that other applications consumed memory, not Icarus+Cocotb.

Here is memory measures for Modelsim ASE (free 32-bit):
```
user@user:~$ ps aux | grep vsimk ; date
user     28191  100  1.5 166984 121712 pts/6   R+   15:58   0:33 /home/user/t/intelFPGA/16.1/modelsim_ase/linuxaloem/vsimk -port 37246 -stdoutfilename /tmp/VSOUT28sezL -c -do runsim.do
user     28201  0.0  0.0   8880   804 pts/10   S+   15:58   0:00 grep --color=auto vsimk
Fri Nov 17 15:58:53 MSK 2017
user@user:~$ 
user@user:~$ ps aux | grep vsimk ; date
user     28191  101  1.9 199752 154552 pts/6   R+   15:58   0:46 /home/user/t/intelFPGA/16.1/modelsim_ase/linuxaloem/vsimk -port 37246 -stdoutfilename /tmp/VSOUT28sezL -c -do runsim.do
user     28205  0.0  0.0   8880   804 pts/10   S+   15:59   0:00 grep --color=auto vsimk
Fri Nov 17 15:59:07 MSK 2017
user@user:~$ 
user@user:~$ ps aux | grep vsimk ; date
user     28191  100  2.6 248904 211740 pts/6   R+   15:58   1:09 /home/user/t/intelFPGA/16.1/modelsim_ase/linuxaloem/vsimk -port 37246 -stdoutfilename /tmp/VSOUT28sezL -c -do runsim.do
user     28209  0.0  0.0   8880   800 pts/10   S+   15:59   0:00 grep --color=auto vsimk
Fri Nov 17 15:59:30 MSK 2017
user@user:~$ 
user@user:~$ ps aux | grep vsimk ; date
user     28191  100  3.3 314440 271276 pts/6   R+   15:58   1:33 /home/user/t/intelFPGA/16.1/modelsim_ase/linuxaloem/vsimk -port 37246 -stdoutfilename /tmp/VSOUT28sezL -c -do runsim.do
user     28212  0.0  0.0   8880   784 pts/10   S+   15:59   0:00 grep --color=auto vsimk
Fri Nov 17 15:59:53 MSK 2017
user@user:~$ 
user@user:~$ ps aux | grep vsimk ; date
user     28191  100  3.8 347208 311816 pts/6   R+   15:58   1:49 /home/user/t/intelFPGA/16.1/modelsim_ase/linuxaloem/vsimk -port 37246 -stdoutfilename /tmp/VSOUT28sezL -c -do runsim.do
Fri Nov 17 16:00:10 MSK 2017
```
When using 32-bit Modelsim (free version) there are still memory leak but with speed of 5 MBytes/s. Strange correlation: less bits -> twice less leakage speed.

LinuxFanatic777, 2017-11-21:
I tried to fix vco file to add Valgrind memory debugger, but vsim (Modelsim) is crashing under debugger. May be it's still possible to debug memory leaks of invpi.so library with help of some other tools? MCheck for example.

LinuxFanatic777, 2017-11-27:
I tried two more things:

1) /path/to/modelsim/vco
```
MALLOCDEBUG=/home/user/mtrace.out /path/to/modelsim/linux_x86_64/vsim $@
```
or
```
MALLOC_TRACE=/home/user/mtrace.out /path/to/modelsim/linux_x86_64/vsim $@
```
None of these two variants produces output, but I'm sure it starts through these lines (wrapped by echoes).

2) Debug Python-side with help of tracemalloc, as a result - there is no places that consume memory a lot.

Still suspect Cocotb's native libraries, or may be Modelsim itself behave wrong in such scenario with VPI interface? How to debug this place? Please, help. What else can I do? I do not want to stop using Cocotb.

imphil, 2023-01-23:
Closing as outdated.

closed 2023-01-23 by imphil

yakovlevyury, 2023-07-13:
It looks like this issue is still relevant, maybe reopen it?

ktbarrett, 2023-07-13:
@yakovlevyury Relevant in what way? Provide details in a new issue.