cocotb/cocotb #472: Add support to Drive and Force signals

leftink, 2016-07-05:
With the current approach of "depositing" data into signals, the value is placed in the signal and remains unless changed or something triggers the resolve() function to be invoked.  This can make testing a bi-directional bus complicated because of the fact that we are not actually driving the signal the resolve function isn't invoked and so cases where both sides of the bus are driving at the same time, an error resulting on an 'X', on the bus, wouldn't be seen.

Further, we lack the ability to navigate down into a design and take control of a component since every time a value would be deposited it would most likely be overwritten by a driver in the module.

The FLI, VHPI, and VPI all support three ways to "set" a value:  deposit, drive, freeze/unfreeze.

The python interface to support this would possibly look like:
```python
dut.sig = 5   # Current approach, uses "default" method
dut.sig = (5, DRIVE)   # Tuple is used to determine method for setting the value
dut.sig = DRIVE(5)   # Wrapper function to create the tuple
```
Basically, there would be constants for DEPOSIT, DRIVE, UNDRIVE?, FREEZE, UNFREEZE.

The default method would be DEPOSIT, in the python, but should support being overridden through an environment variable.

eric-wieser, 2020-04-09:
Closed via #657, I think

closed 2020-04-09 by eric-wieser

ktbarrett, 2020-04-09:
@eric-wieser `Drive` does not have support yet. It can be supported on Questa using the FLI. I can file another issue specifically about `Drive` if you want to keep this closed.

eric-wieser, 2020-04-09:
Drive is different to Force and Deposit?

eric-wieser, 2020-04-09:
Perhaps best for you to open a new issue, since you have the background to know what's missing.