cocotb/cocotb #477: Instructions for non-default install location

hutch31, 2016-07-26:
Please provide instructions for installing cocotb to a non-default location (other than /usr/local).

For my evaluation I created a Python virtual environment (cocotest) and installed cocotb under it:

cocotest/cocotb-1.0

I created a startup script:
# Activate virtual environment
coco/bin/activate
# Add Cocotb modules to path
export PYTHONPATH=`pwd`/cocotb-1.0

This finds the "cocotb" module but not any of its sub-modules:
>>> import cocotb
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/guy/devel/cocotest/cocotb-1.0/cocotb/__init__.py", line 41, in <module>
    import cocotb.handle
  File "/Users/guy/devel/cocotest/cocotb-1.0/cocotb/handle.py", line 45, in <module>
    import simulator

leftink, 2016-07-27:
@hutch31 it is finding the submodules. Your issue is it isn't finding the simulator module which is only present after building the C libraries.  The library location must also be part of the python path since the simulator module is actually in a shared object.

hutch31, 2016-07-27:
Is there another step required to build the simulator module?  After I do a make install I don't see any shared libs present in the install directory.

leftink, 2016-07-28:
Assuming you have built the shared libraries, i.e. cocotb.so, simulator.so, etc., then I believe your only issue is with the export of PYTHONPATH.  The export you have only configures it to find the python files in the cocotb namespace.  You also need to have a path to the build libraries as well.  simulator.so is actually a python extension module and must be in the PYTHONPATH for `import simulator` to work properly.

If you are using the cocotb make system for running tests then all of that is setup for you.

closed 2016-08-08 by hutch31