cocotb/cocotb #480: Propagate Exceptions through the coroutine chain

leftink, 2016-07-28:
Currently, if a coroutine raises an exception an error is created and the test is over.  There currently is no chance for the "parent" coroutine to process the exception.  Ideally it would be nice to support something like the following:

'''python
@cocotb.coroutine
def err():
    yield NullTrigger()
    raise ValueError("Example of raising an error")

@cocotb.coroutine
def catch_err():
    try:
        yield err()
    except ValueError:
        print("Handle the error")
'''

The mechanism is already supported through the "throw()" function of the generator base coroutines.  It would just require modifications to the scheduler to "re-throw" the exception into the calling coroutine.  If at the top of the call stack then use raise_error().

@stuarthodgson and @chiggs  Was there a reason for not having this capability, i.e. not supported in certain versions of python, or never saw a need?  Is this functionality that you would be ok with being added?

chiggs, 2016-07-28:
Sounds like a good idea - I have a feeling there is a slight complication with `throw()` but it's so long ago I'm afraid I can't remember any concrete details.  Definitely worth seeing if this is feasible now though.

As an aside, I had a prototype of a better debug reporting mechanism for when things go wrong with coroutines which may be relevant - I'll try and push that to a branch in case it's useful.

leftink, 2016-07-29:
Sounds good.

I did a very basic prototype of this in python 3.5.1 and it worked.  The one caveat was you couldn't catch the exception and throw it into the other coroutine, you had to 're-construct' it to use the throw() feature.  I hadn't tried to see what happens if the coroutine doesn't catch the exception that is thrown.

TL-SjoerdOptLand, 2018-02-19:
Any news on this? Did the prototype make it to the master? Is there anything I could do to help this issue forward?

eric-wieser, 2018-08-01:
Looks like I re-reported this at #632

eric-wieser, 2018-08-08:
@TL-SjoerdOptLand: See #640 for a working implementation

closed 2019-01-28 by imphil