cocotb/cocotb #481: Debugging function and external decorators in Aldec Active-HDL

diarmuidcwc, 2016-08-03:
I've been using cocotb for the past 6 months and have been progressing well with it. Generally learning it by following the examples and the unit tests in conjunction with the source

However I've come up against an issue when I'm trying to get the function and external decorators to work. 

My environment
```
Windows 7 
Python 2.7.11
cocotb commit 522d74417
Aldec Active-HDL10.3
```

I have a very simple testcase to understand externals and functions
```python
import cocotb
from cocotb.triggers import ( Timer)
from cocotb.result import ReturnValue
from cocotb.decorators import external
import logging

@cocotb.coroutine
def wait_us(useconds=1000):
    yield Timer(useconds*1000*1000)

@cocotb.function
def send_a_packet(dut):
    dut.log.info("In function send_a_packet")
    yield wait_us(1)
    dut.log.info("Returning from send_a_packet")
    raise ReturnValue(1)

def test_ext_function(dut):
    dut.log.info("external")
    return 2

@cocotb.test()
def test_ext_function(dut):

    yield wait_us(10)
    dut.log.setLevel(logging.DEBUG)

    dut.log.info("Kicking function....")
    send_a_packet(dut)
    #value = yield external(test_ext_function)(dut)
    value = 0
    dut.log.info("Return = {}".format(value))

```

The function or external never seems to execute, In my debug I've followed tried to debug the issue.
Here's some debug output
```
run
  0.00ns INFO     cocotb.gpi                                  gpi_embed.c:248  in embed_sim_init                  Running on Aldec HDL Simulator version 10.03.3558
  0.00ns INFO     cocotb.gpi                                  gpi_embed.c:249  in embed_sim_init                  Python interpreter initialised and cocotb loaded!
  0.00ns INFO     cocotb                                      __init__.py:112  in _initialise_testbench           Unable to determine Cocotb version from Unknown
  0.00ns INFO     cocotb                                      __init__.py:131  in _initialise_testbench           Seeding Python random module with 1470210821
  0.00ns INFO     cocotb.regression                         regression.py:161  in initialise                      Found test testcase.test_ext_function
  0.00ns INFO     cocotb.regression                         regression.py:262  in execute                         ←[44m←[30mRunning test 1/1:←[39m←[49m test_ext_function
  0.00ns DEBUG    cocotb.scheduler                           scheduler.py:410  in add                             Starting to add new coroutine test_ext_function
  0.00ns DEBUG    cocotb.scheduler                           scheduler.py:435  in add                             Adding new coroutine test_ext_function
  0.00ns DEBUG    cocotb.scheduler                           scheduler.py:455  in schedule                        Scheduler called test_ext_function
  0.00ns INFO     ..tine.test_ext_function.0x36095990       decorators.py:189  in send                            Starting test: "test_ext_function"
                                                                                                                  Description: None
  0.00ns DEBUG    cocotb.scheduler                           scheduler.py:470  in schedule                        Coroutine test_ext_function yielded wait_us (mode 1)
  0.00ns DEBUG    cocotb.scheduler                           scheduler.py:495  in schedule                        Scheduling nested co-routine: wait_us
  0.00ns DEBUG    cocotb.scheduler                           scheduler.py:399  in queue                           Appending wait_us
  0.00ns DEBUG    cocotb.scheduler                           scheduler.py:410  in add                             Starting to add new coroutine wait_us
  0.00ns DEBUG    cocotb.scheduler                           scheduler.py:435  in add                             Adding new coroutine wait_us
  0.00ns DEBUG    cocotb.scheduler                           scheduler.py:455  in schedule                        Scheduler called wait_us
[... 28 lines cut ...]
KERNEL: Time: 10000001 ps,  Iteration: 0,  TOP instance.
KERNEL: stopped at time: 10000001 ps
VSIM: Simulation has finished. There are no more test vectors to simulate.
coverage write
VSIM: Warning: Code Coverage is turned off.
endsim
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:91   in gpi_print_registered_impl       VHPI registered
VSIM: Simulation has finished.
```

**Important Note**
I've modified decorators.py to add a timeout to the wait on event. If I don't the sim hangs forever, This is the issues I am debugging. I did this so that I could see the rest of the python behaviour
```python
coro = cocotb.scheduler.queue(execute_function(self, self._event))
timeout = self._event.wait(15.0)'
```
From following the code through, it seems that the decorator is doing its job and adding the function to the scheduler queue.
In the scheduler it is appended to self._pending_coros.
However the self.add method is never called on these functions. Other scheduled coroutines are added, which can be seen in the log output however the coroutine in the function is never added.

I'm having trouble understanding what triggers the self.add method?

Maybe someone who understands this aspect of the code will point me in the right direction. Either my simulator is having issues with something here (although everything else I've tried works so far) or I'm doing something wrong.

Any comments, hints or tips would be welcome?

leftink, 2016-08-08:
I believe your issue is the function has to be run on a different thread.  If you invoke the function on the main thread of execution then you are actually blocking the scheduler from running.

diarmuidcwc, 2016-08-09:
Thanks for the reply. Do you have an example of such a usage? I tried to follow the hal example but I guess unsuccessfully...

marlonjames, 2018-08-24:
I think the issue is that the @cocotb.function decorator is for functions that are called FROM the external context.
So you use @cocotb.external to call some other code in another thread, then when the code in that other thread needs to call back into cocotb, it can call any @cocotb.function decorated function.

So you'd want to call send_a_packet(dut) from test_ext_function().

closed 2026-04-08 by ktbarrett