cocotb/cocotb #482: Use -f file for verilog sources

hutch31, 2016-08-08:
Our existing projects are built around file list files which are used for both simulation and synthesis.  Is there a way to incorporate these into the makefile?  Adding "-f mylist.f" results in an error message that it's unable to find file '-f'.

kirknodeng, 2016-11-02:
If you insert little statements in your makefiles, It is possible to get rtl lists.
ex)
In your Makefiles,
We already have VERILOG_SOURCES & INCDIRS variables.
export them.

Simply, 
$(shell echo ${INCDIRS} > rtl_list_${TOPLEVEL}.f)
$(shell echo ${VERILOG_SOURCES} >> rtl_list_${TOPLEVEL}.f)

I think that is a lot of better ways to have rtl lists.

cmarqu, 2019-10-14:
This can be handled by setting ``EXTRA_ARGS`` (see https://cocotb.readthedocs.io/en/latest/building.html#make-variables).

imphil, 2019-10-15:
Yes, EXTRA_ARGS is the way to go here. Please reopen if that doesn't work for you.

closed 2019-10-15 by imphil