cocotb/cocotb #486: Python 3 Compatible?

jiapei100, 2016-09-12:
It looks like cocotb is NOT compatible with Python 3 yet?

> RefactoringTool: Warnings/messages while refactoring:
> RefactoringTool: ### In file ./cocotb/generators/bit.py ###
> RefactoringTool: Line 67: You should use 'operator.mul(1)' here.
> RefactoringTool: There was 1 error:
> RefactoringTool: Can't parse ./tests/test_cases/test_exit_error/test_exit.py: ParseError: bad input: type=1, value='a', context=('', (14, 20))

Cheers
Pei

jeremyherbert, 2016-09-12:
Hi Pei,

I'm not exactly sure what the issue is, but I run it with Python 3.5. And the Travis CI build tests for python3 support.

Perhaps there is a bug in your RefactoringTool software?

LinuxFanatic777, 2017-05-16:
Cocotb is compatible at least with Python 3.4 and 3.5 (on Linux)

eric-wieser, 2019-01-13:
@imphil: Pretty sure this can be closed

imphil, 2019-01-13:
Yes, these days cocotb is Python 3 compatible and tested there.

closed 2019-01-13 by imphil