cocotb/cocotb #488: Target 'regression' always re-runs simulation

AlexanderSpirin, 2016-09-15:
I suggest this changes:

1. $(SIM_BUILD) should be  order only prerequisite in all simulators/Makefile.*
2. Target results.xml should be changed to $(SIM_BUILD)/results.xml
3. Add $(MODULE).py to $(SIM_BUILD)/results.xml prerequisites

ktbarrett, 2020-05-05:
I don't see how this is a bad thing? We *want* to run *all* of the regressions every time we start them up, otherwise they aren't regressions.

cmarqu, 2020-05-05:
According to https://docs.cocotb.org/en/latest/building.html#make-targets, the ``sim`` target is the one that is supposed to unconditionally run all simulations.

ktbarrett, 2020-05-05:
@themperek You know more about the makefiles, are two targets for running sims right now? One for running only if the test modules change and the other for running every time?

eric-wieser, 2020-05-06:
> are two targets for running sims right now?

`make results.xml` and `make sim`

ktbarrett, 2020-05-06:
Well the requested behavior is not already available using `results.xml` as a target, since it does not re-run if test modules change.

eric-wieser, 2020-05-06:
> since it does not re-run if test modules change.

It does as long as you remember to add your test modules to `CUSTOM_SIM_DEPS`.

eric-wieser, 2020-05-06:
I think the point about using an order-only prequisite is a valid one.

Using `$(MODULE).py` as a prequisite fails for:

1. Modules on `sys.path` (like `cocotb.ipython_support`)
2. Dependencies of `$(MODULE).py`
3. Comma-separated `MODULE`

3 is trivial, 2 is impossible and probably not worthwhile, and 1 is necessary to avoid breakage of the ipython example.

We could solve 1 by doing something clever with `importlib`.

ktbarrett, 2020-05-06:
So OP's point 1 is valid and actionable; point 2 is up-for-debate; and point 3 seems to be a matter of documentation of `CUSTOM_SIM_DEPS` while discussing the [make targets](https://docs.cocotb.org/en/latest/building.html#make-targets)? Otherwise supporting `MODULE` as a dependency seems unreasonable/out-of-scope based on your description.

eric-wieser, 2020-05-30:
> 1. $(SIM_BUILD) should be  order only prerequisite in all simulators/Makefile.*

Fixed in gh-1876

ktbarrett, 2025-10-16:
No more movement, we already support the functionality, and Makefiles are on their way out.

closed 2025-10-16 by ktbarrett