cocotb/cocotb #489: ImportError: No module named site

jfosorio, 2016-09-18:
I am trying to "build" cocotb using anaconda python (3.5) and i get the error described below. It basically seams that I can not import the module *site*. What is weird as if I type:
```
python -c "import site" 
```

Following a anaconda installation I followed the instructions  #244 and defined the variable:

```
export PYTHONHOME=/Users/felipe/anaconda3
```

## System
- OSX
- anaconda 3.5

## Error message
```
cd test_cases/test_plusargs && /Applications/Xcode.app/Contents/Developer/usr/bin/make
/Applications/Xcode.app/Contents/Developer/usr/bin/make results.xml
PYTHONPATH=/Users/felipe/test/cocotb/build/libs/x86_64:/Users/felipe/test/cocotb:/Users/felipe/test/cocotb/tests/test_cases/test_plusargs: LD_LIBRARY_PATH=/Users/felipe/test/cocotb/build/libs/x86_64::/Users/felipe/anaconda3/lib:/Users/felipe/anaconda3/lib:/Users/felipe/anaconda3/lib:/Users/felipe/anaconda3/lib MODULE=plusargs \
        TESTCASE= TOPLEVEL=tb_top TOPLEVEL_LANG=verilog \
        /Users/felipe/anaconda3/bin/vvp -M /Users/felipe/test/cocotb/build/libs/x86_64 -m gpivpi sim_build/sim.vvp   +foo=bar +test1 +test2 +options=fubar
ImportError: No module named site
make[3]: *** [results.xml] Error 1
make[2]: *** [sim] Error 2
make[2]: Target `all' not remade because of errors.
make[1]: *** [test_plusargs] Error 2
cd test_cases/test_verilog_access && /Applications/Xcode.app/Contents/Developer/usr/bin/make
/Applications/Xcode.app/Contents/Developer/usr/bin/make results.xml
Skipping simulation as VHDL is not supported on simulator=icarus
cd test_cases/test_vhdl_access && /Applications/Xcode.app/Contents/Developer/usr/bin/make
/Applications/Xcode.app/Contents/Developer/usr/bin/make results.xml
Skipping simulation as VHDL is not supported on simulator=icarus
make[1]: Target `all' not remade because of errors.
make: *** [do_tests] Error 2
```

leftink, 2016-09-19:
You are using Anaconda, do you have PYTHONHOME set in your environment?  If not try setting it to "/Users/Felipe/anaconda3"

jfosorio, 2016-09-19:
Yes I did that. Before this error message I had a different one and following the instructions of  issue #244 I set PYTHONHOME precisely to this value and still get the error mention before.

leftink, 2016-09-22:
I have hit that issue before.  For me it occurs when I have both Python2 and Python3 images installed on my machine.   My issue had to do with the search order for the python-config file, i.e. it was finding python-config which was for Python2 and my python binary was running Python 3.

I put a fix into Makefile.pylib.Linux to address this issue, i.e. looks for a python config binary that matches the python executable version.  I don't know think this fix made it into the other Makefile.pylib.*.  You might try updating to the latest and updating the makefile corresponding to your OS to match the one for Linux.

themperek, 2020-04-16:
I believe this should be solved now.

Since this is an issue from 2016 I will close but feel free to reopen if the problem persists.

closed 2020-04-16 by themperek