cocotb/cocotb #492: Linux Mint 18 + Riviera Issues

saturn77, 2016-10-05:
I  recently did a new build with Linux Mint Cinammon 18 and installed cocotb. 

Running tests with iverilog worked fine. However, when running the adder test with "make SIM=aldec" I get the following errors, where the "dlopen" fails to open. 

Wondering if anyone has a good idea on the most effective way to resolve this. 

Thanks. James

KERNEL: PLI/VHPI kernel's engine initialization done.
# PLI: Loading library 'libvpi.so'
# PLI: Function "dlopen" failed with error "/home/james/Aldec/Riviera-PRO-2016.06-x64/bin/Linux64/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /home/james/repos/cocotb/build/libs/x86_64/libvpi.so)" when attempting to open "libvpi.so" library.
# PLI:  Cannot load the "libvpi.so" library. The library does not exist or is corrupted.
# ELAB2: Warning: ELAB2_0113 Task or function "$dumpfile" not defined in module "adder" from library "work".
# ELAB2: Warning: ELAB2_0113 Task or function "$dumpvars" not defined in module "adder" from library "work".
# ELAB2: Create instances ...
# ELAB2: Create instances complete.
# ELAB2: Elaboration final pass complete - time: 0.0 [s].
# KERNEL: Kernel process initialization done.
# Allocation: Simulator allocated 6395 kB (elbread=1450 elab2=4809 kernel=134 sdf=0)
# KERNEL: ASDB file was created in location /home/james/repos/cocotb/examples/adder/tests/sim_build/dataset.asdb
run -all
# KERNEL: Simulation has finished. There are no more test vectors to simulate.
endsim
# VSIM: Simulation has finished.
make[1]: Leaving directory '/home/james/repos/cocotb/examples/adder/tests'

jeremyherbert, 2016-10-05:
Hi Saturn,

I'm not a Riviera user, but is Linux Mint 18 a supported platform? It looks like Riviera was compiled with a different version of glibc++ than the cocotb PLI module.

saturn77, 2016-10-05:
Jeremy, 
Yes, Linux Mint 18 is essentially Ubuntu 16.04 Xenial. It should be supported. The Riviera-Pro version that I am using is 2016.06 x 64, this is the latest version from Aldec. I could try 2016.02 or a previous version. But it does look like an incompatibility. 
Not sure how to fix this, kind of frustrating. Perhaps a soft link somewhere is needed. 
James

petspats, 2016-10-15:
Hey,
I think i had same issue on Ubuntu 16.
Try this:
`ln -s /usr/lib/x86_64-linux-gnu/libstdc++.so.6  /home/james/Aldec/Riviera-PRO-2016.06-x64/bin/Linux64/libstdc++.so.6`

stuarthodgson, 2017-05-29:
Old I know but this is due to the sim vendors insisting on shipping ancient versions of tools with their tools. Also just using their tools is also not a solution that solves all the problems, I do a lot of cosimulation and need a more up to date compiler than the tools provide.

closed 2017-05-29 by stuarthodgson