cocotb/cocotb #494: Integrate pytest

javValverde, 2016-10-21:
First of all, thanks for this amazing tool. After using it, I cannot think of any other way in which I could write testbenches.

I think it is a great idea if cocotb uses pytest as the testing framework. It supports different kinds of setup/teardowns, parameterization, can generate junit results etc. As far as I know, it is also capable of running tests in unittest and nose formats.

I could contribute in this feature, but I don't really know how to start.

themperek, 2016-10-21:
https://github.com/potentialventures/cocotb/issues/280

eric-wieser, 2018-12-21:
Turns out you can already do this right now (!):
```python
import cocotb
import pytest
import decorator
import concurrent.futures
import Queue

class CocotbExecutor:
    def __init__(self):
        self.queue = Queue.Queue()

    def submit(self, f, *args, **kwargs):
        fut = concurrent.futures.Future()
        @cocotb.coroutine
        def wrapper():
            try:
                ret = yield f(*args, **kwargs)
            except Exception as e:
                fut.set_exception(e)
            else:
                fut.set_result(ret)
            yield Timer(1)
        self.queue.put(wrapper())
        return fut

ex = CocotbExecutor()

@decorator.decorator
def coroutine_test(f, *args, **kwargs):
    return ex.submit(cocotb.coroutine(f), *args, **kwargs).result()

@cocotb.test()
def the_pytest_hack(dut):
    tpe = concurrent.futures.ThreadPoolExecutor(1)
    the_tests = tpe.submit(pytest.main, ['--pyargs', __name__, '-v', '-s'])
    while True:
        try:
            coro = ex.queue.get(timeout=0.5)
        except Queue.Empty:
            if the_tests.done():
                break
        else:
            yield coro
    assert the_tests.result() == 0

@pytest.fixture(scope='module')
def dut():
    return cocotb.regression._dut
```

Once you have that in place, you can write tests (in that file) the pytest way:

```
@coroutine_test
def test_foo(dut)
    yield RisingEdge(dut.clk)
    assert True

@coroutine_test
def test_bar(dut)
    yield RisingEdge(dut.clk)
    assert False
```

You'll get a pytest session embedded within your cocotb one

ktbarrett, 2020-05-05:
xref #667, #1062.

ktbarrett, 2020-11-28:
@eric-wieser I think we can replace the regression manager with pytest using a somewhat similar approach. This depends on #791.

We could change the implementation of `cocotb.test` decorator to add the decorated test coroutine to a scheduler when called and also ensure tests start/end in normal mode. This would translate them to normal functions that pytest can run. With this solution pytest could also run undecorated non-coroutine functions if desired. We can also support running tests using pytest without a simulator, as long as those tests don't call into the simulator.

The end result would look very similar to what we have now, but we could extend it with pytest functionality

```python
@pytest.parameterize("a", [1, 2, 3])
@cocotb.test()
async def test_thing(a):
    # do stuff ...
```

rw1nkler, 2023-10-06:
@ktbarrett Are there any plans to allow for parameterizing cocotb tests with pytest, as in your example?

> ```python
> @pytest.parameterize("a", [1, 2, 3])
> @cocotb.test()
> async def test_thing(a):
>     # do stuff ...
> ```

themperek, 2023-10-06:
@rw1nkler 
See: https://github.com/cocotb/cocotb/blob/master/tests/pytest/test_parallel_cocotb.py

@ktbarrett Should this be closed?

rw1nkler, 2023-10-06:
I believe `test_parallel_cocotb.py` is different than the snippet presented earlier. The parallel test uses pytests parameterization to set parameters for Cocotb Runner invocation, while @ktbarrett's snippet is about parameterizing `cocotb.test` coroutine.

ktbarrett, 2023-10-06:
We won't be able to use pytest for this without modifying how cocotb does test discovery, but we can absolutely provide something similar to replace TestFactory.

adalyn, 2024-06-24:
I'm not a contributor, but am a regular user.  Feel free to remove this comment if inappropriate.

I use cocotb+pytest in a similar but slightly different way that seems appropriate to add to this discussion.

My tests are native pytests written for black-box testing on the actually hardware using parametrized tests and fixtures.  I also develop and run these same tests unchanged in COCOTB simulations.  They have no direct dependency on cocotb (comes via fixtures).

I do this by abstracting the DUT and reentering the cocotb thread using function/coroutine decorators instead.  If I need a cocotb feature from the test such as hierarchical paths for checking internal signals I if/else around it or filter out the test when running in hardware.  Recall this is for blackbox tests that can be run on the hardware.

This abstraction is done by making an API that is called similar to wait() example below (cocotb.function).  I use this abstraction for uarts/cocotb_bus or other interfaces depending on environment.  Usually, the tests can be built up just on a comms api that I build register access on.

cocotbext-pytest: (equivalent to a conftest.py, I've made this an internal package for ease of reuse)
```
@cocotb.test()
def run_pytests(dut):
    assert cocotb.external(pytest.main)(['-p','cocotbext.pytest', '-v', '-s']) == 0

@pytest.fixture(scope='module')
def sim_handle():  # i use sim_handle to make this more obvious when running cooctb
    return cocotb.regression._dut

@pytest.fixture(scope='module')
def wait():
    # use in place of system sleep.  You can also patch sleep but that can have some other issues (needs limited scope)
    @cocotb.function
    def _sleep(sec):
        yield cocotb.Timer(sec, "s")  # function is calling a coroutine which is where it gets back into the cocotb-thread
    return _sleep
```

mock: conftest.py
```
@pytest.fixture(scope='module')
def sim_handle():
    return None

@pytest.fixture(scope='module')
def wait():
    def _sleep(sec):
        loggier.info("Mock sleep for %fs", sec)  # dont sleep when mocking tests - faster runtime
    return _sleep
```

test_cocotb_or_mock_or_hardware.py
```
def test_me(wait, sim_handle):
    wait(1)  # waits 1sec in simulation time, 1sec realtime or 0sec depending on fixture above.  no yield
    if sim_handle is not None:
        assert sim_handle.data_o == 0x12  # just a random assertion for illustration, not require for directed black-box tests
```

When running cocotb I use makefiles and set MODULE=cocotbext.pytest

pytest without cocotb can be run from the command line but I usually have a make target for pytest and some extra flags to indicate if it's hardware or mocked.

tymonx, 2025-03-15:
I can help here 👋 I was able to create `pytest` plugin that allows to use `pytest` transparently and with all `pytest` capabilities. Architecture is similar to `pytest-xdist` where controller, `pytest` parent is calling simulator runner (compilation, elaboration, running test) from test and `pytest` worker running with Cocotb to run and manage Cocotb tests. This allows to use `pytest` fixture, setup, teardown, parameterize, makrers, handle exceptions, use existing pytest plugins!..., etc.  with Cocotb tests. It would be perfect if that feature would be part of Cocotb 2.0. This require some re-work in current regression manager. My proposition is to just abstract regression manager as pluginable component. Then any test framework can be used. We could use any existing Python RPC object-proxy server-client for that. Including from built-in `multiprocessing` package. Because this is needed for `pytest` controller (simulator runner) and `pytest` worker (Cocotb tests).

ktbarrett, 2025-03-15:
@tymonx That would be fantastic! Thank you!

For some more start up info:

I've done a lot of work refactoring things internally to be able to support this. The scheduler can finally be used standalone. cocotb module initialization should be fairly separated out, but will need tweaks. `PYGPI_USERS` should be a useful point to go between loading the RM vs the pytest worker. `Test` objects are a potentially confluence point between the two regression systems. I'm not entirely sure how configurable pytest's test discovery is, so the way the `cocotb.test` and RegressionManager communicate tests to each other may cause some trouble.

tymonx, 2025-03-15:
@ktbarrett 

> I'm not entirely sure how configurable pytest's test discovery is, so the way the cocotb.test and RegressionManager communicate tests to each other may cause some trouble.

It is quite easy. I will explain that on my first `primitive` approach that was using only `pytest` with Cocotb simulator runner 
(similar to existing [cocotb-test](https://github.com/themperek/cocotb-test)) without having `pytest` _"inside"_ running simulator.

1. `pytest` will traverse recursively inside all directories
2. `pytest` will try to load any `*.py` file for introspection (checking functions, classes, ...)
3. `pytest` plugin developer can use `pytest` hooks to detect custom tests from any file type. There is a YAML example: [Working with non-python tests](https://docs.pytest.org/en/stable/example/nonpython.html)
4. What I did, as part of `pytest` plugin, I just load Python file, look for any Python functions decorated with `cocotb.test` decorator and emit it as a new test for `pytest`
5. Because in this primitive approach, Cocotb test couldn't be run directly from `pytest` (`host`), I simple use Cocotb function names to set `MODULE` and `TESTCASE` environment variables for Cocotb simulator runner
6. `stdout` from running Cocotb simulator runner was captured to extract output per test (using regexp to catch test name)
7. `pytest` plugin was managing everything in that way, that 1 Cocotb test == 1 `pytest` test. Hiding all the steps above
8. No modifications in Cocotb guts were needed back then. It was working and tested with Cocotb `1.8.1` and above. I was able to select and run specific Cocotb tests using `pytest -k '<filter>' ...` with all reporting benefits including XML JUnit from `pytest`

But to fully integrate `pytest` with Cocotb (to use fixtures, parameterize etc.) we need `pytest` also inside running simulator. To use it like this:

```pytest
@cocotb.test
@pytest.mark.parametrize("x", (1, 2, 3))
@pytest.mark.parametrize("y", (4, 5))
async my_cocotb_test(dut, x, y, some_pytest_fixture) -> None:
```

This is something like existing [pytest-xdist](https://pypi.org/project/pytest-xdist/) is trying to solve. Where `pytest` controller process (parent process: invoking simulator runner) is controlling `pytest` worker (child process: running simulator + Cocotb + `pytest`).

If we rework and abstract Cocotb Regression Manager, then we could move all heavy lifting part of discovering, creating and managing tests to `pytest` and Cocotb will only run created and provided test from `pytest`. Pytest has this hook [pytest.hookspec.pytest_runtest_call](https://docs.pytest.org/en/stable/reference/reference.html#pytest.hookspec.pytest_runtest_call). It could pass discoverd and created test function to Cocotb:

```pytest
# Silly pseudo code
def pytest_runtest_call(self, item: Item) -> None:
    self.cocotb.run_test(item.func)
```

This also open discussion to use whole `pytest` facility instead of developing own (like custom `cocotb.parametrize` in Cocotb 2.0) and focus on other things.

tymonx, 2025-03-15:
I think better to explain everything by using real code example in action 😅 I will try to prepare a Pull Request PoC. Be aware that I would be lazy here and I will integrate `pytest` more tightly inside Cocotb guts. But this will be also a good opportunity to discuss how this kind of integration in Cocotb should look like from Cocotb community point of view. Personally I think it should be more pluginable where `pytest` is only an option and not as strong required dependency.

ktbarrett, 2025-03-15:
> * `pytest` will try to load any `*.py` file for introspection (checking functions, classes, ...)

There are many test which introspect the DUT when creating tests (parameterization values, skips, etc.). The only way to safely find all test is to first load a simulation and then find tests.

> * What I did, as part of `pytest` plugin, I just load Python file, look for any Python functions decorated with `cocotb.test` decorator and emit it as a new test for `pytest`

In 2.0 cocotb tests are now stuck in `__cocotb_tests__` variable in the defining module.

> Personally I think it should be more pluginable where pytest is only an option and not as strong required dependency.

I agree. That was the plan. We will definitely need to maintain the RegressionManager for a while before people have an opportunity to move to the new system, or for users who can't stomach pytest dependency.

tymonx, 2025-03-15:
> There are many test which introspect the DUT when creating tests (parameterization values, skips, etc.). The only way to safely find all test is to first load a simulation and then find tests.

@ktbarrett Can you point me to existing tests/examples/documentation 🤔 I could utilize [pytest_generate_tests](https://docs.pytest.org/en/stable/reference/reference.html#pytest.hookspec.pytest_generate_tests) hook for that. It is used example when test is decorated with `@pytest.mark.parametrize`.

ktbarrett, 2025-03-15:
I'm not sure there are any examples, but try accessing a signal through `cocotb.top` in a test module like so:

```python
@cocotb.test(skip=not cocotb.top.ALLOW_VALID_GAPS.value)
async def test_valid_gaps(_) -> None:
    ...
```