cocotb/cocotb #497: Hierarchy access for generate statement

kirknodeng, 2016-11-08:
When we instantiate some block with generate statement. 
How to access that hierarchical module?

example) When I want to access "SubBlock[3].u_SubModule.a". 
genvar i;
generate 
  for(i=0;i<5;i=i+1) begin : SubBlock
    SubModule u_SubModule(
       .a(...),
       .b(...),
    );
  end
endgenerate

First, I tried to do this 
RisingEdge(dut.SubBlock[3].u_SubModule.a) -> cocotb can't find SubBlock

and, Second
RisingEdge(dut.u_SubModule.a) -> cocotb find their object. but, which module accessed? 
                                                         I instantiated five modules.

leftink, 2016-11-08:
The first way should work I believe.  What simulator are you using?

You should try running test_array.  It tests something similar.

kirknodeng, 2016-11-09:
I tried to running test_array using VCS. It's similar. but, different.

So, I want to make this case test into test_cases. Can I make it? 
It is first time to make test case and push request. Is it okay?

ktbarrett, 2020-05-05:
If you are still having trouble create a new issue.

mattvenn, 2020-09-21:
I believe there is still an issue with hierarchy access - please see this MVCE
https://github.com/mattvenn/cocotb-mvce-generate-bug/blob/master/test.py

mattvenn, 2020-09-21:
This is with icarus 10.3 compiled from source

ktbarrett, 2020-09-21:
Running with debugging I get [this](https://github.com/cocotb/cocotb/files/5255524/output.log). If I set line 9 to `True`, the test passes and I get [this](https://github.com/cocotb/cocotb/files/5255513/output.log). I'll look into this more later, but there is definitely a problem with the discovery of pseudo-region. Calling in the expert on pseudo-regions @garmin-mjames.

ktbarrett, 2020-09-21:
I see the problem and I'm astonished that this hasn't been caught before...

When attempting to access `genblk1`, the VPI implementation is unaware that it is a pseudo-region, causing `vpi_handle_by_name` to fail. When iterating over "top" beforehand, the pseudo-region is discovered, created, and cached so that following accesses succeeds. There are a lot of ways to fix this issue, I'm not sure which is best.

I'm starting to think we should be iterating over all related objects, properly creating and caching those related-objects, whenever we access an object. This approach is also the best solution for #1128.

garmin-mjames, 2020-09-21:
@ktbarrett It might be enough to call `_discover_all` before looking for sub handles.

ktbarrett, 2020-09-22:
```diff
--- a/cocotb/handle.py
+++ b/cocotb/handle.py
@@ -249,6 +249,11 @@ class HierarchyObject(RegionObject):
         except KeyError:
             pass
 
+        if not self._discovered:
+            self._discover_all()
+
+        return self.__get_sub_handle_by_name(name)
+
         # Cache to avoid a call to the simulator if we already know the name is
         # invalid. Unclear if we care, but we had this before.
         if name in self._invalid_sub_handles:
```

This resolves the problem. If someone wants they could make a PR.

closed 2020-10-04 by ktbarrett

garmin-mjames, 2020-10-19:
Adding a follow-up note, icarus doesn't reveal `vpiGenScopeArray` through the VPI.
cocotb VPI code around handling pseudo-regions assumes tool support for `vpiGenScopeArray`:

https://github.com/cocotb/cocotb/blob/a77f8b01328a3c233acb98001e6ded551ea02362/cocotb/share/lib/vpi/VpiImpl.cpp#L305

ktbarrett, 2020-10-19:
Is that a correct assumption? Is Icarus lacking support? Or is it something we need to handle on our end?

garmin-mjames, 2020-10-19:
icarus supports only `vpiGenScope`, but not the enclosing `vpiGenScopeArray`. I think it's missing functionality from the standard.
Looking at Section 37.83 of the SV standard, it definitely looks like a **gen scope** is always inside a **gen scope array**.

In other simulators I'm pretty sure I've seen an array even if there's only one generate inside. I'll double check though.

garmin-mjames, 2020-10-19:
The VHPI code does the iteration in `VhpiImpl.cpp` looking for a match and creates the pseudo-region. The comments indicate inconsistency across tools.
We could move the iteration from #2079 out of `handle.py` into the VPI code to match the behavior.

PR #2079 (2020-10-04): Iterate over handle if sub-handle lookup fails
Fixes #497. 

Pseudo-region lookup may fail, so iterate over sub-handles using `_discover_all` and try lookup again.

Once this was added, it uncovered another issue where iterating on a module would find the vpiPort for
a variable and create a `HierarchyObject` rather than the underlying variable itself which becomes a
`ModifiableObject` or subclass.

---
Todo:
- [x] Test
- [x] Newsfragment?