cocotb/cocotb #498: Windows 10 64-bit + Cocotb instruction

OrionBytes, 2016-11-17:
It works well with
- http://msys2.github.io/
- MSYS + internal make  ( >pacman -S make )
- Windows GCC 5.1.0 64 bit from here - http://tdm-gcc.tdragon.net/
- Anaconda Python3 64-bit ( after >conda install mingw libpython )

And small fix in Makefile.pylib.Msys (Python3 print syntax)
https://gist.github.com/OrionBytes/fa4b46f956c5dc4637fd45f2e8c768a4

I'm not sure is it optimal solution or not - but it seems working with QuestaSim

OrionBytes, 2016-11-17:
And it works with Intel Python - here is recipe for Anaconda
https://software.intel.com/en-us/articles/using-intel-distribution-for-python-with-anaconda

Visual Studio supports both Python(PTVS) and GCC (plugins).
In addition it's possible to use Performance Analysis from Intel in case of intensive verification flow
https://software.intel.com/en-us/videos/performance-analysis-of-python-applications-with-intel-vtune-amplifier

LinuxFanatic777, 2016-11-29:
Thank you, interesting info!
I plan to use cocotb also on Windows 64 bit, because developers are using Windows while I prefer to run cocotb on Linux. Glad to see that Windows OS is also supported.

LinuxFanatic777, 2016-12-08:
I trying to reproduce your success. Makefile.pylib.Msys fix was helpful. I'm using Windows 7 64-bit and Modelsim Starter Edition which is 32-bit. Due to 32 bit of modelsim I have to use msys2 32-bit + Anaconda 3 32-bit. It builds libcocotbutils.dll and then I see the following error:

```
make results.xml
make[1]: Entering directory '/f/tests/test_cocotb'
echo "if [file exists work] {vdel -lib work -all}" > sim_build/runsim.do
echo "vlib work" >> sim_build/runsim.do
echo "vmap -c" >> sim_build/runsim.do
echo "vmap work work" >> sim_build/runsim.do
echo "vlog -work work +define+COCOTB_SIM -timescale 1ns/100ps -mfcu +acc=rmb -sv  /f/tests/test_cocotb/top_cocotb.v" >> sim_build/runsim.do
echo "vsim -onfinish stop -pli libvpi.dll work.top_cocotb" >> sim_build/runsim.do
echo "add wave -recursive /*" >> sim_build/runsim.do
make -C /c/work/cocotb-master/lib/gpi_log SIM=modelsim
make[2]: Entering directory '/c/work/cocotb-master/lib/gpi_log'
gcc -m32 -g -DDEBUG -shared -DFILTER -DMODELSIM -c -I/c/work/cocotb-master/include -IC:\anaconda332\include  -o /c/work/cocotb-master/build/obj/i686/gpi_logging.o gpi_logging.c
In file included from C:\anaconda332\include/Python.h:8:0,
                 from gpi_logging.c:30:
C:\anaconda332\include/pyconfig.h:291:23: error: unknown type name '__int64'
 # define PY_LONG_LONG __int64
```
I'm using GCC 5.3 from msys2.

OrionBytes, 2016-12-08:
I'm not sure about 64bit vs 32bit conflicts details.
But you can also try windows GCC here - I use the same for the 64 bit successfully
http://tdm-gcc.tdragon.net/download
Under msys2 I keep only make

LinuxFanatic777, 2016-12-08:
It's funny but now I got all modules compiled successfuly after "pacman -R gcc" and installing of TDM32 GCC! Now I'm getting error "Load of libvpi.dll failed". When using Linux 64-bit I have to use Anaconda 32-bit and all other 32-bit software to get linked with Modelsim (which is 32-bit). My assumption is that Modelsim for Windows also require such trick. However after this "Load of libvpi.dll failed" it seems that Modelsim for Windows could use 64-bit VPI iterface...
```
$ cat run_win
#!/bin/bash

export PYTHON_DIR=/c/anaconda332
export PATH=/c/TDM-GCC-32/bin:$PATH
export PATH=/c/intelFPGA/16.1/modelsim_ase/win32aloem:$PATH

rm -Rf sim_build/*
mkdir sim_build

export GUI=1
make ARCH=i686 SIM=modelsim
```
Modelsim internal console output:
```
vsim -onfinish stop -pli "libvpi.dll" work.top_cocotb
Start time: 15:15:10 on Dec 08,2016
Loading sv_std.std
Loading work.top_cocotb
Loading libvpi.dll
** Error: (vsim-3193) Load of "libvpi.dll" failed: File not found.
** Error (suppressible): (vsim-PLI-3002) Failed to load PLI object file "libvpi.dll".
   Time: 0 ns  Iteration: 0  Root: /
Error loading design
Error: Error loading design
```

OrionBytes, 2016-12-08:
Yes, I think that TDM-gcc has better linking with windows python.
What is the make output without ARCH=i686
just > make SIM=modelsim

And what is the output with Icarus (common simulator for the Cocotb checks)?
You can find windows Icarus here - http://bleyer.org/icarus/
Don't forget to add the folder to PATH after installation

LinuxFanatic777, 2016-12-08:
Same result without ARCH.
```
/c/intelFPGA/16.1/modelsim_ase/win32aloem/vsim.exe: PE32 executable (console) Intel 80386, for MS Windows
/c/work/cocotb-master/build/libs/i686/libvpi.dll: PE32 executable (DLL) (console) Intel 80386, for MS Windows
```
Both executables is 32-bit. Why it can not find libvpi.dll? OK, I just put libvpi.dll into dir where vsim.exe and now it can see library, however unable to load:
```
# Loading work.top_cocotb
# Loading C:/intelFPGA/16.1/modelsim_ase/win32aloem/libvpi.dll
# ** Error: (vsim-3193) Load of "C:/intelFPGA/16.1/modelsim_ase/win32aloem/libvpi.dll" failed: DLL dependent library not found.
# ** Error (suppressible): (vsim-PLI-3002) Failed to load PLI object file "C:/intelFPGA/16.1/modelsim_ase/win32aloem/libvpi.dll".
#    Time: 0 ns  Iteration: 0  Root: /
# Error loading design
```
Then I copy **all** DLLs from c:\work\cocotb-master\build\libs\i686\ to vsim.exe path - same result.
```
$ ldd libvpi.dll
        ntdll.dll => /c/Windows/SysWOW64/ntdll.dll (0x77320000)
        kernel32.dll => /c/Windows/syswow64/kernel32.dll (0x74f90000)
        KERNELBASE.dll => /c/Windows/syswow64/KERNELBASE.dll (0x75c00000)
        ??? => ??? (0x6f6c0000)
        ??? => ??? (0x10000000)
        ??? => ??? (0x73c30000)
        ??? => ??? (0x74ee0000)
```
Also, when run Dependency Walker I see that it needs more DLLs: DBGCORE GPSVC.DLL IESHIMS.DLL

OrionBytes, 2016-12-08:
Hm, I will try to check your situation in VM on Saturday.
Before you can check Icarus, but the thing is Win64 I think.

LinuxFanatic777, 2016-12-08:
Yes please, try Modelsim Starter Edition (which is free to download) in Windows **64-bit**.
Unfortunately all developers at my work are using Windows 64-bit, also they are highly dependent on vendor libraries and components so Icarus is helpless (however I love Icarus very much :) )...

OrionBytes, 2016-12-10:
Sorry, still without time to check. Will try later.

LinuxFanatic777, 2016-12-12:
May be it will be easier to fix this error (instead of Modelsim 32-bit), here is my result with Modelsim 64-bit (TDM-GCC-64, MSys2 64-bit, Anaconda 64-bit):
```
make results.xml
make[1]: Entering directory '/f/tests/test_cocotb'
echo "if [file exists work] {vdel -lib work -all}" > sim_build/runsim.do
echo "vlib work" >> sim_build/runsim.do
echo "vmap -c" >> sim_build/runsim.do
echo "vmap work work" >> sim_build/runsim.do
echo "vlog -work work +define+COCOTB_SIM -timescale 1ns/100ps -mfcu +acc=rmb -sv  f:/tests/test_cocotb/top_cocotb.v" >> sim_build/runsim.do
echo "vsim -onfinish stop -pli libvpi.dll work.top_cocotb" >> sim_build/runsim.do
echo "add wave -recursive /*" >> sim_build/runsim.do
make -C /c/work/cocotb-master/lib/gpi_log SIM=modelsim
make[2]: Entering directory '/c/work/cocotb-master/lib/gpi_log'
gcc -g -DDEBUG -shared -DMS_WIN64 -DFILTER -DMODELSIM -shared -Wl,-no-undefined -Wl,-enable-runtime-pseudo-reloc-v2 -Wl,--enable-auto-import -L/c/anaconda364/libs -o /c/work/cocotb-master/build/libs/x86_64/libgpilog.dll /c/work/cocotb-master/build/obj/x86_64/gpi_logging.o   -lPython35
C:/anaconda364/libs/Python35.lib: error adding symbols: File in wrong format
collect2.exe: error: ld returned 1 exit status
make[2]: *** [/c/work/cocotb-master/makefiles/Makefile.rules:58: /c/work/cocotb-master/build/libs/x86_64/libgpilog.dll] Error 1
make[2]: Leaving directory '/c/work/cocotb-master/lib/gpi_log'
make[1]: *** [/c/work/cocotb-master/lib/Makefile:43: /c/work/cocotb-master/build/libs/x86_64/libgpilog.dll] Error 2
make[1]: Leaving directory '/f/tests/test_cocotb'
make: *** [/c/work/cocotb-master/makefiles/Makefile.sim:72: sim] Error 2
```
However, Modelsim 32-bit Started Edition still highly interesting, because it's free for anyone.

OrionBytes, 2016-12-12:
Hm... strange. TDM-GCC-64, MSys2 64-bit, Anaconda 64-bit works well for me with QuestaSim.
I'm not sure about >make results.xml
But >make SIM=questa in cocotb main directory
Or >make SIM=questa in examples/<any_example>
Or with my modules, works well enough

OrionBytes, 2016-12-12:
Try to check the new issue - https://github.com/potentialventures/cocotb/issues/502
As I understand they got working cocotb under windows too

LinuxFanatic777, 2016-12-12:
My command is:
```
make SIM=modelsim
```
This "make results.xml" is just an output during command execution.

jeremyherbert, 2016-12-12:
Hi @OrionBytes 

Sorry to chime in at this late stage, but your fix in the gist will break for python 2.7 because print is not a function. Have a look at the code I posted in the issue you linked (#502) to see one way of dealing with this (I think it's the most common way) so that it works with both versions of python.

OrionBytes, 2016-12-12:
Hi, @jeremyherbert 
thank you for the comment - sure, my changes are for Python 3 only.
I saw https://github.com/potentialventures/cocotb/pull/495  - that's why I didn't submit a pull.
In addition - I'm not an expert to advise the best option to deal with Python 2 and 3 at the same time.
Your solution looks perfect!

LinuxFanatic777, 2016-12-13:
2 OrionBytes: I found similar topic here: https://github.com/potentialventures/cocotb/issues/428 will try these solutions

OrionBytes, 2016-12-16:
Maybe some way like:
>>
Try this...
    Download gendef for your version of mingw (32 or 64 bit), and in msys shell...
    Run gendef /c/windows/system32/python32.dll
    Run dlltool -D python32.dll -d python32.def -l libpython32.a
    Copy libpython32.a to your ./python32/libs directory.

If your libpython32.a file is 0 bytes, something went wrong. Double-check that you downloaded the correct version of gendef for your version of mingw/msys. If you're running a 64-bit build, you'll likely have to download the gendef binaries and compile yourself, but that's simple enough.
Hope that helps.
>>
from http://stackoverflow.com/questions/6731100/link-to-python-with-mingw

LinuxFanatic777, 2016-12-16:
Excuse me, but my problem with Python35.lib file with LIB extensions. I had successfully built libpython35.a with gendef/dlltool but this file is not the reason of problem. Or I misunderstand your suggestion?

LinuxFanatic777, 2016-12-26:
Is there any news? I want to solve this problem and then send a patch to cocotb for better support of Windows.

LinuxFanatic777, 2017-01-28:
I created cross-platform script on Python 3 that launch my testing environment on Linux and Windows.
During first run it creates configuration file with PATH lines to be configured by developer (paths to Modelsim, Anaconda, Mingw/TDM-GCC on Windows, and of course PATH to cocotb). If all tools visible then script patching cocotb files and start testing.

If anybody is interested in such script I can show source code.

iHumus, 2018-06-04:
@OrionBytes can you help me a bit?
I also work with V3S on VisualStudio, and I'm trying to run cocotb.
I added the visualGDB, but I can't figure what are the next steps.
can you guide me?

LinuxFanatic777, 2018-06-04:
Which exact problems do you have? May be I can help you too?
And of course, is there any vital reason to use Visual Studio necessarily?
I think UNIX-tools is a target platform, and MinGW give the way of supporting on Windows too. But VS might be "alien technology" for using Cocotb...

iHumus, 2018-06-05:
All our development is on Windows and VisualStudio, so it will be nice to have everything on the same IDE.
But I'll be happy if it will just work.

I tried with msys2, but got errors when MAKE.
and I want to integrate it to VisualStudio, but don't know how

LinuxFanatic777, 2018-06-05:
1) Visual Studio Code is cross-platform IDE
2) Cocotb is Python-based framework
3) I do not see any reason of integrating Cocotb into IDE
4) Which exact errors do you have? Show us the link to logs
5) Are you sure you are not trolling us? :)

iHumus, 2018-06-05:
1. VisualStudio Code is cross-platform, but VisualStudio isn't, and V3S only works on it.
2. I know that, but @OrionBytes wrote that he was able to run everything in one place (since VisualStudio has python support)
3. again, just for simplicity, when we have everything in one place (but it's not necessary of course)
4. Here is the log when I try to run it on msys2: [https://gist.github.com/iHumus/f809a4974f77605de87e8b18420195fc](url)
5. Just trying to get some help...

elgorwi, 2018-06-07:
It looks like you're using 32 bit modelsim with 64 bit tools. I recommend installing 32 bit msys2 and 32 bit python (eg. anaconda) in a sensible location (Eg. C:\tools\) and prepending them to your PATH variable.
You can check if the correct versions are found by typing following on a msys2 shell:
>which gcc
>which python
You can also try running the endian swapper example after uncommenting the debug in Makefile.pylib. That should give some more clues. (don't forget to comment out again to actually run a sim)
Also do a search for modelsim, there's a few threads already with more details.

closed 2020-04-09 by eric-wieser

themperek, 2020-04-09:
Easiest way to use conda httpss://github.com/cocotb/cocotb/wiki/Tier-2-Setup-Instructions#conda-based-installation-all-os

omkars3, 2022-11-20:
Hi,
I am using cocotb in windows through ming64. Does anyone know how to install modelsim inside ming64 ?
Installed Modelsim(Intel version) in windows but it did not work.

themperek, 2022-11-20:
>Installed Modelsim(Intel version) in windows but it did not work.

What did not work?

omkars3, 2022-11-20:
Ming64 is not able to locate modelsim.

themperek, 2022-11-20:
>Ming64 is not able to locate modelsim.

Need to add path to modalism (`vsim`) to `PATH` environment variable?

omkars3, 2022-11-20:
Yes I am getting this error itself